coolsnowwolf/lede #4360: 云编译默认支持SSR+不隐藏

z826540272, 2020-04-17:
用 https://github.com/P3TERX/Actions-OpenWrt 云编译脚本
P3TERX脚本灵活性和功能方面优于当前仓库脚本,如果你使用当前仓库脚本,将CONFIG段内容复制到新建.config文件上传即可.
云编译配置文件 diy.sh 写入以下内容即默认开启ssr菜单显示和添加 helloword (SSR+) 仓库内容.
echo 0xDEADBEEF > package/base-files/files/config/google_fu_mode
echo "src-git helloworld https://github.com/fw876/helloworld" >> feeds.conf.default

z826540272, 2020-04-17:
上面缺少了一行,完整文件为:
`#!/bin/bash
mkdir package/base-files/files/config
echo 0xDEADBEEF > package/base-files/files/config/google_fu_mode
echo "src-git helloworld https://github.com/fw876/helloworld" >> feeds.conf.default`

garypang13, 2020-04-18:
光这样还是不行的,
diy.sh后还需要执行 ./scripts/feeds update -a && ./scripts/feeds install -a

FengNning, 2020-04-18:
我是这样的
#!/bin/bash
mkdir package/base-files/files/config
echo 0xDEADBEEF > package/base-files/files/config/google_fu_mode
echo "src-git helloworld https://github.com/fw876/helloworld" >> feeds.conf.default
 ./scripts/feeds update -a && ./scripts/feeds install -a

garypang13, 2020-04-18:
> 我是这样的
> #!/bin/bash
> mkdir package/base-files/files/config
> echo 0xDEADBEEF > package/base-files/files/config/google_fu_mode
> echo "src-git helloworld https://github.com/fw876/helloworld" >> feeds.conf.default
> ./scripts/feeds update -a && ./scripts/feeds install -a

对的,这样就没问题.

mjyhj, 2020-04-18:
mkdir package/base-files/files/config
echo 0xDEADBEEF > package/base-files/files/config/google_fu_mode

这个config早就已经不需要了，现在没有也不会隐藏的

martin0607, 2020-04-18:
我是直接在workflow文件里添加这个的
[image]

iamggs, 2020-04-18:
> 我是直接在workflow文件里添加这个的
> [image]

学习了。

madcat0513, 2020-04-18:
为啥我改了feeds.conf.default后，menuconfig 里还没出来？

tengyuankoo, 2020-04-19:
git clone https://github.com/fw876/helloworld package/luci-app-ssr-plus
./scripts/feeds update -a
./scripts/feeds install -a

z826540272, 2020-04-19:
> git clone https://github.com/fw876/helloworld package/luci-app-ssr-plus
> /scripts/feeds update -a && ./scripts/feeds install -a
不需要更新feeds,因为你直接复制到仓库里了,而不是feed源

closed 2020-07-04 by coolsnowwolf