coolsnowwolf/lede #4364: Deepin 15.11 编译失败，

mcucore, 2020-04-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
编译失败，找不到PO2LMO
编译系统DEEPIN 15.11
### (2) 路由器型号和固件版本

A：
X86-32
### (3) 详细日志

A：
po2lmo ./po/zh-cn/ssr-plus.po /home/user/lede/build_dir/target-i386_pentium4_musl/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus/usr/lib/lua/luci/i18n/ssr-plus.zh-cn.lmo
bash: po2lmo: command not found
Makefile:120: recipe for target '/home/user/lede/build_dir/target-i386_pentium4_musl/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus.installed' failed
make[3]: *** [/home/user/lede/build_dir/target-i386_pentium4_musl/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus.installed] Error 127
make[3]: Leaving directory '/home/user/lede/feeds/helloworld/luci-app-ssr-plus'
time: package/feeds/helloworld/luci-app-ssr-plus/compile#0.13#0.04#0.19
package/Makefile:111: recipe for target 'package/feeds/helloworld/luci-app-ssr-plus/compile' failed
make[2]: *** [package/feeds/helloworld/luci-app-ssr-plus/compile] Error 2
make[2]: Leaving directory '/home/user/lede'
package/Makefile:107: recipe for target '/home/user/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile' failed
make[1]: *** [/home/user/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
/home/user/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

mcucore, 2020-04-18:
应该是编译顺序的问题，床新编译就会这样，先make package/feeds/luci/luci-base/compile V=99，再编译就好了。先不关问题，老大看到我再关！

EnnawYang, 2020-04-18:
[image]
装了个deepin跑了一遍，很顺畅啊。 话说64位它不香吗？

mcucore, 2020-04-18:
挺喜欢DEEPIN，本地化比较好，需要的都有了。编译固件也好用，可以和WINDOWS一样3389远程桌面操作，不用在手机那么小的屏幕SSH了。

closed 2020-04-18 by mcucore