coolsnowwolf/lede #4368: make[4]: *** [node] Error 2 make: *** [world] Error 2求如何解决

kissaway, 2020-04-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
编译出错
### (1) 具体问题

A

### (2) 路由器型号和固件版本

A：Raspberry pi 3

### (3) 详细日志

A：See <http://bugs.openwrt.org/> for instructions.
tools/v8_gypfiles/v8_base_without_compiler.target.mk:594: recipe for target '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/api/api.o' failed
make[5]: *** [/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/api/api.o] Error 4
rm 44d6460fd90b9a06296029e141b24c213bac5c41.intermediate f74fbe74ca082eff7f73d5013e545f35d8c0e5e0.intermediate 294532b628f5c66b71a193e1d12b2cf24b97d30d.intermediate 4e6ac6541edf1a300093ee349c6da4a392c42aea.intermediate
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/node-v12.15.0'
Makefile:150: recipe for target '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/node-v12.15.0/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#110.93#11.81#197.91
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
package/Makefile:107: recipe for target '/home/ubuntu/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

ecco2005, 2020-04-18:
make FORCE_UNSAFE_CONFIGURE=1 -j1 V=s
重新拉源，然后用这个编译。

关于这个recipe for tartget XXX failed真的是伤脑筋，找不到具体问题在哪。

kissaway, 2020-04-18:
> make FORCE_UNSAFE_CONFIGURE=1 -j1 V=s
> 重新拉源，然后用这个编译。
> 
> 关于这个recipe for tartget XXX failed真的是伤脑筋，找不到具体问题在哪。

我以为是系统的事儿，我连系统都重做了୧( ⁼̴̶̤̀ω⁼̴̶̤́ )૭

closed 2020-04-19 by coolsnowwolf