coolsnowwolf/lede #4369: lean固件怎么添加passwall

z826540272, 2020-04-18:


ecco2005, 2020-04-18:
经测试，passwall科学上网的性能，远不如SSR-PLUS。负载均衡功能也没有你想象中的好用

guaguagaga, 2020-04-18:
负载均衡IP跳来跳去很蛋疼的啊

Twinzo1, 2020-04-18:
src-git lienol https://github.com/Lienol/openwrt-package

Promix953, 2020-04-18:
passwall停止开发了

DanielChen008, 2020-04-19:
passwall 开发者都弃更了，还要了做啥，而且我之前集成后对比过，性能确实不如 ssr-plus

z826540272, 2020-04-19:
git clone https://github.com/kenzok8/openwrt-packages.git package/openwrt-packages
git clone https://github.com/kenzok8/small.git package/small
解决了,执行上述两行命令后可在menu config菜单中看到luci-app-passwall

z826540272, 2020-04-19:
> src-git lienol https://github.com/Lienol/openwrt-package
你的这个不对,lienol仓库现在没有passwall

M1A6I3L, 2020-04-19:
楼上几位也不能一棒子打死PASSWALL啊，性能又不是一切。

至少对我来说多节点代理挺香的，俩设备代理日本，剩下的是香港。

passwall第653个commits就删代码了，需要可以用这位的，正好在652。
https://github.com/cnxzp/openwrt-package

badcrazy, 2020-04-25:
之前试过，当初奔着均衡去的，但是我的订阅解析出来都是乱码，我就放弃了

closed 2020-07-04 by coolsnowwolf