coolsnowwolf/lede #4371: rg100a-aa编译报错

jiatongwu, 2020-04-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 编译报错

### (2) 路由器型号和固件版本

A：rg100a-aa

### (3) 详细日志

A：
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.c
Hunk #1 succeeded at 1905 (offset 1 line).
Hunk #2 succeeded at 1944 (offset 1 line).

Applying /mnt/vdy1/lede/target/linux/brcm63xx/patches-4.14/001-4.15-13-bcm63xx_enet-remove-unneeded-include.patch using plaintext:
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.h

Applying /mnt/vdy1/lede/target/linux/brcm63xx/patches-4.14/001-4.16-01-bcm63xx_enet-just-use-enet-as-the-clock-name.patch using plaintext:
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.c
Hunk #1 succeeded at 1720 (offset 1 line).
Hunk #2 succeeded at 1760 (offset 1 line).

Applying /mnt/vdy1/lede/target/linux/brcm63xx/patches-4.14/001-4.16-02-bcm63xx_enet-use-platform-data-for-dma-channel-numbe.patch using plaintext:
patching file arch/mips/bcm63xx/dev-enet.c
patching file arch/mips/include/asm/mach-bcm63xx/bcm63xx_dev_enet.h
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.c
Hunk #1 succeeded at 1756 (offset 1 line).
Hunk #2 succeeded at 1787 (offset 1 line).

Applying /mnt/vdy1/lede/target/linux/brcm63xx/patches-4.14/001-4.16-03-bcm63xx_enet-remove-pointless-mac_id-check.patch using plaintext:
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.c
Hunk #1 succeeded at 1791 (offset 1 line).

Applying /mnt/vdy1/lede/target/linux/brcm63xx/patches-4.14/001-4.16-04-bcm63xx_enet-use-platform-device-id-directly-for-mii.patch using plaintext:
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.c
Hunk #1 succeeded at 1754 (offset 1 line).
Hunk #2 succeeded at 1821 (offset 1 line).
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.h

Applying /mnt/vdy1/lede/target/linux/brcm63xx/patches-4.14/001-4.21-01-BCM63XX-fix-switch-core-reset-on-BCM6368.patch using plaintext:
patching file arch/mips/bcm63xx/reset.c
Hunk #1 FAILED at 120.
1 out of 1 hunk FAILED -- saving rejects to file arch/mips/bcm63xx/reset.c.rej
Patch failed!  Please fix /mnt/vdy1/lede/target/linux/brcm63xx/patches-4.14/001-4.21-01-BCM63XX-fix-switch-core-reset-on-BCM6368.patch!
Makefile:29: recipe for target '/mnt/vdy1/lede/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.14.172/.prepared_cf55965be7a449bf9b8312b648014344' failed
make[4]: *** [/mnt/vdy1/lede/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.14.172/.prepared_cf55965be7a449bf9b8312b648014344] Error 1
make[4]: Leaving directory '/mnt/vdy1/lede/target/linux/brcm63xx'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/mnt/vdy1/lede/target/linux'
time: target/linux/compile#34.65#38.55#64.85
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/mnt/vdy1/lede'
target/Makefile:18: recipe for target '/mnt/vdy1/lede/staging_dir/target-mips_mips32_musl/stamp/.target_compile' failed
make[1]: *** [/mnt/vdy1/lede/staging_dir/target-mips_mips32_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/mnt/vdy1/lede'
/mnt/vdy1/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

jiatongwu, 2020-04-18:
我再重新pull 再编译试试

jiatongwu, 2020-04-19:
我重新pull了 还是编译报错

  427 git pull
  428  ls
  429  make menuconfig
  430  ./scripts/feeds update -a && ./scripts/feeds install -a
  431  make defconfig
  432  make menuconfig
  433  history
  434  make -j16 download v=s && make -j4 V=s
  435  history

Applying /mnt/vdy1/lede2/target/linux/brcm63xx/patches-4.14/001-4.16-04-bcm63xx_enet-use-platform-device-id-directly-for-mii.patch using plaintext:
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.c
Hunk #1 succeeded at 1754 (offset 1 line).
Hunk #2 succeeded at 1821 (offset 1 line).
patching file drivers/net/ethernet/broadcom/bcm63xx_enet.h

Applying /mnt/vdy1/lede2/target/linux/brcm63xx/patches-4.14/001-4.21-01-BCM63XX-fix-switch-core-reset-on-BCM6368.patch using plaintext:
patching file arch/mips/bcm63xx/reset.c
Hunk #1 FAILED at 120.
1 out of 1 hunk FAILED -- saving rejects to file arch/mips/bcm63xx/reset.c.rej
Patch failed!  Please fix /mnt/vdy1/lede2/target/linux/brcm63xx/patches-4.14/001-4.21-01-BCM63XX-fix-switch-core-reset-on-BCM6368.patch!
Makefile:29: recipe for target '/mnt/vdy1/lede2/build_dir/target-mips_mips32_musl/linux-brcm63xx_generic/linux-4.14.176/.prepared_711a1caf77aab4c873f89fded2946e0c' failed
make[4]: *** [/mnt/vdy1/lede2/build_dir/target-mips_mips32_musl/linux-brcm63xx_generi

xpc0938, 2020-04-19:
看起来是打patch失败了，pull前可能还要执行:
make clean
rm -rf ./tmp
(还有也可以对出错的包单独clean,但是这样太麻烦，说不定后面还有其他错误; make clean 一般可以解决此类问题;如果不能解决问题，就把make clean换成make dirclean，还不能解决就改成make distclean,注意执行前备份.config文件。
解释:
make clean
[清除编译产生的中间文件]
make dirclean
[make clean]+[清除交叉编译工具及工具链目录]
make distclean
[清除所有相关的东西，包括下载的软件包，配置文件，feed内容等; 这个命令，下载的包也会被删除，下次编译会重新下载软件包，很慢，一般慎用！]
最后如果还是不能解决问题，就真的需要找具体原因了。

jiatongwu, 2020-04-19:
那我重新 git clone 一下 再重新编译 我再试试

AmadeusGhost, 2020-04-19:
#4372 ,等合并或者自己照着改下

PR #4372 (2020-04-19): bcm63xx: refresh patches
fix #4371
closed 2020-04-19 by coolsnowwolf