coolsnowwolf/lede #4373: 编译ssr plus提示错误 缺少依赖,怎么添加依赖呢

gc89925, 2020-04-18:
编译ssr plus提示错误
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
 * 	microsocks
 * 	tcpping
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * resolve_conffiles: Existing conffile /home/gc89925/1907/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/gc89925/1907/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/gc89925/1907'
package/Makefile:108: recipe for target '/home/gc89925/1907/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/gc89925/1907/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/gc89925/1907'
/home/gc89925/1907/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

gc89925, 2020-04-18:
gc89925@tw01:~/1907$ make package/feeds/luci/luci-base/compile V=99
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'microsocks', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'tcpping', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'redsocks2', which does not exist
WARNING: Makefile 'package/feeds/lienol/luci-app-verysync/Makefile' has a dependency on 'verysync', which does not exist
make[1]: Entering directory '/home/gc89925/1907'
make[2]: Entering directory '/home/gc89925/1907/feeds/luci/modules/luci-base'
make[2]: Leaving directory '/home/gc89925/1907/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.05#0.48#1.57
make[2]: Entering directory '/home/gc89925/1907/package/libs/toolchain'

closed 2020-04-19 by coolsnowwolf

xiehuayi75, 2021-04-19:
依赖问题可以执行lede / package $下运行或openwrt / package $下运行

git clone https://github.com/kenzok8/small.git