coolsnowwolf/lede #4374: 重启openwrt后,ssr运行模式为GFW列表模式,不能访问youtube

githubjerry001, 2020-04-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如果ssr运行模式为"GFW列表模式",重启路由器后,则不能访问youtube,手动切换回"绕过中国大陆IP"模式,则可以访问youtube,这时切换回"GFW列表模",youtube可以正常访问(有时要来回切换数次才可). 因为设置路由器每天自动重启,这样每天都要手动切换运行模式几次才能在GFW模式访问youtube.
在强制走代理的域名里加了"youtube.com",但无效.

### (2) 路由器型号和固件版本

A：J1900, OpenWrt R20.4.8

### (3) 详细日志

A：

githubjerry001, 2020-04-19:
到目前为止,只有youtube受影响,其他国外网站,如推特,facebook都不受影响
有试过把一个GFW模式下不能访问的的网站加到强制走代理的域名列表里,是可以生效的
不知道youtube有什么特殊的,在GFW和强制走代理的域名都不生效, 但在模式间切换后,又可以生效,而且在路由器重启前一直生效.

githubjerry001, 2020-04-19:
这个看起来是DNS问题,访问过一次,有缓存后就能解决.
求彻底解决此问题的方法.

Michaol, 2020-04-20:
这个问题我也遇到过，确实改为绕过大陆IP后，打开一下再切回GFW就好了。尝试了一下在br-lan里面手动加入DNS（使用自定义的 DNS 服务器），我把8888,8844之类的加了之后，能解决问题。你试试看，告诉我结果。我不确定是我的问题还是普遍问题。

deleisoft, 2020-04-20:
我也遇到了类似问题，使用GFW模式，没有被墙的域名也会走代理，配置强制不走代理的域名也没有生效，依然会走代理，不知道是否与DNS有关

jxph1123, 2020-04-20:
需要代理的端口 设置为所有端口试试

Michaol, 2020-04-20:
特地测试了一下午，自定义DNS实际上没效果，还是需要绕过大陆IP一段时间后，切换回GFW才能用。代理端口什么的都试过了，依然这样，很奇怪。

hxlls, 2020-04-21:
这个我以前遇到过，具体就是用了其他的DNS 加速之类的软件，或者是老的版本，openwrt 自己测试时候默认用的ISP的DNS ，不是本身的DNS地址，所以会造成测试不通，从而引起切换，新的版本，不用其他的DNS 软件没问题，如果非要使用，可以在lan 的DNS 地址处填上127.0.0.1作为DNS地址也就是通过PDNS 缓存去查询判断谷歌是否通畅，还有一个容易引起问题的原因是，带配置升级，这个就不需要说怎么解决了把

githubjerry001, 2020-04-27:
> 这个我以前遇到过，具体就是用了其他的DNS 加速之类的软件，或者是老的版本，openwrt 自己测试时候默认用的ISP的DNS ，不是本身的DNS地址，所以会造成测试不通，从而引起切换，新的版本，不用其他的DNS 软件没问题，如果非要使用，可以在lan 的DNS 地址处填上127.0.0.1作为DNS地址也就是通过PDNS 缓存去查询判断谷歌是否通畅，还有一个容易引起问题的原因是，带配置升级，这个就不需要说怎么解决了把

试过在lan DNS填上127.0.0.1,不能解决问题
尝试在Turbo ACC 网络加速--上游 DNS 服务器(国内)里填入本地的DNS,好像有点作用,重启后有时可以,有时还是不可以.
我确实如你所说,是带配置升级的,下次试试不带配置升级试试.

closed 2020-07-04 by coolsnowwolf

hnwywwh, 2022-03-22:
Ignore resolve file 取消此选项试试
DNS forwardings 也留空，因为一般interface里面已经指定DNS了

caileipk, 2022-04-03:
> 

你说的对，DNS forwarding 设为空就行了。但是一重启又会有了。

LouisLiu0707, 2022-10-05:
我的情况也是开启GFW可以打开youtube但加载不了视频，ipv6DNS关了也无济于事，后来我把op防火墙里的SYN-flood 防御关掉，然后就可以看视频了，方法记录下来，仅供参考

397414198, 2022-12-03:
解决了吗？

realabertian, 2023-03-21:
> Ignore resolve file 取消此选项试试 DNS forwardings 也留空，因为一般interface里面已经指定DNS了
> 
> 这个不能算是固件的问题，仔细梳理下DNS设置的解析路线

此方法可以可以解决问题
重点是 在 网络 - DHCP/DNS 中 服务器设置 选择 HOSTS 和解析文件, 将 忽略解析文件 取消勾选,并且解析文件 文本框 留空即可

linuxbelletian, 2023-08-12:
遇到相同问题，最新版的代码