coolsnowwolf/lede #4377: 无法通过Zerotier访问SMB

TXMAN, 2020-04-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：**软路由通过Zerotier组网后，电脑无法使用Zerotier内网IP访问软路由上开启的SMB服务。windows网络诊断显示：“该设备或资源(192.168.196.1-注：软路由在zerotier的内网IP)未设置为接受端口“文件和打印机共享(SMB)”上的连接。”**
推断此问题的出现原因可能是openwrt上的SMB服务未绑定zerotier IP地址。

### (2) 路由器型号和固件版本

A：OpenWrt R20.4.8

Aes64X, 2020-04-19:
这个不需要绑定吧。。。 Zerotier组往后内网是完全对接入设备开放的。我使用NFS共享，不需要其他操作，组网后就能使用KODI在手机上用移动网络远程播放视频、管理路由页面或者其他操作。

guochan2019, 2020-04-19:
检查自己设置，我用Zerotier远程SMB很久了，不存在你说的情况

lxwq, 2020-04-20:
把模板中的interfaces = |INTERFACES|注释掉试试看,

kltk, 2020-04-20:
> 把模板中的interfaces = |INTERFACES|注释掉试试看,

我是在后面加上 zerotier 的 interface，安全方面的考虑

TXMAN, 2020-04-25:
感谢以上二位，问题成功解决。在samba的模板中interfaces =部分加上zerotier的interface就能远程访问了

closed 2020-04-25 by TXMAN

John-Ming-yang, 2020-05-08:
求问应该加什么啊

yubanmeiqin9048, 2020-05-22:
ZeroTier需要通过配置使用吗？（看到网上说要加什么防火墙规则）