coolsnowwolf/lede #4384: uefi-gpt-squashfs固件每次更新都会丢失分区

winddriver, 2020-04-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：uefi-gpt-squashfs格式固件，每次更新都会丢失分区信息，导致所有docker相关数据丢失

XiaoliChan, 2020-04-24:
是的，我也是这样

winddriver, 2020-04-28:
请问，有什么办法能够保留分区更新固件吗？

ziyangdaoren, 2020-05-05:
一样的问题

qq1316237299, 2020-05-08:
uefi不能保留的把 我换回传统就可以保存的

jzhou404, 2020-06-11:
我也是， 看来是UEFI固件的问题，只要升级，分区就恢复到默认，自己的分区全部丢失

cokyrain, 2020-06-16:
不仅如此，调整分区大小后的新固件，还会丢失配置

 .config - OpenWrt Configuration
 > Target Images 
        *** Image Options ***                                                           │ │  
  │ │             (**16**) Kernel partition size (in MB)                                                  │ │  
  │ │             (**160**) Root filesystem partition size (in MB)

closed 2020-07-04 by coolsnowwolf