coolsnowwolf/lede #4389: 4m 小闪存  如何编译出更小的固件，只想要ssr plus

jiatongwu, 2020-04-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：4m小闪存的路由器 如何编译出更小的固件，只想包含 ssr plus?

### (2) 路由器型号和固件版本

A：wr841n v7

### (3) 详细日志

A：

mcucore, 2020-04-20:
CPU太渣，丢进垃圾桶，重新买个16M闪存CPU强劲的路由。非要4M建议官方源码编译。昨天也试图用LEAN源码编译QCA9558 4M带WPAD-FULL固件，剩骨头了编译成功，刷上不能用，懒得找原因，用回官方源码的4M。

martin0607, 2020-04-20:
L大都打算放弃8M以下设备了

Her0R0cket, 2020-04-20:
怎么可能.你是准备把系统精简了只留酸奶吗

Aes64X, 2020-04-20:
买个大些闪存的没什么难度吧？7620A的一大堆，几十块。

coolsnowwolf, 2020-04-20:
4M ROM 实际上只能使用 3.275MB 最大，一点都不能超过

内核+最基础的 rootfs + luci 就有 3.1MB 了，剩下 175KB，看你的造化了

daofeng2015, 2020-04-20:
现在随便一百多买个x86工控板加垃圾二手无线路由不香吗？

jiatongwu, 2020-04-20:
> 4M ROM 实际上只能使用 3.275MB 最大，一点都不能超过
> 
> 内核+最基础的 rootfs + luci 就有 3.1MB 了，剩下 175KB，看你的造化了

> 4M ROM 实际上只能使用 3.275MB 最大，一点都不能超过
> 
> 内核+最基础的 rootfs + luci 就有 3.1MB 了，剩下 175KB，看你的造化了

ssr plus 加上它的依赖包总共占多大?

mcucore, 2020-04-20:
> 4M ROM 实际上只能使用 3.275MB 最大，一点都不能超过
> 
> 内核+最基础的 rootfs + luci 就有 3.1MB 了，剩下 175KB，看你的造化了

官方19.07.2源码，编译QCA9558 3.5M使用中没发现问题，只换了个WPAD-FULL，去掉OPKG和LOGD。现在官方源码内核也比18.06大了些，以前4M还能多塞几个东西。破AP懒得换闪存，高通信号还是可以，没找到其他好的代用品。MTK开源驱动太渣渣，没法用。

yinzhefuyun, 2020-04-24:
> > 4M ROM 实际上只能使用 3.275MB 最大，一点都不能超过
> > 内核+最基础的 rootfs + luci 就有 3.1MB 了，剩下 175KB，看你的造化了
> 
> > 4M ROM 实际上只能使用 3.275MB 最大，一点都不能超过
> > 内核+最基础的 rootfs + luci 就有 3.1MB 了，剩下 175KB，看你的造化了
> 
> ssr plus 加上它的依赖包总共占多大?

一个v2主体就有3.4MB

badcrazy, 2020-04-25:
放弃吧，就算编出来也带不动

1049066154, 2020-04-26:
SSR plus 加上它的以来就超过4M了,上古产品还是别折腾了

lunatickochiya, 2020-04-29:
https://github.com/lunatickochiya/barrier_breaker/blob/barrier_breaker/README

有兴趣可以看看这里

closed 2020-07-04 by coolsnowwolf