coolsnowwolf/lede #4390: 树莓派3B WiFi无法启动

wangzhuok, 2020-04-20:
反馈bug/问题模板，提建议请删除

 1.关于你要提交的问题
树莓派3B 32位系统在启动时WiFi不能启动
Q：是否搜索了issue (使用 "x" 选择)
[x] 没有类似的issue

2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：树莓派3B 32bit 最新版

### (3) 详细日志

Sat Apr 18 21:06:39 2020 daemon.notice netifd: radio0 (754): Command failed: Request timed out
Sat Apr 18 21:06:39 2020 daemon.notice netifd: radio0 (754): Command failed: Not found
Sat Apr 18 21:06:39 2020 daemon.notice netifd: radio0 (754): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sat Apr 18 21:06:39 2020 daemon.notice netifd: radio0 (754): Command failed: Invalid argument
Sat Apr 18 21:06:40 2020 user.notice root: ip link set dev wlan0 up
Sat Apr 18 21:06:40 2020 kern.info kernel: [   45.052090] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sat Apr 18 21:06:51 2020 daemon.notice netifd: radio0 (1543): Command failed: Not found
Sat Apr 18 21:06:51 2020 daemon.notice netifd: radio0 (1543): command failed: No error information (-524)
Sat Apr 18 21:06:52 2020 daemon.notice netifd: radio0 (1616): command failed: No error information (-524)
Sat Apr 18 21:06:52 2020 daemon.notice netifd: radio0 (1616): command failed: I/O error (-5)
Sat Apr 18 21:06:52 2020 kern.warn kernel: [   56.512091] netlink: 4 bytes leftover after parsing attributes in process `iw'.
Sat Apr 18 21:06:52 2020 daemon.notice netifd: radio0 (1616): command failed: I/O error (-5)
Sat Apr 18 21:06:52 2020 daemon.notice netifd: radio0 (1616): command failed: Too many open files in system (-23)

AmadeusGhost, 2020-04-23:
可以试下我的pr：#4420

wangzhuok, 2020-04-23:
> 可以试下我的pr：#4420

大佬，改了之后需要make clean吗？

AmadeusGhost, 2020-04-23:
> 大佬，改了之后需要make clean吗？

直接`git pull`吧，已经merge了，问了一个有树莓派的他的3B无线正常了

wangzhuok, 2020-04-23:
> > 大佬，改了之后需要make clean吗？
> 
> 直接`git pull`吧，已经merge了，问了一个有树莓派的他的3B无线正常了

Thu Apr 23 16:56:36 2020 daemon.notice netifd: radio1 (9654): command failed: Not supported (-95)
Thu Apr 23 16:56:36 2020 daemon.notice netifd: radio1 (9654): Command failed: Invalid argument
Thu Apr 23 16:56:36 2020 daemon.notice netifd: radio1 (9654): command failed: No such device (-19)
Thu Apr 23 16:56:37 2020 daemon.notice netifd: radio1 (9698): WARNING: Variable 'data' does not exist or is not an array/object
大佬，client模式无法启动，以上是日志

AmadeusGhost, 2020-04-23:
> 大佬，master可以正常了，client模式无法启动，以上是日志

[image]
问了一个有树莓派3B的大佬是正常的

wangzhuok, 2020-04-24:
> > 大佬，master可以正常了，client模式无法启动，以上是日志
> 
> [image]
> 问了一个有树莓派3B的大佬是正常的

抱歉我没有说清楚，可以搜索无线，在保存应用之后不能启动，无论是自带的还是挂卡都不行

kk3375, 2020-04-24:
新3最新编译固件扫描网络加入后，无线未开启，点击停用在启动还是无效，前几天编译的无线还可以正常使用

AmadeusGhost, 2020-04-25:
> 大佬，master可以正常了，client模式无法启动，以上是日志

试试这个commit：https://github.com/AmadeusGhost/lede/commit/4a60fe4464e19b6ee29b03b1be96fa8571ccfe59
我猜可能这个地方问题（

wangzhuok, 2020-04-25:
> > 大佬，master可以正常了，client模式无法启动，以上是日志
> 
> 试试这个commit：[AmadeusGhost@4a60fe4](https://github.com/AmadeusGhost/lede/commit/4a60fe4464e19b6ee29b03b1be96fa8571ccfe59)
> 我猜可能这个地方问题（

我直接从openwrt拷贝一份netifd的源码过去试试

AmadeusGhost, 2020-04-25:
> 我直接从openwrt拷贝一份netifd的源码过去试试

注意加上最底下那个patch，而且我这个commit可能没用（小声）

wangzhuok, 2020-04-25:
> > 我直接从openwrt拷贝一份netifd的源码过去试试
> 
> 注意加上commit最底下那个patch，试了下好像没什么问题
> [image]

可以连接上了，DNS转发有点问题

AmadeusGhost, 2020-04-25:
> 可以连接上了，DNS转发有点问题，树莓派的DNS无法解析，与什么组件有关呢？

我那个commit最底下那个patch打了没有，官方改了路径，太蛋疼了就打patch改回来了

wangzhuok, 2020-04-25:
> > 可以连接上了，DNS转发有点问题，树莓派的DNS无法解析，与什么组件有关呢？
> 
> 我那个commit最底下那个patch打了没有，官方改了路径，太蛋疼了就打patch改回来了

打上了，手动设置了DNS也不行，诊断里面不能ping域名，可以ping IP

AmadeusGhost, 2020-04-26:
> 打上了，手动设置了DNS也不行，诊断里面不能ping域名，可以ping IP

试试这个commit：https://github.com/AmadeusGhost/lede/commit/500a975aa8a5c635dca01930ca72d836b00a5742

AmadeusGhost, 2020-05-01:
> 打上了，手动设置了DNS也不行，诊断里面不能ping域名，可以ping IP

试下这个commit：https://github.com/AmadeusGhost/lede/commit/ee87d9cccdcae9a9bafa5eeca4ebf5065c2f5545
[image]
今天有空试了下是可以正常中继的

kk3375, 2020-05-01:
> > 打上了，手动设置了DNS也不行，诊断里面不能ping域名，可以ping IP
> 
> 试下这个commit：[AmadeusGhost@ee87d9c](https://github.com/AmadeusGhost/lede/commit/ee87d9cccdcae9a9bafa5eeca4ebf5065c2f5545)
> [image]
> 今天有空试了下是可以正常中继的

新3今天测试已可以正常无线中继了

PR #4518 (2020-05-02): netifd: add dynamic wireless reconfiguration
Set new option 'reconf' in 'wifi-device' section to enable dynamic re-configuration on that radio. Also fix wifi relay and 'netifd: radio1 (9654): Command failed'.
Run-tested-by: @hyird 
fix #4319 , fix #4390 , fix #4474 , fix #4503 , fix #4516
closed 2020-05-02 by coolsnowwolf