coolsnowwolf/lede #4393: ssr plus 当选择绕过中国大陆ip模式时，就不能访问google了？

jiatongwu, 2020-04-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ssr plus 当选择绕过中国大陆ip模式时，就不能访问google了
     当选择gfw列表模式，就可以访问google，这个正常吗？

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

lylxoooooo, 2020-04-20:
没遇到过这个问题呢，可以尝试一下刷新一下电脑的dns缓存

shu0802, 2020-04-20:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ ]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：ssr plus 当选择绕过中国大陆ip模式时，就不能访问google了
> 当选择gfw列表模式，就可以访问google，这个正常吗？
> 
> ### (2) 路由器型号和固件版本
> A：x86_64
> 
> ### (3) 详细日志
> A：

可能与你的节点有关系，原来我也遇到过这样的情况

jiatongwu, 2020-04-20:
我的节点是香港服务器

hxlls, 2020-04-20:
自己把节点从国内IP段取消，或者想国内IP段的上游提交对应信息

plutoghost, 2020-05-06:
改一下iptables规则的顺序就可以

slitilanber, 2020-05-12:
一直用绕过模式，从未出过上不了G网的问题

closed 2020-07-04 by coolsnowwolf