coolsnowwolf/lede #4394: pull到最新版本后菜单中ssr plus没有了

tufu9441, 2020-04-20:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：有段时间没编译了，更新到最新版本代码后，Luci-Applications中找不到luci-app-ssr-plus的选项了，回退到 b97c4e8bb5707c90ba83885e115ba362c418d9b8 后才找回来，是现在食用的姿势有变化了吗？

### (2) 路由器型号和固件版本

A：Newifi D2

### (3) 详细日志

[image]

EnnawYang, 2020-04-20:
请仔细翻阅前人的issue

guochan2019, 2020-04-20:
https://github.com/coolsnowwolf/lede/commit/af803f2569bbbaac88d001890a8a7646abf0ce0d#comments

tufu9441, 2020-04-20:
> [af803f2#comments](https://github.com/coolsnowwolf/lede/commit/af803f2569bbbaac88d001890a8a7646abf0ce0d#comments)

明白了，感谢两位！

closed 2020-04-20 by tufu9441