coolsnowwolf/lede #4399: PPTP 服务器连接问题

Blueskyhou, 2020-04-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
  
## 2. 详细叙述

### (1) 具体问题

A：看到commit ，最新去掉了PPTP服务器，所以尝试用LT2P/IPSEC, 但WINDOWS 10 发现怎么也连不上 IPSEC， 百度了一遍也没解决，回到 挂在路由后面的NAS 上面建立 PPTP服务器，使用端口转发，发现也连接不上（如果挂在K2P梅林后面可以）。最后又回到了X86 openwrt. 基于lean的源码，编译进了一个第三方PPTP server源. 但最后还没能解决。还有一个比较奇怪，DDNS内网不能访问，但是windwos, PPTP却可以连接上, 换到外网又连接不上，日志如下。清帮忙看下有无解决的方法。谢谢！

### (2) 路由器型号和固件版本

A： X86 ， OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)

### (3) 详细日志

A：Mon Apr 20 22:23:53 2020 daemon.info pptpd[23699]: CTRL: Client 117.136.40.215 control connection started
Mon Apr 20 22:23:53 2020 daemon.info pptpd[23699]: CTRL: Starting call (launching pppd, opening GRE)
Mon Apr 20 22:23:53 2020 daemon.notice pppd[23700]: pppd 2.4.8 started by root, uid 0
Mon Apr 20 22:23:53 2020 daemon.info pppd[23700]: Using interface ppp2
Mon Apr 20 22:23:53 2020 daemon.notice pppd[23700]: Connect: ppp2 <--> /dev/pts/1
Mon Apr 20 22:24:24 2020 daemon.warn pppd[23700]: LCP: timeout sending Config-Requests
Mon Apr 20 22:24:24 2020 daemon.notice pppd[23700]: Connection terminated.
Mon Apr 20 22:24:24 2020 daemon.info : 05[KNL] interface ppp2 deleted
Mon Apr 20 22:24:24 2020 daemon.notice pppd[23700]: Modem hangup
Mon Apr 20 22:24:24 2020 daemon.info pppd[23700]: Exit.

ElectronicElephant, 2020-04-23:
看起来是nat loopback的问题，按道理来说最近的几个commit应该是已经解决了的

georip, 2020-06-04:
A：PPTP SERVER多用户状态上能连接但不能上网以及访问路由器。
在pptp连接模式下
单用户连接没有问题，且能正常上网和访问路由器管理。但是多用户添加以后，pptp能正常连接
但是，不能上网以及访问管理路由了。
通过对比前后配置文件，发现是配置文件中的“mppe required,no40,no56,stateless ”被删除造成。补充后即可正常，但是如果再在WEB下配置，又会丢失“mppe required,no40,no56,stateless ”造成故障
#debug
#logfile /tmp/pptp-server.log
auth
name "pptp-server"
lcp-echo-failure 3
lcp-echo-interval 60
default-asyncmap
mtu 1482
mru 1482
nobsdcomp
nodeflate
#noproxyarp
#nomppc
require-mschap-v2
refuse-chap
refuse-mschap
refuse-eap
refuse-pap
#plugin radius.so
#radius-config-file /etc/radius.conf
mppe required,no40,no56,stateless
ms-dns 192.168.2.1
(2) 路由器型号和固件版本

A：R619AC
使用lean 3月2日源码编译固件。
(3) 详细日志

A：

closed 2020-07-04 by coolsnowwolf