coolsnowwolf/lede #4401: 弱弱的问一句，ssr-plus是不再更新了吗

ch1993128, 2020-04-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

Aes64X, 2020-04-21:
想想自己家在哪，不要给别人找麻烦，也不要给自己找麻烦。

EnnawYang, 2020-04-21:
> 想想自己家在哪，不要给别人找麻烦，也不要给自己找麻烦。

哈哈哈哈 戳中要害

ch1993128, 2020-04-21:
你们这个问题很有意思啊，我的家在地球上面啊，同时我也是中国人，外面的世界也很精彩，这个问题是不该提，不说了我去拿快递去了，这顺丰怎么还砸门，真没素质

closed 2020-04-21 by ch1993128

ch1993128, 2020-04-21:
有意思啊，我难道不是生活在地球上吗？有些话题是不该提起，我也怕查顺风快递啊，所以关了吧，我匿了 ------------------ 原始邮件 ------------------

  

  
&#x54C8;&#x54C8;&#x54C8;&#x54C8; 戳中要害
 
&mdash;

AwesomeSky, 2020-04-25:
> 你们这个问题很有意思。我的家在地球上面啊，同时我也是中国人。外面的世界很精彩呢。这个问题是不该提，所以关了他吧。不说了我去拿快递去了，这顺丰怎么还砸门，真没素质
> 
> github这种万恶之源我们要坚决抵制！！！

https://github.com/fw876/helloworld
有时候多想想为什么 自己总是伸手 获得信息的渠道多种多样 
没必要非要感觉别人有为你解决一切的义务
连接在上面了 希望你能会编译 不会的话 建议查查教程 不要提类似XXX怎么编译的问题了
百度一下很多教程的

ch1993128, 2020-04-25:
我可不需要hello world，也不是伸手党，更没有在提过什么编译报错的问题，另外trojan和别的我也会自己整合到op里面去，提这个问题的意思无非是询问一下。 对不起我道歉，给水军喷子们发挥的机会了 ------------------ 原始邮件 ------------------

  

 
  
https://github.com/fw876/helloworld
 有时候多想想为什么 自己总是伸手 获得信息的&#x6E20;道多种多样
 
&mdash;

AwesomeSky, 2020-04-25:
那你既然这么说了 何必发issue呢 commits翻一翻不就知道了 
就像上面两位的态度一样 就不该和你好好说话

ch1993128, 2020-04-25:
> 
> 那你既然这么说了 何必发issue呢 commits翻一翻不就知道了
> 就像上面两位的态度一样 就不该和你好好说话

看见你们的评论我就知道不合时宜，不是关了么，确实这种事情不应该让别人去背责任，没毛病的

Aes64X, 2020-04-26:
这issues没必要在讨论了吧？ 我一直都认为任何时候都不应在公开场合讨论这些，毕竟在内来说定义模糊，请喝茶的理由也是随便编个就能让你喝而并不需要合情合理又合法。
牢记，劝人爬梯，天打雷劈。这东西知道就知道了，不知道也别当个好奇宝宝。别人不知道也别凑上去。

ch1993128, 2020-04-26:
> 这issues没必要在讨论了吧？ 我一直都认为任何时候都不应在公开场合讨论这些，毕竟在内来说定义模糊，请喝茶的理由也是随便编个就能让你喝而并不需要合情合理又合法。
> 牢记，劝人爬梯，天打雷劈。这东西知道就知道了，不知道也别当个好奇宝宝。别人不知道也别凑上去。

大声告诉我closed怎么念？