coolsnowwolf/lede #4403: 硬件加速貌似未生效

wpm369079365, 2020-04-21:
歌华链  7621
未开硬件加速200m宽带  占用25 %左右  ，开了之后还是25%。貌似硬件加速并未开启

bluehj777, 2020-04-21:
MT6521的一键加速上个月有一个内核补丁，打上后才可以开启HWNAT，否则现在都是用软加速，遗憾的是这个补丁打上后会造成路由不定时重启，所以后来就给去掉了
https://github.com/coolsnowwolf/lede/issues/3880

wpm369079365, 2020-04-21:
> MT6521的一键加速上个月有一个内核补丁，打上后才可以开启HWNAT，否则现在都是用软加速，遗憾的是这个补丁打上后会造成路由不定时重启，所以后来就给去掉了
> #3880

希望大神们能早点修复这个BUG吧，HWNAT效果不是一般的好

coolsnowwolf, 2020-04-21:
HWNAT 会造成内核崩

wpm369079365, 2020-04-21:
> HWNAT 会造成内核崩

缺少硬件加速确实可惜了~~·感谢lean

wpm369079365, 2020-04-21:
> HWNAT 会造成内核崩

https://github.com/openwrt/openwrt/blob/master/target/linux/generic/pending-4.14/641-netfilter-nf_flow_table-support-hw-offload-through-v.patch     此patch貌似能解决问题（来自一个另一个大神的回复）.

bluehj777, 2020-04-21:
我也一直关注这个问题，最初讨论是这个

https://github.com/openwrt/openwrt/pull/2815
目前这个有时间可以试试，这里有讨论
https://github.com/KevinMX/openwrt-phicomm-k2p-build/issues/10

bluehj777, 2020-04-21:
> > HWNAT 会造成内核崩
> 
> https://github.com/openwrt/openwrt/blob/master/target/linux/generic/pending-4.14/641-netfilter-nf_flow_table-support-hw-offload-through-v.patch 此patch貌似能解决问题（来自一个另一个大神的回复）.

我也一直关注这个问题，最初讨论是这个

https://github.com/openwrt/openwrt/pull/2815
目前这个有时间可以试试，这里有讨论
https://github.com/KevinMX/openwrt-phicomm-k2p-build/issues/10

bluehj777, 2020-04-23:
今天早上刷上了替换过https://github.com/openwrt/openwrt/blob/master/target/linux/generic/pending-4.14/641-netfilter-nf_flow_table-support-hw-offload-through-v.patch 的固件，路由是小娱C5 ，目前看，100M宽带，普通多线程满速下载luci界面里的CPU占用基本小娱10%,迅雷下载基本小娱20%，硬件加速有效果，但是依然出现过路由重启现象一次。

wpm369079365, 2020-04-23:
> 今天早上刷上了替换过https://github.com/openwrt/openwrt/blob/master/target/linux/generic/pending-4.14/641-netfilter-nf_flow_table-support-hw-offload-through-v.patch 的固件，路由是小娱C5 ，目前看，100M宽带，普通多线程满速下载luci界面里的CPU占用基本小娱10%,迅雷下载基本小娱20%，硬件加速有效果，但是依然出现过路由重启现象一次。

话说开了硬件加速 100m 应该小于 10%吧，我没有编译能力，有的话也编译一个版本试一下了

TXH201622, 2020-04-23:
htop里看cpu吧，tcp的话，硬件加速生效应该完全不占cpu。

closed 2020-04-26 by wpm369079365

KevinMX, 2020-05-09:
目前HWNAT(K2P)经测试似乎仅对有线网络(PPPoE<-->LAN, LAN<-->LAN)生效，WiFi下并未生效，CPU占用较高。

MeIsReallyBa, 2020-05-09:
> 目前HWNAT(K2P)经测试似乎仅对有线网络(PPPoE<-->LAN, LAN<-->LAN)生效，WiFi下并未生效，CPU占用较高。

我看了MTK SDK的代码，WIFI的hwnat ，无线驱动和HWNAT本身都需要写代码做相应的处理，很明显OP现在是缺少这一部分代码的。

bluehj777, 2020-05-09:
> > 目前HWNAT(K2P)经测试似乎仅对有线网络(PPPoE<-->LAN, LAN<-->LAN)生效，WiFi下并未生效，CPU占用较高。
> 
> 我看了MTK SDK的代码，WIFI的hwnat ，无线驱动和HWNAT本身都需要写代码做相应的处理，很明显OP现在是缺少这一部分代码的。

我这里测试环境是纯有线路由，用的是小娱C3？wifi是用的Ap接入。目前看即使是有线hwnat起作用，但是会造成路由不定时死机，没有规律。

MeIsReallyBa, 2020-05-09:
> > > 
> > 
> > 
> > 我看了MTK SDK的代码，WIFI的hwnat ，无线驱动和HWNAT本身都需要写代码做相应的处理，很明显OP现在是缺少这一部分代码的。
> 
> 我这里测试环境是纯有线路由，用的是小娱C3？
op的hwnat不是mtk官方写的，不稳定也正常。看5.4内核重写hwnat会不会好点吧。现在要用hwnat还是别用op了。

bluehj777, 2020-05-09:
> > > > 
> > > 
> > > 
> > > 我看了MTK SDK的代码，WIFI的hwnat ，无线驱动和HWNAT本身都需要写代码做相应的处理，很明显OP现在是缺少这一部分代码的。
> > 
> > 
> > 我这里测试环境是纯有线路由，用的是小娱C3？
> 
> op的hwnat不是mtk官方写的，不稳定也正常。看5.4内核重写hwnat会不会好点吧。现在要用hwnat还是别用op了。

请问目前还有哪个支持hwnat的开源的系统靠谱点儿？padavan？

MeIsReallyBa, 2020-05-09:
> > > > > 
> > > > 
> > > > 
> > > > 我看了MTK SDK的代码，WIFI的hwnat ，无线驱动和HWNAT本身都需要写代码做相应的处理，很明显OP现在是缺少这一部分代码的。
> > > 
> > > 
> > > 我这里测试环境是纯有线路由，用的是小娱C3？
> > 
> > 
> > op的hwnat不是mtk官方写的，不稳定也正常。看5.4内核重写hwnat会不会好点吧。现在要用hwnat还是别用op了。
> 
> 请问目前还有哪个支持hwnat的开源的系统靠谱点儿？padavan？

mtk的第三方固件就OP PADAVAN还有潘多拉。另外两个的hwnat都是用的mtk官方sdk的代码

xjboss, 2020-05-24:
好像最新版有HWNAT了，我打开HWNAT路由器几分钟就炸了，这个HWNAT还是不太稳啊。