coolsnowwolf/lede #4409: SSR Plus + 设置了强制不走代理的域名无效，实际还是走了代理

bitzhi, 2020-04-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

SSR Plus + 设置了强制不走代理的域名无效，实际还是走了代理。
[image]
[image]

### (2) 路由器型号和固件版本

主机型号 | UBNT-ERX (CpuMark : 4597.084682 Scores)
架构 | MediaTek MT7621 ver:1 eco:3 x 4
固件版本 | OpenWrt R20.3.19 / LuCI Master (git-20.088.29068-cebe120)
内核版本 | 4.14.172

### (3) 详细日志

2020-04-22 01:42:06 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-04-22 01:31:26 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-04-22 01:30:48 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-04-22 01:23:30 ShadowsocksR No Problem.
2020-04-22 01:12:23 ShadowsocksR No Problem.
2020-04-22 01:01:16 ShadowsocksR No Problem.

hcym, 2020-04-21:
应该有效吧，一直有几个域名直通。

springsunx, 2020-04-26:
我的是设置模式为绕过中国大陆IP模式的话，和全局一样，所有都走代理。
现在只能改成gfw。不知道什么情况。

xinxijishuwyq, 2020-05-01:
> 我的是设置模式为绕过中国大陆IP模式的话，和全局一样，所有都走代理。
> 现在只能改成gfw。不知道什么情况。

我今天也碰到了。。。

hcym, 2020-05-01:
我的一直正常，用那个clash神器反而不稳定，论坛反复提醒异常登录，烦不胜烦。大雕的这个黑白名单从来就没事。也是有遇到一个怪事，haproxy负载均衡在n1上只能gfwlist，在x86上面只能路由表，反之就是不行，活见鬼了

xinxijishuwyq, 2020-05-01:
> > 我的是设置模式为绕过中国大陆IP模式的话，和全局一样，所有都走代理。
> > 现在只能改成gfw。不知道什么情况。
> 
> 我今天也碰到了。。。

我今天碰到的问题是由该pr引起的。
https://github.com/fw876/helloworld/pull/1
已通过手动修改/etc/config/shadowsocksr文件解决解决

coaens, 2020-05-31:
这个具体怎么修改呢？

closed 2020-07-04 by coolsnowwolf

JustinhoCHN, 2020-08-12:
> etc/config/shadowsocksr

请问是如何修改？我也遇到这个问题了

hekang26, 2020-12-10:
> > > 我的是设置模式为绕过中国大陆IP模式的话，和一体一样，所有都走代理。
> > > 现在只能改成gfw。不知道什么情况。
> > 
> > 
> > 我今天也碰到了。。。
> 
> 我今天碰到的问题是由该PR引起的。
> [fw876 / HelloWorld的＃1](https://github.com/fw876/helloworld/pull/1)
> 已通过手动修改的/ etc /配置/ shadowsocksr文件解决解决

我也遇到这个问题了  请问如何修改?

usamr, 2021-01-28:
我也是这样的问题K2P和软路由用的openwrt，无论是gfw列表还是大陆以外，访问本地网站都是代理的地址，指定本地不走代理都不好用，实在找不到方法了。

acse-kx20, 2021-10-23:
我近期也遇到了同样的问题，即“不走代理的域名”和“强制走代理的域名”两个名单无效，我试了各种方法，最后发现我这里是DNS的问题，解决方法：

1. 关闭OpenWRT-Turbo ACC网络加速-DNS缓存。
2. SSR PLUS-状态-重置PDNSD缓存（SSR PLUS-客户端-DNS解析方式默认为“使用PDNSD TCP查询并缓存”）。
3. 如果电脑使用的是Chrome浏览器，在设置中将“使用安全DNS”选项关闭（基于Chromium内核的浏览器应该都有这个选项）。
4. 清空电脑浏览器、操作系统内的DNS缓存（两个都要清理，方法Google）。
5. 将电脑的首选DNS设为软路由而不是主路由（我使用的是旁路由模式）。

仅供参考，希望有用：）

oceange8, 2021-11-15:
我也碰到了这个问题，我是发现同花顺炒股软件手机端总是感觉延迟，服务器设置为电信，结果总是变为其他。网络诊断是走了代理。同花顺域名为10jqka.com.cn和hexin.cn
我目前设置为绕过中国大陆ip。使用PDNSD解析。启用BBR加速，启用DNS缓存。在上游DNS服务器添加了本地DNS。
我只是使用小白，对一些技术术语不甚明白，请问该怎么设置。谢谢

RockNHawk, 2021-12-01:
> 我近期也遇到了同样的问题，即“不走代理的域名”和“强制走代理的域名”两个名单无效，我试了各种方法，最后发现我这里是DNS的问题，解决方法：
>
> 1. 关闭OpenWRT-Turbo ACC网络加速-DNS缓存。
> 2. SSR PLUS-状态-重置PDNSD缓存（SSR PLUS-客户端-DNS解析方式默认为“使用PDNSD TCP查询并缓存”）。
> 3. [...]

有用，感想！

AndyFCui, 2022-08-22:
可不可以设置只代理单个域名啊，我在海外，如果选择回国模式那么微信什么的，抖音都会变得很慢，可不可以只设置几个域名代理，其他的都不代理。。。。。