coolsnowwolf/lede #4410: 我试着编译了一下OPENWRT，但是在LuCI里面没有ssr-plus

hlz781105, 2020-04-22:
我试着编译了一下OPENWRT，但是在LuCI里面没有ssr-plus。
是我的哪一步搞错了还是有什么隐藏的没有找到。
先谢谢各位大神了。

martin0607, 2020-04-22:
没有搞错，就是没有的

Michaol, 2020-04-22:
首先提问之前要习惯先去查找旧issue和TG，google也行。这是一个好习惯，既不麻烦别人，也能加深认识。这个问题已经说了好多天了，自己先翻查一下吧。helloworld~~~

hlz781105, 2020-04-22:
搜索了一下好像解决了，现在编译中
feeds.conf.default里helloworld的注释去掉
./scripts/feeds update -a
./scripts/feeds install -a
又执行了一次
make menuconfig
然后就有了ssr plus选项了。
现在编译中，几个小时后就知道行不行了

hcym, 2020-04-22:
生在强国要自强不息噢

ztfot, 2020-04-22:
看见了，现在编译中，不知道能不能成功

Michaol, 2020-04-22:
这就对头了

ztfot, 2020-04-22:
好像编译错误

ztfot, 2020-04-22:
Makefile:327: recipe for target '/home/ztfot/lede/build_dir/hostpkg/go-1.14/.built' failed
make[3]: *** [/home/ztfot/lede/build_dir/hostpkg/go-1.14/.built] Error 2
make[3]: Leaving directory '/home/ztfot/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#68.45#9.61#79.43
package/Makefile:111: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/ztfot/lede'
package/Makefile:107: recipe for target '/home/ztfot/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ztfot/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ztfot/lede'
/home/ztfot/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

ztfot, 2020-04-22:
> 搜索了一下好像解决了，现在编译中
> feeds.conf.default里helloworld的注释去掉
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> 又执行了一次
> make menuconfig
> 然后就有了ssr plus选项了。
> 现在编译中，几个小时后就知道行不行了

你的编译成功了吗？

dodogao, 2020-04-22:
执行命令 make package/feeds/luci/luci-base/compile V=99

ztfot, 2020-04-22:
> 执行命令 make package/feeds/luci/luci-base/compile V=99

从哪步开始执行？

dodogao, 2020-04-22:
> > 执行命令 make package/feeds/luci/luci-base/compile V=99
> 
> 从哪步开始执行？

报错之后/开始编译之前，都可以

xiehuayi75, 2020-04-22:
1. 在lede目录，修改feeds.conf.default文件 
2. 第五行的  #src-git helloworld  把最前面的#去掉
3. 执行 ./scripts/feeds update -a && ./scripts/feeds install -a

ztfot, 2020-04-22:
> > > 执行命令 make package/feeds/luci/luci-base/compile V=99
> > 
> > 
> > 从哪步开始执行？
> 
> 报错之后/开始编译之前，都可以

还是会报错
Makefile:618: recipe for target '/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc.installed' failed
make[2]: *** [/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/ztfot/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.12#0.04#0.17
package/Makefile:111: recipe for target 'package/libs/toolchain/compile' failed
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/ztfot/lede'
/home/ztfot/lede/include/toplevel.mk:225: recipe for target 'package/feeds/luci/luci-base/compile' failed
make: *** [package/feeds/luci/luci-base/compile] Error 2

ztfot, 2020-04-22:
> * 在lede目录，修改feeds.conf.default文件
> * 第五行的  #src-git helloworld  把最前面的#去掉
> * 执行 ./scripts/feeds update -a && ./scripts/feeds install -a

然后我执行了
make menuconfig
make -j8 download V=s
make -j1 V=s
最后还是产生大段报错

Michaol, 2020-04-22:
确定下载齐全了么？我刚中午增增减减了几个工具，又编译一次，没有问题。

ztfot, 2020-04-22:
> 确定下载齐全了么？我刚中午增增减减了几个工具，又编译一次，没有问题。

我用的是国外高频vps编译，下载齐全，就是加了一个ssr plus就编译失败

ztfot, 2020-04-22:
> 确定下载齐全了么？我刚中午增增减减了几个工具，又编译一次，没有问题。

执行 ./scripts/feeds install -a会在开头产生大量warning

Michaol, 2020-04-22:
我是本地win10 ubuntu 18.04编译，下面是我之前的步骤。去掉#我是直接用文本编辑，懒得打命令。

cd lede
1.git pull
2.去掉#号
3. rm ./tmp -fr
4. ./scripts/feeds update -a && ./scripts/feeds install -a
5. make menuconfig
6. make package/feeds/luci/luci-base/compile V=99
7. make -j$(($(nproc) + 1)) V=s

ztfot, 2020-04-22:
> cd lede

你这个是第二次编译吧
我第一次编译最后一步
make -j8 download V=s
make -j1 V=s
换成这个

Michaol, 2020-04-22:
我这个星期已经编译好几次了，都没问题。试试直接make download，不要-j8。防意外。

ztfot, 2020-04-22:
> 我这个星期已经编译好几次了，都没问题。试试直接make download，不要-j8。防意外。
那我第一次编译
git clone https://github.com/coolsnowwolf/lede
cd lede
git pull
去掉#号
 ./scripts/feeds update -a && ./scripts/feeds install -a
make menuconfig
make download
make package/feeds/luci/luci-base/compile V=99
make -j1 V=s
这样不知道会不会报错

Michaol, 2020-04-22:
按照之前P3大佬的教程，检查文件完整性

find dl -size -1024c -exec ls -l {} \;

此命令可以列出下载不完整的文件（根据我多次编译的经验得出小于1k的文件属于下载不完整），如果存在这样的文件可以使用find dl -size -1024c -exec rm -f {} \;命令将它们删除，然后重新执行make download下载并反复检查，确认所有文件完整可大大提高编译成功率，避免浪费时间。

ztfot, 2020-04-22:
> 按照之前P3大佬的教程，检查文件完整性
> 
> find dl -size -1024c -exec ls -l {} ;
> 
> 此命令可以列出下载不完整的文件（根据我多次编译的经验得出小于1k的文件属于下载不完整），如果存在这样的文件可以使用find dl -size -1024c -exec rm -f {} ;命令将它们删除，然后重新执行make download下载并反复检查，确认所有文件完整可大大提高编译成功率，避免浪费时间。

确保下载完整后执行make package/feeds/luci/luci-base/compile V=99 有大段错误
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
Collecting package info: done
Collecting target info: done
time: target/linux/prereq#0.11#0.01#0.13
make[1]: Entering directory '/home/ztfot/lede'
make[2]: Entering directory '/home/ztfot/lede/feeds/luci/modules/luci-base'
. /home/ztfot/lede/include/shell.sh; bzcat /home/ztfot/lede/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.prepared66aff93c9579bf19d55999023fcd607e_6664517399ebbbc92a37c5bb081b5c53
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.configured
make -C src/ clean po2lmo
make[3]: Entering directory '/home/ztfot/lede/feeds/luci/modules/luci-base/src'
rm -f po2lmo parser.so version.lua *.o
cc    -c -o po2lmo.o po2lmo.c
cc    -c -o template_lmo.o template_lmo.c
cc  -o po2lmo po2lmo.o template_lmo.o
make[3]: Leaving directory '/home/ztfot/lede/feeds/luci/modules/luci-base/src'
make -C /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1 bin/LuaSrcDiet.lua
make[3]: Entering directory '/home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
make[3]: Nothing to be done for 'bin/LuaSrcDiet.lua'.
make[3]: Leaving directory '/home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
install -d -m0755 /home/ztfot/lede/staging_dir/hostpkg/bin
install -m0755 src/po2lmo /home/ztfot/lede/staging_dir/hostpkg/bin/po2lmo
install -m0755 /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /home/ztfot/lede/staging_dir/hostpkg/bin/LuaSrcDiet
mkdir -p /home/ztfot/lede/staging_dir/hostpkg/stamp
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
touch /home/ztfot/lede/staging_dir/hostpkg/stamp/.luci-base_installed
make[2]: Leaving directory '/home/ztfot/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.35#0.50#1.90
make[2]: Entering directory '/home/ztfot/lede/package/libs/toolchain'
touch /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.prepared_34ac9a1ef8dd17ffcf2bc1b765bcc888_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain
touch /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.prepared_34ac9a1ef8dd17ffcf2bc1b765bcc888_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.configured_*
rm -f /home/ztfot/lede/staging_dir/target-x86_64_musl/stamp/.toolchain_installed
(cd /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/./; if [ -x ./configure ]; then find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r chmod u+w; find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/ztfot/lede/scripts/config.guess; find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r chmod u+w; find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /home/ztfot/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY [... cut ...]
make[2]: *** [/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/ztfot/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.12#0.04#0.17
package/Makefile:111: recipe for target 'package/libs/toolchain/compile' failed
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/ztfot/lede'
/home/ztfot/lede/include/toplevel.mk:225: recipe for target 'package/feeds/luci/luci-base/compile' failed
make: *** [package/feeds/luci/luci-base/compile] Error 2

Michaol, 2020-04-22:
这我就不清楚了，我不是大神，也是在一边学一边练，期待大神解答。

aaronchin, 2020-04-22:
make clean 然后重新编译

hlz781105, 2020-04-22:
把我的过程写下来，给和我一样的小白少踩一点坑。

1、sudo apt-get update
 
2、 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf

3、 git clone https://github.com/coolsnowwolf/lede 

4、 cd lede 进入目录

5、XFTP登陆然后修改feeds.conf.default里helloworld的注释去掉#
（xshell 6 xftp 6都可以从官网下载免费的家庭学校版使用）

6、./scripts/feeds update -a

7、./scripts/feeds install -a

8、make menuconfig

9、make -j8 download V=s

然后是等。。。。。。我的现在编译中。。。。不知道什么时候好。弄好了再给各位大神汇报。

ztfot, 2020-04-22:
> 把我的过程写下来，给和我一样的小白少踩一点坑。
>
> 1、sudo apt-get update
>
> 2、 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev [...]

你用什么编译啊，怎么这么慢，我3个小时就编译好了，我用的是vultr的高频vps，美国

ztfot, 2020-04-22:
> 把我的过程写下来，给和我一样的小白少踩一点坑。
>
> 1、sudo apt-get update
>
> 2、 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev [...]

我编译失败了好几次，用你这个，不知道为什么

ztfot, 2020-04-22:
大家看看编译到最后这是什么错误
( cd /home/ztfot/lede/build_dir/hostpkg/go-1.14/src ; GOROOT_FINAL=/home/ztfot/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/ztfot/lede/build_dir/hostpkg/go-1.14/.go_bootstrap GOCACHE=/home/ztfot/lede/build_dir/hostpkg/go-1.14/.go_work/cache CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/ztfot/lede/build_dir/hostpkg/go-1.14/.go_bootstrap. (go1.4-bootstrap-20170531 linux/amd64)
Building Go toolchain1 using /home/ztfot/lede/build_dir/hostpkg/go-1.14/.go_bootstrap.
# bootstrap/cmd/compile/internal/ssa
/home/ztfot/lede/build_dir/hostpkg/go-1.14/src/cmd/compile/internal/ssa/rewritedecArgs.go:226[/home/ztfot/lede/build_dir/hostpkg/go-1.14/pkg/bootstrap/src/bootstrap/cmd/compile/internal/ssa/rewritedecArgs.go:230]: internal compiler error: out of memory
go tool dist: FAILED: /home/ztfot/lede/build_dir/hostpkg/go-1.14/.go_bootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 2
Makefile:327: recipe for target '/home/ztfot/lede/build_dir/hostpkg/go-1.14/.built' failed
make[3]: *** [/home/ztfot/lede/build_dir/hostpkg/go-1.14/.built] Error 2
make[3]: Leaving directory '/home/ztfot/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#62.37#7.60#71.11
package/Makefile:111: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/ztfot/lede'
package/Makefile:107: recipe for target '/home/ztfot/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ztfot/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ztfot/lede'
/home/ztfot/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

hue715, 2020-04-22:
> 确保下载完整后执行make package/feeds/luci/luci-base/compile V=99 有大段错误
> Checking 'working-make'... ok.
> Checking 'case-sensitive-fs'... ok.
> Checking 'proper-umask'... ok.
> Checking 'gcc'... ok.
> Checking [...]

看到文件缺失，之前完整编译过吗？toolchain编译完成了吗？

ztfot, 2020-04-22:
> 看到文件缺失，之前完整编译过吗？toolchain编译完成了吗？

老哥你看我上面最新的错误，我第一次编译，如果不去#可以完美编译

hlz781105, 2020-04-22:
> > 把我的过程写下来，给和我一样的小白少踩一点坑。
> > 1，sudo apt-get更新
> > 2，sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-完整的msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g ++-multilib antlr3 gperf
> > 3，git clone https://github.com/coolsnowwolf/lede
> > 4，cd lede进入目录
> > 5，XFTP登陆然后修改feeds.conf.default里helloworld的注释去掉＃
> > （xshell 6 xftp 6都可以从官网下载免费的家庭学校版使用）
> > 6，。/ scripts / feeds更新-a
> > 7，。/ scripts / feeds安装-a
> > 8，制作菜单配置
> > 9，make -j8下载V = s
> > 然后是等。。。。。。。我的现在编译中。。。。。不知道什么时候好。弄好了再给各位大神汇报。
> 
> 你用什么编译啊，怎么这么慢，我3个小时就编译好了，我用的是vultr的高频vps，美国
虚拟机。配制有点差啊。

hlz781105, 2020-04-22:
我的也是Error 2。。。。。。。。。
继续折腾

ztfot, 2020-04-22:
> 我的也是Error 2。。。。。。。。。
> 继续折腾

> 我的也是Error 2。。。。。。。。。
> 继续折腾

我编译了3次都是不行

hlz781105, 2020-04-22:
> > 我的也是Error 2。。。。。。。。。
> > 继续折腾
> 
> > 我的也是Error 2。。。。。。。。。
> > 继续折腾
> 
> 我编译了3次都是不行

不去掉#号时可以。现在不行了。。。

ztfot, 2020-04-22:
> > > 我的也是Error 2。。。。。。。。。
> > > 继续折腾
> > 
> > 
> > > 我的也是Error 2。。。。。。。。。
> > > 继续折腾
> > 
> > 
> > 我编译了3次都是不行
> 
> 不去掉#号时可以。现在不行了。。。

我也是这样的

hlz781105, 2020-04-22:
找了一台VPS来编译。
可是开始最后一步时才想起来，VPS的用户默认就是ROOT的。。。
ROOT用户编译会出现什么问题？

ztfot, 2020-04-22:
> 找了一台VPS来编译。
> 可是开始最后一步时才想起来，VPS的用户默认就是ROOT的。。。
> ROOT用户编译会出现什么问题？

反正我用root没成功过，添加个用户就好了

hlz781105, 2020-04-22:
这个是我自己的虚拟机编译的报错

bash: po2lmo: command not found
Makefile:120: recipe for target '/home/h/lede/build_dir/target-aarch64_cortex-a72_musl/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus.installed' failed
make[3]: *** [/home/h/lede/build_dir/target-aarch64_cortex-a72_musl/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus.installed] Error 127
make[3]: Leaving directory '/home/h/lede/feeds/helloworld/luci-app-ssr-plus'
time: package/feeds/helloworld/luci-app-ssr-plus/compile#0.17#0.15#0.32
package/Makefile:111: recipe for target 'package/feeds/helloworld/luci-app-ssr-plus/compile' failed
make[2]: *** [package/feeds/helloworld/luci-app-ssr-plus/compile] Error 2
make[2]: Leaving directory '/home/h/lede'
package/Makefile:107: recipe for target '/home/h/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile' failed
make[1]: *** [/home/h/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/h/lede'
/home/h/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

hlz781105, 2020-04-22:
> > 找了一台VPS来编译。
> > 可是开始最后一步时才想起来，VPS的用户默认就是ROOT的。。。
> > ROOT用户编译会出现什么问题？
> 
> 反正我用root没成功过，添加个用户就好了

今天有点晚了，明天继续折腾。
在没有去掉#号时，可以正确编译。
现在不行了，文件没有下载完？

VPS那边很快就错误了。重置了明天折腾。

ztfot, 2020-04-22:
> > > 找了一台VPS来编译。
> > > 可是开始最后一步时才想起来，VPS的用户默认就是ROOT的。。。
> > > ROOT用户编译会出现什么问题？
> > 
> > 
> > 反正我用root没成功过，添加个用户就好了
> 
> 今天有点晚了，明天继续折腾。
> 在没有去掉#号时，可以正确编译。
> 现在不行了，文件没有下载完？
> 
> VPS那边很快就错误了。重置了明天折腾。

我都快一天了，编译了4，5次都不行

hue715, 2020-04-22:
> 大家看看编译到最后这是什么错误
> ( cd /home/ztfot/lede/build_dir/hostpkg/go-1.14/src ; GOROOT_FINAL=/home/ztfot/lede/staging_dir/hostpkg/lib/go-cross [...]

有error:out of memory，内存不足

ztfot, 2020-04-22:
> 有error:out of memory，内存不足

刚刚换了4G内存的高频cpu 美国vps，不知道其他错误还会不会出来

hue715, 2020-04-22:
> 这个是我自己的虚拟机编译的报错
>
> bash: po2lmo: command not found
> Makefile:120: recipe for target '/home/h/lede/build_dir/target-aarch64_cortex-a72_musl/luci-app-ssr-plus/.pkgdir/luci-app-ssr-plus.installed' failed [...]

需要先编译luci-base，就不会有po2lmo报错了

ztfot, 2020-04-22:
> 需要先编译luci-base，就不会有po2lmo报错了

这个luci-base怎么编译a，给条命令呗
make package/feeds/luci/luci-base/compile V=99
我之前用这个命令还是错的

ztfot, 2020-04-22:
> 需要先编译luci-base，就不会有po2lmo报错了

文件下载完全
执行make package/feeds/luci/luci-base/compile V=99 有大段错误
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
Collecting package info: done
Collecting target info: done
time: target/linux/prereq#0.11#0.01#0.13
make[1]: Entering directory '/home/ztfot/lede'
make[2]: Entering directory '/home/ztfot/lede/feeds/luci/modules/luci-base'
. /home/ztfot/lede/include/shell.sh; bzcat /home/ztfot/lede/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.prepared66aff93c9579bf19d55999023fcd607e_6664517399ebbbc92a37c5bb081b5c53
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.configured
make -C src/ clean po2lmo
make[3]: Entering directory '/home/ztfot/lede/feeds/luci/modules/luci-base/src'
rm -f po2lmo parser.so version.lua .o
cc -c -o po2lmo.o po2lmo.c
cc -c -o template_lmo.o template_lmo.c
cc -o po2lmo po2lmo.o template_lmo.o
make[3]: Leaving directory '/home/ztfot/lede/feeds/luci/modules/luci-base/src'
make -C /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1 bin/LuaSrcDiet.lua
make[3]: Entering directory '/home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
make[3]: Nothing to be done for 'bin/LuaSrcDiet.lua'.
make[3]: Leaving directory '/home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
install -d -m0755 /home/ztfot/lede/staging_dir/hostpkg/bin
install -m0755 src/po2lmo /home/ztfot/lede/staging_dir/hostpkg/bin/po2lmo
install -m0755 /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /home/ztfot/lede/staging_dir/hostpkg/bin/LuaSrcDiet
mkdir -p /home/ztfot/lede/staging_dir/hostpkg/stamp
touch /home/ztfot/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
touch /home/ztfot/lede/staging_dir/hostpkg/stamp/.luci-base_installed
make[2]: Leaving directory '/home/ztfot/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.35#0.50#1.90
make[2]: Entering directory '/home/ztfot/lede/package/libs/toolchain'
touch /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.prepared_34ac9a1ef8dd17ffcf2bc1b765bcc888_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain
touch /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.prepared_34ac9a1ef8dd17ffcf2bc1b765bcc888_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.configured_
rm -f /home/ztfot/lede/staging_dir/target-x86_64_musl/stamp/.toolchain_installed
(cd /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/./; if [ -x ./configure ]; then find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r chmod u+w; find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/ztfot/lede/scripts/config.guess; find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r chmod u+w; find /home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /home/ztfot/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/ztfot/lede/staging_dir/target-x86_64_musl/usr/include -I/home/ztfot/lede/staging_dir/target-x86_64_musl/include -I/home/ztfot/lede/staging_dir/toolcha [... cut ...]
make[2]: *** [/home/ztfot/lede/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/ztfot/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.12#0.04#0.17
package/Makefile:111: recipe for target 'package/libs/toolchain/compile' failed
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/ztfot/lede'
/home/ztfot/lede/include/toplevel.mk:225: recipe for target 'package/feeds/luci/luci-base/compile' failed
make: *** [package/feeds/luci/luci-base/compile] Error 2

hue715, 2020-04-22:
> 文件下载完全
> 执行make package/feeds/luci/luci-base/compile V=99 有大段错误
> Checking 'working-make'... ok.
> Checking 'case-sensitive-fs'... ok.
> Checking 'proper-umask'... ok.
> Checking 'gcc'... ok.
> Checking [...]

看到toolchain报错，单独的软件编译需要先编译toolchain或者完整编译过（此时toolchain已编译），就不会有个这个报错。

hlz781105, 2020-04-22:
哎，这二天再折腾下，不行就有空了就报佐大的培训班了。

hn315, 2020-04-22:
一开始我卡在./scripts/feeds update -a这里，油管4K全局，就是更新不了，清早6点起床终于过了这一关；后面卡在helloworld，打命令替换不了#，手动硬改终于看到；第三次卡make -j8 download V=s 1天天的，还是在不断手动从网上下载DL，虚拟机有点坑人，网上http能下载，虚拟机里就是下不了；终于挺进make -j1 V=s，等了半天了，里面还时不时有DL没下载全，编个译这么难，费我2天了，还有跑码……

hue715, 2020-04-22:
> 一开始我卡在./scripts/feeds update -a这里，油管4K全局，就是更新不了，清早6点起床终于过了这一关；后面卡在helloworld，打命令替换不了#，手动硬改终于看到；第三次卡make -j8 download V=s 1天天的，还是在不断手动从网上下载DL，虚拟机有点坑人，网上http能下载，虚拟机里就是下不了；终于挺进make -j1 V=s，等了半天了，里面还时不时有DL没下载全，编个译这么难，费我2天了，还有跑码……

如果网不稳定，其实还不如用github actions

hlz781105, 2020-04-22:
给VPS加一个用户。用不用给ROOT权限？

hn315, 2020-04-22:
编译时每次下载缺失的DL 系统会重试5次，几K的速度，要疯

hlz781105, 2020-04-22:
如位大神告诉我加用户给什么权限，怎么设置谢谢。
sudo adduser abc 这个命令成功
但是这个用户权限好像不够
sudo vim /etc/sudoers  这个命令不行
sudo: unable to resolve host kkk
sudo: vim: command not found

hlz781105, 2020-04-22:
make -j8 download V=s
这个能不能重复执行？

hue715, 2020-04-22:
> make -j8 download V=s
> 这个能不能重复执行？

可以的

ztfot, 2020-04-22:
> 如位大神告诉我加用户给什么权限，怎么设置谢谢。
> sudo adduser abc 这个命令成功
> 但是这个用户权限好像不够
> sudo vim /etc/sudoers 这个命令不行
> sudo: unable to resolve host kkk
> sudo: vim: command not found

不用加权限，只要不用root就可以了，我是chmod 777 -R /home 的，就可以开始编译了

ztfot, 2020-04-22:
> 如位大神告诉我加用户给什么权限，怎么设置谢谢。
> sudo adduser abc 这个命令成功
> 但是这个用户权限好像不够
> sudo vim /etc/sudoers 这个命令不行
> sudo: unable to resolve host kkk
> sudo: vim: command not found

搜索ubuntu 18.04添加用户

ztfot, 2020-04-22:
> make -j8 download V=s
> 这个能不能重复执行？

个人比较喜欢nano ，其次是vi ，不用vim

minfengyk, 2020-04-22:
> 这个luci-base怎么编译a，给条命令呗
> make package/feeds/luci/luci-base/compile V=99
> 我之前用这个命令还是错的

也遇到同样问题，make package/feeds/luci/luci-base/compile V=99，再接着编译，（前提是文件下载完全），或者第一次编译时取消勾选SSR，编译成功后，再勾选SSR，第二次编译，很快的。两种方法亲测可行。

[43 more replies]

closed 2020-04-26 by coolsnowwolf