coolsnowwolf/lede #4411: 编译固件时报错

lanxianfeng, 2020-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译固件的时候报错

### (2) 路由器型号和固件版本

A：erx sfp

### (3) 详细日志

A：make[3]: Entering directory `/home/xflan/lede/package/lean/pdnsd-alt'
touch /home/xflan/lede/build_dir/target-mipsel_24kc_musl/pdnsd-1.2.9b-par/.prepared_55fb305a81337d821d895e38763a2d3b_6664517399ebbbc92a37c5bb081b5c53_check
. /home/xflan/lede/include/shell.sh; gzip -dc /home/xflan/lede/dl/pdnsd-1.2.9b-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz | tar -C /home/xflan/lede/build_dir/target-mipsel_24kc_musl/pdnsd-1.2.9b-par/.. -xf -
tar: This does not look like a tar archive

AmadeusGhost, 2020-04-22:
`tar: This does not look like a tar archive tar`
你网络问题，dl这个包（pdnsd-1.2.9b-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz）删了重新下

lanxianfeng, 2020-04-22:
现在报这个 Create index file './feeds/helloworld.index' 
WARNING: Makefile 'package/feeds/telephony/freeswitch-stable/Makefile' has a dependency on 'libopusenc', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch-stable/Makefile' has a dependency on 'libopusfile', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'trojan', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'kcptun-client', which does not exist
WARNING: Makefile 'package/feeds/lienol/luci-app-trojan-server/Makefile' has a dependency on 'trojan', which does not exist

hxlls, 2020-04-22:
warning 哪里是报错了？这个是警告的意思

ElectronicElephant, 2020-04-23:
home/xflan/lede/dl/pdnsd-1.2.9b-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz | tar -C /home/xflan/lede/build_dir/target-mipsel_24kc_musl/pdnsd-1.2.9b-par/.. -xf - tar: This does not look like a tar 

这个文件没有下载全，删除后重新下载，注意网络环境

lanxianfeng, 2020-04-26:
主要是网络问题 最新不好使 感谢

closed 2020-04-26 by lanxianfeng