coolsnowwolf/lede #4417: 自己在软路由上建的IPSec服务器手机连接正常，内网正常，国外网站正常（Google等），就是国内网站不行，试过把手工竹手工竹梯子关了，变成国内外都不行了，重启问题还是一样的。，内外网一切正常

qq1316237299, 2020-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
的IPSec服务器手机连接正常，内网正常，国外网站正常（Google等），就是国内网站不行
### (1) 具体问题

A：的IPSec服务器手机连接正常，内网正常，国外网站正常（Google等），就是国内网站不行

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: set name of thread 5767 to PLUGIN[diskspac
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: thread created with task id 5777
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: set name of thread 5777 to WEB_SERVER[stat
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: thread created with task id 5768
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: set name of thread 5768 to PLUGIN[tc]
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: thread created with task id 5780
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: set name of thread 5780 to STATSD_COLLECTO
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: STATSD collector thread started with taskid 5780
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: POLLFD: LISTENER: listening on 'udp:[::1]:8125'
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: POLLFD: LISTENER: listening on 'udp:127.0.0.1:8125'
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: POLLFD: LISTENER: listening on 'tcp:[::1]:8125'
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: POLLFD: LISTENER: listening on 'tcp:127.0.0.1:8125'
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_metrics/counters.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_metrics/timers.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_metrics/meters.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_metrics/histograms.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_metrics/sets.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/gauges.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/counters.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/timers.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/meters.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/histograms.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/sets.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/gauges.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/counters.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/timers.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/meters.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/histograms.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/sets.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/unknown.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_events/errors.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_reads/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_reads/tcp.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_reads/udp.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_bytes/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_bytes/tcp.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_bytes/udp.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_packets/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_packets/tcp.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.statsd_packets/udp.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.tcp_connects/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.tcp_connects/connects.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.tcp_connects/disconnects.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.tcp_connected/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.tcp_connected/connected.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.private_charts/main.db.
Thu Apr 23 01:18:01 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/netdata.private_charts/charts.db.
[... 538 lines cut ...]
Thu Apr 23 01:38:51 2020 daemon.info : 16[ENC] parsed TRANSACTION response 2740415864 [ HASH CPA(X_STATUS) ]
Thu Apr 23 01:38:51 2020 daemon.info : 16[IKE] IKE_SA xauth_psk[4] established between 115.207.93.246[115.207.93.246]...219.135.212.114[192.168.100.158]
Thu Apr 23 01:38:51 2020 authpriv.info : 16[IKE] IKE_SA xauth_psk[4] established between 115.207.93.246[115.207.93.246]...219.135.212.114[192.168.100.158]
Thu Apr 23 01:38:51 2020 daemon.info : 16[IKE] scheduling reauthentication in 10044s
Thu Apr 23 01:38:51 2020 daemon.info : 16[IKE] maximum IKE_SA lifetime 10584s
Thu Apr 23 01:38:51 2020 daemon.info : 16[NET] received packet: from 219.135.212.114[4500] to 115.207.93.246[4500] (124 bytes)
Thu Apr 23 01:38:51 2020 daemon.info : 16[ENC] parsed TRANSACTION request 2587909504 [ HASH CPRQ(ADDR MASK DNS NBNS U_BANNER U_DEFDOM U_SPLITDNS U_SPLITINC U_LOCALLAN VER) ]
Thu Apr 23 01:38:51 2020 daemon.info : 16[IKE] peer requested virtual IP %any
Thu Apr 23 01:38:51 2020 daemon.info : 16[CFG] reassigning offline lease to 'qq1316237299'
Thu Apr 23 01:38:51 2020 daemon.info : 16[IKE] assigning virtual IP 192.168.8.200 to peer 'qq1316237299'
Thu Apr 23 01:38:51 2020 daemon.info : 16[ENC] generating TRANSACTION response 2587909504 [ HASH CPRP(ADDR DNS) ]
Thu Apr 23 01:38:51 2020 daemon.info : 16[NET] sending packet: from 115.207.93.246[4500] to 219.135.212.114[4500] (92 bytes)
Thu Apr 23 01:38:52 2020 daemon.info : 07[NET] received packet: from 219.135.212.114[4500] to 115.207.93.246[4500] (668 bytes)
Thu Apr 23 01:38:52 2020 daemon.info : 07[ENC] parsed QUICK_MODE request 3096370330 [ HASH SA No ID ID ]
Thu Apr 23 01:38:52 2020 daemon.info : 07[IKE] received 28800s lifetime, configured 3600s
Thu Apr 23 01:38:52 2020 daemon.info : 07[ENC] generating QUICK_MODE response 3096370330 [ HASH SA No ID ID ]
Thu Apr 23 01:38:52 2020 daemon.info : 07[NET] sending packet: from 115.207.93.246[4500] to 219.135.212.114[4500] (172 bytes)
Thu Apr 23 01:38:52 2020 daemon.info : 08[NET] received packet: from 219.135.212.114[4500] to 115.207.93.246[4500] (76 bytes)
Thu Apr 23 01:38:52 2020 daemon.info : 08[ENC] parsed QUICK_MODE request 3096370330 [ HASH ]
Thu Apr 23 01:38:52 2020 daemon.info : 08[IKE] CHILD_SA xauth_psk{4} established with SPIs c9f56666_i 088a3457_o and TS 0.0.0.0/0 === 192.168.8.200/32
[... cut ...] SPIs c9f56666_i 088a3457_o and TS 0.0.0.0/0 === 192.168.8.200/32
Thu Apr 23 01:39:07 2020 kern.info kernel: [ 1291.865047] sh (9898): drop_caches: 3
Thu Apr 23 01:40:04 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/net_drops.br_lan/main.db.
Thu Apr 23 01:40:04 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/net_drops.br_lan/inbound.db.
Thu Apr 23 01:40:04 2020 daemon.info netdata[5640]: Initializing file /var/cache/netdata/net_drops.br_lan/outbound.db.

coolsnowwolf, 2020-04-23:
某些安卓手机就是这样，MTU 没有自动设置造成的

换用 iPhone 即可解决

qq1316237299, 2020-04-23:
> 某些安卓手机就是这样，MTU 没有自动设置造成的
> 
> 换用 iPhone 即可解决

我靠 还有这种操作的。。。  
L大你好，我在ssh里加了iptables -t nat -A POSTROUTING -j MASQUERADE
iptables -I FORWARD -p tcp --syn -i ppp+ -j TCPMSS --set-mss 1356
这两条代码就可以了，这是为什么呢 这是我网上找来的代码，请问L大可以把这两条加入到源码里面吗？

qq1316237299, 2020-04-23:
> 某些安卓手机就是这样，MTU 没有自动设置造成的
> 
> 换用 iPhone 即可解决

要在ssh里打才可以好像，加入到防火墙里面的自定义规则里面还想不太行。

closed 2020-04-28 by coolsnowwolf