coolsnowwolf/lede #4418: 更新到最新版本， 本地编译报错，搞了一晚上无解，求助！

keivenliao, 2020-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

keiven@debian-desktop:~/lede$ make -j1 V=s
make[1]: Entering directory '/home/keiven/lede'
make[2]: Entering directory '/home/keiven/lede'
+ mkdir -p /home/keiven/lede/staging_dir/target-x86_64_musl
+ cd /home/keiven/lede/staging_dir/target-x86_64_musl
+ mkdir -p bin lib include stamp
mkdir -p /home/keiven/lede/build_dir/target-x86_64_musl/stamp
touch /home/keiven/lede/staging_dir/target-x86_64_musl/.prepared
make[3]: Entering directory '/home/keiven/lede/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/keiven/lede/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.00
make[3]: Entering directory '/home/keiven/lede/toolchain/binutils'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/keiven/lede/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.00#0.01
make[3]: Entering directory '/home/keiven/lede/toolchain/gcc/initial'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/keiven/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#0.00#0.00#0.01
make[3]: Entering directory '/home/keiven/lede/toolchain/kernel-headers'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/keiven/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#0.01#0.01#0.03
make[3]: Entering directory '/home/keiven/lede/toolchain/musl'
make[3]: Leaving directory '/home/keiven/lede/toolchain/musl'
time: toolchain/musl/compile#0.09#0.04#0.12
make[3]: Entering directory '/home/keiven/lede/toolchain/gcc/final'
export SHELL="bash"; make CFLAGS="-O2 -I/home/keiven/lede/staging_dir/host/include " CFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CXXFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" GOCFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" gcc_cv_libc_provides_ssp=yes  -C /home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final all
make[4]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final'
make[5]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libiberty'
make[7]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libiberty/testsuite'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libiberty/testsuite'
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libiberty'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/fixincludes'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/fixincludes'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/lto-plugin'
make  all-am
make[7]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/lto-plugin'
make[7]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/lto-plugin'
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/lto-plugin'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/intl'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/intl'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/libiberty'
make[7]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/libiberty'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/fixincludes'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/fixincludes'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/build-x86_64-pc-linux-gnu/libcpp'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=-O2 -I/home/keiven/lede/staging_dir/host/include " "CXXFLAGS=-g -O2" "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc " "LIBCFLAGS=-O2 -I/home/keiven/lede/staging_dir/host/include " "LIBCFLAGS_FOR_TARGET=-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" "MAKE=make" "MAKEINFO=/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0/missing makeinfo --split-size=5000000 --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/keiven/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl" "infodir=/home/keiven/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/share/info" "libdir=/home/keiven/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib" "prefix=/home/keiven/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl" "tooldir=/home/keiven/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/x86_64-openwrt-linux-musl" "AR=ar" "AS=as" "CC=gcc" "CXX=g++" "LD=ld" "LIBCFLAGS=-O2 -I/home/keiven/lede/staging_dir/host/include " "NM=nm" "PICFLAG=" "RANLIB=ranlib" "DESTDIR=" DO=all multi-do # make
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/zlib'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libbacktrace'
make  all-am
[... 3 lines cut ...]
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libbacktrace'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libcpp'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libdecnumber'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/libdecnumber'
make[6]: Entering directory '/home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/gcc'
[... cut ...]
make[3]: *** [Makefile:88: /home/keiven/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gcc-7.5.0-final/.built] Error 2
make[3]: Leaving directory '/home/keiven/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#4.05#1.13#5.12
make[2]: *** [toolchain/Makefile:100: toolchain/gcc/final/compile] Error 2
make[2]: Leaving directory '/home/keiven/lede'
make[1]: *** [toolchain/Makefile:96: /home/keiven/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/keiven/lede'
make: *** [/home/keiven/lede/include/toplevel.mk:227：world] 错误 2

lxwq, 2020-04-23:
loglog,errlog

jw10126121, 2020-04-23:
touch /home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.built
rm -rf /home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc.installed /home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc
mkdir -p /home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc
install -d -m0755 /home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc/lib /home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc/usr/bin
cp -fpR /home/ljw/linOpenwrt/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/lib/ld-musl-*.so* /home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc/lib/
cp: cannot stat '/home/ljw/linOpenwrt/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/lib/ld-musl-*.so*': No such file or directory
Makefile:618: recipe for target '/home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc.installed' failed
make[2]: *** [/home/ljw/linOpenwrt/openwrt/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/ljw/linOpenwrt/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.10#0.04#0.13
package/Makefile:111: recipe for target 'package/libs/toolchain/compile' failed
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/ljw/linOpenwrt/openwrt'
/home/ljw/linOpenwrt/openwrt/include/toplevel.mk:225: recipe for target 'package/feeds/luci/luci-base/compile' failed
make: *** [package/feeds/luci/luci-base/compile] Error 2

xnxy2012, 2020-04-23:
全局上网，就好了

f3404923, 2020-04-24:
先开全局梯子，建议rm -rf ./tmp ./build_dir ,然后make download V=s看看是否有包下载错误，最后在make -j1 V=s

keivenliao, 2020-04-24:
最后， 执行了一次 make clean 再编译就可以了。

kktt007, 2020-04-25:
> 
> 
> 最后， 执行了一次 make clean 再编译就可以了。

我好了，是否rm -rf ./build_dir和make clean都是必须的,反正通过新版helloworld发现rm-rf ./tmp是必须的。另外我好了

ElectronicElephant, 2020-04-30:
`rm -rf ./build_dir` 和 `make clean` 都是必须的

kktt007, 2020-04-30:
> 
> 
> `rm -rf ./build_dir` 和 `make clean` 都是必须的

谢谢，在编译x86固件后再编译R7800,执行了rm -rf./build ./tmp && make clean还不行，然后又加了个make V=s 就好了，已经全局了哎,宝宝心累啊。

closed 2020-07-04 by coolsnowwolf