coolsnowwolf/lede #4427: nlbwmon进程CPU使用率过高

Clackz, 2020-04-23:
斐讯k3 版本是到639f363这个commit的编译的，在运行几天后发现路由器CPU使用率过高，top命令查看发现是nlbwmon这个进程占用很高，重启进程后恢复正常，日志中没有发现异常的地方……还需要观察几天看看……
[image]

ElectronicElephant, 2020-04-23:
很久以前也遇到过，当时的做法就是直接卸了这个鸡肋😂😂😂

stock169, 2020-04-24:
这玩意一直有问题的，不定时不定期只要运行时间久了就一定会拖死CPU的，直接卸掉最好

hcym, 2020-04-24:
还有一个netdata，wan3什么的，自己编译一定要注意去除干净相关依赖包

badcrazy, 2020-04-25:
是的，我改用bandwidthd了，虽然查看需要ip/bandwidthd，不如nlbw方便，但是不拖cpu

zhouym233, 2020-04-25:
> 是的，我改用bandwidthd了，虽然查看需要ip/bandwidthd，不如nlbw方便，但是不拖cpu

你好，我想问一下bandwidthd是在编译时添加吗？似乎在编译选项里没有找到。

zwillhill, 2020-04-28:
同样的问题，卸载完事。问题是menuconfig的时候无法取消，不知道是哪个依赖了。

badcrazy, 2020-04-29:
> bandwidthd
op官方源就有

badcrazy, 2020-04-29:
> > 是的，我改用bandwidthd了，虽然查看需要ip/bandwidthd，不如nlbw方便，但是不拖cpu
> 
> 你好，我想问一下bandwidthd是在编译时添加吗？似乎在编译选项里没有找到。

op官方源就有

Cathgao, 2020-05-01:
建议直接取消编译这个，因为挺鸡肋，统计也不精确

82kg, 2020-05-20:
> > 是的，我改用bandwidthd了，虽然查看需要ip/bandwidthd，不如nlbw方便，但是不拖cpu
> 
> 你好，我想问一下bandwidthd是在编译时添加吗？似乎在编译选项里没有找到。

bandwidthd有luci-app的，但是L大好像没加这个包

closed 2020-07-04 by coolsnowwolf