coolsnowwolf/lede #4428: 当启用ssrplus的启用广告屏蔽无法打开淘宝的弹出式页面

norikohxb, 2020-04-23:
当我开启了ssrplus的广告屏蔽，就无法在淘宝打开搜索结果的弹出式页面
注意到域名上是
https://click.simba.taobao.com/cc_im?spm=a230r.1.14.6&p=%BB%DD%CD%FE%207.1&s=1757595097&k=577&e=2EqhAgR5guTNCGhuVO4b63XacDax8rtpjIH478icTcb67Bb2oi9laOFZybqpvuTAxcGmv2d5BjnqSHVsCcJxCOhwibnUt9G4dnaFam4UC4F4i6H8jARi%2B7cRgaQNnqUCNSGcFPDOvnlB1ZZxdaze59OlJFLiBpIM32ThuHcGmBplmfdSzEPc7wyEgGzfHwL%2FMCGT1aV%2F%2Bl3JpbAXi5emmIhildjkbKn%2FWSK7qX59E4NjIKQMqTeqgc3gI2Rzbt3rYOb%2F6c2ISFwk%2F3opfURWAAF6g%2FJ0C7N2QiqLwZnzGbL29lRLvN1Gr57WVEDJT%2FfYq0vyCbDHqNBEqgYvFM%2FITjGjdFLO2qDZFlA0YsXcLguXwMC1FtjIqeiR%2BTuYV%2B4%2F%2FAcTs1ADO2MDbzTazmXHylresZ7CPJCxJ3rXejFXkZFgZKrxIvl2%2Fjjd0d06q3580ejzVaNDR3IS6VCHFQWM%2FESqBi8Uz8hO1d5r2EDOjUqNrFv6aceNziAukj0C18KrdWp%2BdOv8FPvpC1k8lsPKiTfO%2FGLBFMzPXuRHNdybUrTS1R6oVJMTgxQg%2BZFVG3tg

比如你在淘宝主页搜索 惠威7.1
然后搜索结果无法点击弹出窗口。

norikohxb, 2020-04-23:
> 1.此项目已经不维护ssrp了
> 2.广告屏蔽为全局
> 3.请到广告规则列表维护者github提交issue

那就是了，这个广告屏蔽就是全局屏蔽，不管墙内墙外清一色斩立决。理解了，那我就关闭他咯

norikohxb, 2020-04-23:
> 1.此项目已经不维护ssrp了
> 2.广告屏蔽为全局
> 3.请到广告规则列表维护者github提交issue

你好yang，请问你这个广告屏蔽是全局的你是在哪里看到说明的？是你自己看代码？还是说哪个地方可以了解每个功能的使用性能？
如果能告诉我是哪里看到的，我也可以去看看这样好有个了解。
一颗心在lean大的lede吊死了

EnnawYang, 2020-04-23:
> > 1.此项目已经不维护ssrp了
> > 2.广告屏蔽为全局
> > 3.请到广告规则列表维护者github提交issue
> 
> 你好yang，请问你这个广告屏蔽是全局的你是在哪里看到说明的？是你自己看代码？还是说哪个地方可以了解每个功能的使用性能？
> 如果能告诉我是哪里看到的，我也可以去看看这样好有个了解。
> 一颗心在lean大的lede吊死了

经常关注issue的人应该都看到过，如果你觉得默认的规则不好用的话可以换其他的，我自己使用默认规则也有误杀，就没用屏蔽广告的功能了，非刚需

norikohxb, 2020-04-23:
> > > 1.此项目已经不维护ssrp了
> > > 2.广告屏蔽为全局
> > > 3.请到广告规则列表维护者github提交issue
> > 
> > 
> > 你好yang，请问你这个广告屏蔽是全局的你是在哪里看到说明的？是你自己看代码？还是说哪个地方可以了解每个功能的使用性能？
> > 如果能告诉我是哪里看到的，我也可以去看看这样好有个了解。
> > 一颗心在lean大的lede吊死了
> 
> 经常关注issue的人应该都看到过，如果你觉得默认的规则不好用的话可以换其他的，我自己使用默认规则也有误杀，就没用屏蔽广告的功能了，非刚需

我发错地方了应该发错到helloworld去，不好意思，我自己都知道那个插件，就是忘记了。
然后你有好的墙外去广告的推荐吗。毕竟他是全局的，其实相当墙外墙内。再有请教你一下，能否通过这个把youtube的5秒钟倒计时广告去掉吗？
我渴望了很久了

EnnawYang, 2020-04-23:
> 我发错地方了应该发错到helloworld去，不好意思，我自己都知道那个插件，就是忘记了。
> 然后你有好的墙外去广告的推荐吗。毕竟他是全局的，其实相当墙外墙内。再有请教你一下，能否通过这个把youtube的5秒钟倒计时广告去掉吗？
> 我渴望了很久了

https://github.com/coolsnowwolf/lede/issues/3208#issuecomment-588589205
广告规则应该是这个，具体是哪个规则不记得了，自己看下吧，去这个项目提issue。去广告我用浏览器插件(PC)，手机用youtube Vanced (android)

norikohxb, 2020-04-23:
> > > > > 1.此项目已经不维护ssrp了
> > > > > 2.广告屏蔽为全局
> > > > > 3.请到广告规则列表维护者github提交issue
> > > > 
> > > > 
> > > > 你好yang，请问你这个广告屏蔽是全局的你是在哪里看到说明的？是你自己看代码？还是说哪个地方可以了解每个功能的使用性能？
> > > > 如果能告诉我是哪里看到的，我也可以去看看这样好有个了解。
> > > > 一颗心在lean大的lede吊死了
> > > 
> > > 
> > > 经常关注issue的人应该都看到过，如果你觉得默认的规则不好用的话可以换其他的，我自己使用默认规则也有误杀，就没用屏蔽广告的功能了，非刚需
> > 
> > 
> > 我发错地方了应该发错到helloworld去，不好意思，我自己都知道那个插件，就是忘记了。
> > 然后你有好的墙外去广告的推荐吗。毕竟他是全局的，其实相当墙外墙内。再有请教你一下，能否通过这个把youtube的5秒钟倒计时广告去掉吗？
> > 我渴望了很久了
> 
> [#3208 (comment)](https://github.com/coolsnowwolf/lede/issues/3208#issuecomment-588589205)
> 广告规则应该是这个，具体是哪个规则不记得了，自己看下吧，去这个项目提issue。去广告我用浏览器插件(PC)，手机用youtube Vanced (android)

非常感谢你的答复可能你平时关注比较多，我也是用vanced。能否再请教你一下这个去广告的设置是针对全局还是国内上网的？
[image]

EnnawYang, 2020-04-24:
> 非常感谢你的答复可能你平时关注比较多，我也是用vanced。能否再请教你一下这个去广告的设置是针对全局还是国内上网的？
> [image]

全局，不过我的op没使用任何去广告的软件

norikohxb, 2020-04-24:
> 全局，不过我的op没使用任何去广告的软件

你的爱奇艺app也是交会员？还是很少看？我主要用lean大的adbyby过滤爱奇艺和奇异果的视频广告很好用

EnnawYang, 2020-04-24:
> 你的爱奇艺app也是交会员？还是很少看？我主要用lean大的adbyby过滤爱奇艺和奇异果的视频广告很好用

很少看国内平台

closed 2020-04-24 by norikohxb