coolsnowwolf/lede #4429: 编译的R3G固件要丢包

xzjue5, 2020-04-23:
老是要丢包断网，看了日志好像老是要重新启动网络

下面是日志
[ 2438.902672] Call Trace:
[ 2438.905161] [<8000c700>] show_stack+0x58/0x100
[ 2438.909642] [<8048e654>] dump_stack+0xa4/0xe0
[ 2438.914040] [<8002b1e8>] __warn+0xe0/0x114
[ 2438.918153] [<8002b2ac>] warn_slowpath_null+0x1c/0x30
[ 2438.923256] [<8eef4330>] mt7603_filter_tx+0x100/0x164 [mt7603e]
[ 2438.929227] [<8eef4498>] mt7603_wtbl_set_ps+0xa0/0x12c [mt7603e]
[ 2438.935299] [<8eef18ec>] mt7603_sta_ps+0x38/0x3ac [mt7603e]
[ 2438.940923] [<8e593fb4>] mt76_rx_poll_complete+0x474/0x544 [mt76]
[ 2438.947079] [<8e591874>] mt76_dma_cleanup+0x178/0x454 [mt76]
[ 2438.952985] ---[ end trace f4e9c2782416c301 ]---
[ 4334.712091] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[ 4336.503628] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4336.509302] mt76x2e 0000:02:00.0: Build: 1
[ 4336.513569] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4336.542102] mt76x2e 0000:02:00.0: Firmware running!
[ 4336.552378] ieee80211 phy1: Hardware restart was requested
[ 4337.592064] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[ 4344.544370] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4344.551087] mt76x2e 0000:02:00.0: Build: 1
[ 4344.556168] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4344.582138] mt76x2e 0000:02:00.0: Firmware running!
[ 4344.592622] ieee80211 phy1: Hardware restart was requested
[ 4370.071981] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[ 4370.935332] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4370.940824] mt76x2e 0000:02:00.0: Build: 1
[ 4370.945051] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4370.971857] mt76x2e 0000:02:00.0: Firmware running!
[ 4370.983622] ieee80211 phy1: Hardware restart was requested
[ 4381.969575] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[ 4384.845145] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4384.850782] mt76x2e 0000:02:00.0: Build: 1
[ 4384.855111] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4384.892646] mt76x2e 0000:02:00.0: Firmware running!
[ 4384.902107] ieee80211 phy1: Hardware restart was requested
[ 4390.392017] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[ 4392.558960] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4392.564886] mt76x2e 0000:02:00.0: Build: 1
[ 4392.568983] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4392.591695] mt76x2e 0000:02:00.0: Firmware running!
[ 4392.602035] ieee80211 phy1: Hardware restart was requested
[ 4395.591644] mt76x2e 0000:02:00.0: MCU message 30 (seq 1) timed out
[ 4398.453176] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4398.458844] mt76x2e 0000:02:00.0: Build: 1
[ 4398.463403] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4398.491686] mt76x2e 0000:02:00.0: Firmware running!
[ 4398.501974] ieee80211 phy1: Hardware restart was requested
[ 4401.191605] mt76x2e 0000:02:00.0: MCU message 30 (seq 5) timed out
[ 4402.557126] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4402.562806] mt76x2e 0000:02:00.0: Build: 1
[ 4402.566982] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4402.591618] mt76x2e 0000:02:00.0: Firmware running!
[ 4402.602091] ieee80211 phy1: Hardware restart was requested
[ 4404.636877] mt76x2e 0000:02:00.0: MCU message 30 (seq 9) timed out
[ 4406.636241] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 4406.642151] mt76x2e 0000:02:00.0: Build: 1
[ 4406.646370] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 4406.671589] mt76x2e 0000:02:00.0: Firmware running!
[ 4406.685986] ieee80211 phy1: Hardware restart was requested

编译的Lienol固件就没有这个问题

jw10126121, 2020-04-24:
我的，好像也会这样。。

xzjue5, 2020-04-24:
> 我的，好像也会这样。。

000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

xzjue5, 2020-04-24:
> 我的，好像也会这样。。

用LienoI的就没问题   我现在用lede的固件LienoI的配置  这样就解决丢包了  很稳定

closed 2020-04-24 by xzjue5