coolsnowwolf/lede #4433: 反复半天左右丢失internet连接，dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start,DNS rebinding protection is active

kevinlyliu, 2020-04-24:
## 1.关于你要提交的问题

软路由lean openwrt半天左右就会丢失internet连接

Q：是否搜索了issue (使用 "x" 选择)
* [ ×] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：小白刚上路，买了个J1900成品软路由，前后试过多个不同版本的固件，目前在用esir精品小包精简版。经常性地每隔半天到接近一天会出现无internet连接的情况。

### (2) 路由器型号和固件版本

A：J1900成品软路由，OpenWrt eSir精品小包 openwrt-spp-v4.1[2020]-x86-64-combined-squashs.img 

### (3) 详细日志

A：
Fri Apr 24 23:21:50 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:02 2020 cron.info crond[24134]: crond (busybox 1.30.0) started, log level 5
Fri Apr 24 23:22:03 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:03 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:03 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:04 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:04 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:04 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:16 2020 cron.info crond[24814]: crond (busybox 1.30.0) started, log level 5
Fri Apr 24 23:22:16 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:16 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:17 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:17 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:17 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:18 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:30 2020 cron.info crond[25482]: crond (busybox 1.30.0) started, log level 5
Fri Apr 24 23:22:30 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:30 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:30 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:31 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:31 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:31 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:44 2020 cron.info crond[26161]: crond (busybox 1.30.0) started, log level 5
Fri Apr 24 23:22:44 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:44 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:44 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:45 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:45 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:45 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:57 2020 cron.info crond[26829]: crond (busybox 1.30.0) started, log level 5
Fri Apr 24 23:22:58 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:58 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:58 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:22:59 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:22:59 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:22:59 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:23:11 2020 cron.info crond[27497]: crond (busybox 1.30.0) started, log level 5
Fri Apr 24 23:23:11 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:23:11 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:23:12 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:23:12 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:23:12 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:23:13 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:23:25 2020 cron.info crond[28982]: crond (busybox 1.30.0) started, log level 5
Fri Apr 24 23:23:25 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:23:25 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:23:25 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
Fri Apr 24 23:23:26 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Apr 24 23:23:26 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Apr 24 23:23:26 2020 user.notice dnsmasq: found already running DHCP-server on interface 'br-lan' refusing to start, use 'option force 1' to override
[... 1198 lines cut ...]
[  403.180387] br-lan: port 3(eth3) entered blocking state
[  403.185806] br-lan: port 3(eth3) entered forwarding state
[  422.078852] e1000e: eth3 NIC Link is Down
[  422.083193] br-lan: port 3(eth3) entered disabled state
[  425.150765] e1000e: eth3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[  425.158783] br-lan: port 3(eth3) entered blocking state
[  425.164186] br-lan: port 3(eth3) entered forwarding state
[  435.503149] e1000e: eth3 NIC Link is Down
[  435.507406] br-lan: port 3(eth3) entered disabled state
[  438.514836] e1000e: eth3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[  438.522860] br-lan: port 3(eth3) entered blocking state
[  438.528265] br-lan: port 3(eth3) entered forwarding state
[ 3589.645320] e1000e: eth2 NIC Link is Down
[ 3589.649659] br-lan: port 2(eth2) entered disabled state
[ 3594.988445] e1000e: eth2 NIC Link is Up 100 Mbps Full Duplex, Flow Control: Rx/Tx
[ 3594.996298] e1000e 0000:03:00.0 eth2: 10/100 speed: disabling TSO
[ 3595.002659] br-lan: port 2(eth2) entered blocking state
[ 3595.008099] br-lan: port 2(eth2) entered forwarding state
[ 6020.182798] e1000e: eth3 NIC Link is Down
[ 6020.187135] br-lan: port 3(eth3) entered disabled state
[ 6043.593176] e1000e: eth3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[ 6043.601219] br-lan: port 3(eth3) entered blocking state
[ 6043.606611] br-lan: port 3(eth3) entered forwarding state
[ 6043.662376] e1000e: eth2 NIC Link is Down
[ 6044.622808] br-lan: port 2(eth2) entered disabled state

hxlls, 2020-04-24:
找错门了，请找固件作者。

kevinlyliu, 2020-04-24:
请问还需要别的信息吗？全新联通光猫桥接，软路由192.168.2.1，eth0, 2, 3分别接一台华硕68u已刷官方最新固件做ap设为192.168.2.221，一台tplink tl-pa1200电力猫协同扩展器，及空闲。

kevinlyliu, 2020-04-24:
> 找错门了，请找固件作者。

也是啊😯

hxlls, 2020-04-24:
不同编译的人做的修改不一样，只能找作者，如果是这个源码未修改编译的还能参考下，这种什么XXX优化的，找固件作者把。帮不了你，顺便说一句，你的固件估计可能是用的L大源码，但是不代表没做其他修改，所以出了什么问题，只能找对应的人

hlz781105, 2020-04-25:
可以自己试着编译一个了。

R3pl4c3r, 2020-04-25:
你的局域网内已经有一个DHCP服务器了，要么关掉那个用op的DHCP服务器，要么关掉op的DHCP服务器用你原来的

kevinlyliu, 2020-04-25:
> 你的局域网内已经有一个DHCP服务器了，要么关掉那个用op的DHCP服务器，要么关掉op的DHCP服务器用你原来的

找不到这个DHCP服务器啊，请问光猫直连，给配了个192.168.1.1，而我的软路由自己设定了192.168.2.1。这样子算是光猫也有DHCP功能吗？但是光猫已经不在2.1这个网段上了，还有几个空着的百兆口。。。

kevinlyliu, 2020-04-25:
最新进展：

在一位朋友指点下关掉了软路由lan口的dhcp，分别单独接硬路由ap（192.168.2.221）的无线，有线，电力猫的无线信号，恢复internet，但IP地址都是192.168.2.X，似乎软路由的dhcp功能关不掉？

kevinlyliu, 2020-04-25:
不好意思，看后台还在冲突。大概率是固件问题，不行继续换固件。确实不该再在这里干扰大家了。。。

hlz781105, 2020-04-25:
局域网组网的一个基本就是一个局域网里面只能有一个DHCP设备，二次路由是另外一种情况。
软路由按路由器的方式配好。
68U用中继模式。
电力猫也用中继或桥接什么的，这个电力猫没有用过，不清楚它是不是路由功能。
这样组网后只有软路由在分配IP。

软路由的DHCP在网络-接口-DHCP- 忽略此接口 更改。

9l, 2020-04-25:
你用电脑直接连上光猫，如果电脑能得到一个192.168的正常ip，那就是你光猫的路由功能dhcp服务这些没有关掉

coolsnowwolf, 2020-04-26:
LAN 里面有其他的 DHCP 服务器，所以 OP 的 DHCP 不会起来

kevinlyliu, 2020-04-27:
> LAN 里面有其他的 DHCP 服务器，所以 OP 的 DHCP 不会起来

谢谢！反复检查，发现华硕路由无线信号下面我接的一个tplink双频无线扩展器居然默认开着DHCP，导致它下面有线连接的一个安卓盒子疯狂寻址。关掉后，暂时没有再发现问题。

closed 2020-04-27 by kevinlyliu