coolsnowwolf/lede #4437: lean大或者hxlls或者Promix953大有空的时候能否指点一下？

norikohxb, 2020-04-25:
之前lean大春节过后那段时间更新很频繁的时候我也是每天更新。就是每天git pull然后编译。但是从3月28日之后我印象是adbyby增加了自定义订阅第三方屏蔽列表的功能还有ssr增加屏蔽广告功能后。我的三星早期的安卓5.0的平板无论是用官方的youtube还是youtubevanced播放视频很容易断线，失联。就是貌似那个ssr的连接很容易中断，在那边转圈半天不播放视频。或者显示无连接。在同一时间我拿出三星s10的手机安卓1.0，在同一个wifi下面用youtube或者是vanced，播放同一个youtube视频，不会卡顿，也很正常。而且这个问题存在我手头3,4个三星的note12.2 安卓5.0的版本。然后我也也因此怀疑dns的问题也发了几个帖子给hxlss或者promix953说不要去说是dns的问题。 最后证实也不是dns的问题。 
总的现在的现象是我用lean大的3月21日自己编译的固件，三星老的平板看youtube没有任何问题。不会中断。
用最新的lean大的lede的固件，新的三星手机没有问题，新的华为平板也没有问题，就是老的安卓5.0的平板的youtube总是断流。而且就算是用新的固件我不启用adbyby的订阅第三方广告，ssrplus不开启全局的屏蔽广告模式。一样也会断流。

而我的编译也没有问题，因为全新的安卓10.0或者9.0的手机都可以轻松的播放youtuber视频，就是安卓5.0的手机播放youtube会转圈圈，你点开一个视频开头缓存个几秒就不播放了。甚至在用老的平板的同时，我也会用s10的手机播放同一个视频，就是s10顺畅，安卓5.0播放前几秒就停住了，而且我还会发现，我必须点击其他节目，然后又可以缓存前面几秒或者1分钟，让你播放，播放完毕又卡顿了，就是不续传的样子。

我自己的谷歌vps，自己的trojan，自己用。而且也排除了当时vps存在问题。
我今天又切换回去我自己存在硬盘的3月21日编译的lean大的lede。
但是如果我还想要用最新的lede，也想要用三星的早期的安卓5.0的平板看youtube，有什么可以共存方式？

我甚至在虚拟机架设了2个lede，一个给安卓5.0的平板用3月21日的lede，一个是给新的安卓9.0以上新的设备用最新的lede。 这样也都可以完美解决问题。

[image]
[image]

hxlls, 2020-04-25:
广告大师和SSRP的屏蔽方式不太一样，广告大师是匹配的规则，SSrp 的屏蔽广告是dnsmasq ，直接从解析部分去屏蔽广告，理论上dnsmasq 的效率更高，并且是全局。至于说部分能上和部分手机不能上，这个说不准，毕竟很多的安卓手机内置有自己的DNS ，这个只能自己找到对应的设置进行设置，强制为路由网关做DNS，并不一定是路由造成的，至于说部分版本可以部分版本不行，这种理论上是不可能的，但是不排除非全新写入，或者保存配置升级带来的错误，只能自己测试和判断

norikohxb, 2020-04-25:
> 广告大师和SSRP的屏蔽方式不太一样，广告大师是匹配的规则，SSrp 的屏蔽广告是dnsmasq ，直接从解析部分去屏蔽广告，理论上dnsmasq 的效率更高，并且是全局。至于说部分能上和部分手机不能上，这个说不准，毕竟很多的安卓手机内置有自己的DNS ，这个只能自己找到对应的设置进行设置，强制为路由网关做DNS，并不一定是路由造成的，至于说部分版本可以部分版本不行，这种理论上是不可能的，但是不排除非全新写入，或者保存配置升级带来的错误，只能自己测试和判断

主要是我最新编译的lede我2个广告都不开。还是一样存在安卓5.0的youtube问题。

norikohxb, 2020-04-25:
> 广告大师和SSRP的屏蔽方式不太一样，广告大师是匹配的规则，SSrp 的屏蔽广告是dnsmasq ，直接从解析部分去屏蔽广告，理论上dnsmasq 的效率更高，并且是全局。至于说部分能上和部分手机不能上，这个说不准，毕竟很多的安卓手机内置有自己的DNS ，这个只能自己找到对应的设置进行设置，强制为路由网关做DNS，并不一定是路由造成的，至于说部分版本可以部分版本不行，这种理论上是不可能的，但是不排除非全新写入，或者保存配置升级带来的错误，只能自己测试和判断

我引入那个说法并不是想要起到误导的作用，只是想要说明lede的时间上的差异。3月21日的版本，安卓5.0的平板的youtube app看视频不会中断，用了最新的lede，在安卓9.0以上电脑看youtube正常的情况下，安卓5.0的平板，开一个视频刚开头的第一次缓冲的可以看，看完就不加载了，然后你如果再切换一个视频，又是开头可以看。，然后我开头也以为是去广告的问题所以吧去广告的都关闭了。结果发现还是一样，所以问题应该差别在内核或者ssr上面

norikohxb, 2020-04-27:
昨天晚上看到helloworld有更新了今天又编译了一次。测试一下最新的固件，一样存在问题，三星老款平板，youtube的app无法看视频。用网页可以，youtube的视频缓存很容易不足就卡住。基本上是点开一个图片的时候会出现几秒钟的缓存让你播放然后就卡死。同一台平板用chrome没有问题，同一个局域网下面三星s10的手机播放youtube app无问题。。测试的lede关闭了adbyby，关闭了netflixd的分流。
[image]
[image]
[image]
[image]
[image]
[image]

norikohxb, 2020-04-27:
> 广告大师和SSRP的屏蔽方式不太一样，广告大师是匹配的规则，SSrp 的屏蔽广告是dnsmasq ，直接从解析部分去屏蔽广告，理论上dnsmasq 的效率更高，并且是全局。至于说部分能上和部分手机不能上，这个说不准，毕竟很多的安卓手机内置有自己的DNS ，这个只能自己找到对应的设置进行设置，强制为路由网关做DNS，并不一定是路由造成的，至于说部分版本可以部分版本不行，这种理论上是不可能的，但是不排除非全新写入，或者保存配置升级带来的错误，只能自己测试和判断

lean大hxlls大Promix953大我解决了问题了。我通过adguardhome找到了屏蔽的这个域名securepubads.g.doubleclick.net把它加入到adbyby的不走代理的域名里面。三星安卓5.0的平板+youtube的app就不会有问题饿了。

但是现在我还是想要请教你们这个问题，同样的lede，为什么3月21日的lede不需要特别把securepubads.g.doubleclick.net加入到adbyby白名单？而在4月份的版本就需要把securepubads.g.doubleclick.net加入到白名单方可以让老的平板上youtube的app（新的手机也不需要这样操作，一样可以通过）

不知道能否解答一下我的问题

closed 2020-04-27 by norikohxb