coolsnowwolf/lede #4438: ssr plus无法订阅

KNulL0, 2020-04-25:
2020-04-25 14:11:06 服务正在暂停
2020-04-25 14:10:37 重启服务成功
2020-04-25 14:10:37 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:434: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:379: in function </usr/share/shadowsocksr/subscribe.lua:294>
	[C]: in function 'foreach'
	/usr/share/shadowsocksr/subscribe.lua:436: in function </usr/share/shadowsocksr/subscribe.lua:434>
2020-04-25 14:10:37 stack traceback:
2020-04-25 14:10:37 /usr/share/shadowsocksr/subscribe.lua:392: attempt to concatenate field 'server_port' (a nil value)

coolsnowwolf, 2020-04-26:
server_port , 把机场的广告关键字去掉

closed 2020-04-27 by coolsnowwolf

gdstchdr, 2020-04-29:
还是不行，版本是luci-app-ssr-plus_177-1_all