coolsnowwolf/lede #4445: WRT32X 能获取到V6，但是无法上网

haig233, 2020-04-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
WRT32X 能获取到V6，但是无法上网
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
WRT32X 能获取到V6，但是无法上网，网络连接里详细信息，看到IPv6是有IP的
[image]
### (1) 具体问题

A：看图吧
[image]
[image]
[image]

### (2) 路由器型号和固件版本

A：WRT32X

xzjue5, 2020-04-26:
[image]
把这个删了

haig233, 2020-04-26:
删了还是不行

xzjue5, 2020-04-26:
DHCP/DNS-高级设置-禁止解析 IPv6 DNS 记录  取消试试

haig233, 2020-04-26:
> DHCP/DNS-高级设置-禁止解析 IPv6 DNS 记录 取消试试

也试过了，也不行

xzjue5, 2020-04-26:
http://www.test-ipv6.com 测试一下

haig233, 2020-04-26:
> http://www.test-ipv6.com 测试一下

昨晚试了很多种方法，都不行，IP.sb测试都没有V6

xzjue5, 2020-04-26:
v6的防火墙设置了吗

haig233, 2020-04-26:
> v6的防火墙设置了吗

需要动防火墙吗
之前4月15号之前编译的固件都是可以直接用的，后面的，就一直用不了V6

xzjue5, 2020-04-26:
[image]
看看是不是这样的

zyizhuo, 2020-04-26:
问题一样。我发现一个现象，刚启动的几秒内可以ping通外网ipv6，几秒后ping显示“Network unreachable”

xzjue5, 2020-04-26:
是不是设置了负载均衡

haig233, 2020-04-26:
> 是不是设置了负载均衡

刚刚刷进去的固件，没改什么

xzjue5, 2020-04-26:
> > 是不是设置了负载均衡
> 
> 刚刚刷进去的固件，没改什么

看一下有没有负载均衡，有的话就删掉里面的配置

zyizhuo, 2020-04-26:
我用了一个去掉了一堆组件的精简版后莫名其妙没问题了

zyizhuo, 2020-04-26:
> 
> 
> 问题一样。我发现一个现象，刚启动的几秒内可以ping通外网ipv6，几秒后ping显示“Network unreachable”

我用了一个去掉了一堆组件的精简版后莫名其妙没问题了。

zyizhuo, 2020-04-26:
> > 问题一样。我发现一个现象，刚启动的几秒内可以ping通外网ipv6，几秒后ping显示“Network unreachable”
> 
> 我用了一个去掉了一堆组件的精简版后莫名其妙没问题了。

又多了一个线索：用老版本正常的IPv6备份后导入到最新版本后IPv6正常

hyird, 2020-04-27:
mwan3安装了没有 装了的话就卸载了

haig233, 2020-04-27:
> mwan3安装了没有 装了的话就卸载了

卸载了能用V6了再装回去么？我需要多拨

hyird, 2020-04-27:
多拨和v6是冤家 自己改mwan3 恩山有

haig233, 2020-04-27:
> 多拨和v6是冤家 自己改mwan3 恩山有

就好像我上面说得，我刚刚把固件刷进去，并没有动过Mwan3的设置啊，这个插件只要有都会有影响么

hcwang, 2020-04-27:
负载均衡中的“策略”，将balanced的备用成员从“不可达”改为“使用默认路由表”试试。

shikkoku, 2020-04-29:
我刚编译的X86-64版和楼主问题一样。而且我还没有编译mwan3进去。

yunmi323, 2020-05-03:
有些地方只给v6地址，不能联网。之前广东电信也这样有地址只能上电信官网其他不能上，现在好了还可以看v6线路直播。建议等等吧，工信部要求年底基本弄好。

shikkoku, 2020-05-03:
> 有些地方只给v6地址，不能联网。之前广东电信也这样有地址只能上电信官网其他不能上，现在好了还可以看v6线路直播。建议等等吧，工信部要求年底基本弄好。

但我编译原版Openwrt IPv6就没有问题，也有其他类似的issue反馈回退旧版本的代码后编译也恢复正常了。

zhouym233, 2020-05-05:
遇到类似情况。MT7621平台，未编译mwan3，正常获取dhcpv6-pd，ssh到路由去ping公网v6是通的，路由后端的设备能获取公网v6地址但ping不通。

zhouym233, 2020-05-06:
> 遇到类似情况。MT7621平台，未编译mwan3，正常获取dhcpv6-pd，ssh到路由去ping公网v6是通的，路由后端的设备能获取公网v6地址但ping不通。

https://github.com/coolsnowwolf/lede/issues/3797#issuecomment-600538499
删除重建wan6接口之后重启就好了，玄学.....

zyizhuo, 2020-05-13:
> 有些地方只给v6地址，不能联网。之前广东电信也这样有地址只能上电信官网其他不能上，现在好了还可以看v6线路直播。建议等等吧，工信部要求年底基本弄好。

刚开机的几秒内可以ping通ipv6外网...用opnsense也没问题

haig233, 2020-05-13:
> 有些地方只给v6地址，不能联网。之前广东电信也这样有地址只能上电信官网其他不能上，现在好了还可以看v6线路直播。建议等等吧，工信部要求年底基本弄好。

不存在你说的这种可能，我用电脑直连猫或者用路由器原版固件是一切正常的，能访问纯V6网站

wushangwei, 2020-06-02:
x64也是一样的问题，我发现用新版的固件，路由器端都ping不通v6，我用恩山里lusty编译的固件，5月9号的固件也不行，但是5月1号的固件就可以；5月9号的固件的更新日志里有个内核支持网桥，不知道是不是和这个有关系，因此是不是五月初某次更新影响到了ipv6

ZqinKing, 2020-07-04:
你是不是把wan6的借口删掉了？如果wan6的接口删掉了的话，防火墙就没有ipv6的放行规则，需要自行手动添加！

closed 2020-07-04 by coolsnowwolf