coolsnowwolf/lede #4448: R7800 在编译软路由成功后编译错误 v2ray-core-4.23.1/.built] Error 1

kktt007, 2020-04-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译软路由通过,没问题,然后现在切换编译R7800,出现错误,然后make clean后编译产生新的错误,如下,默认的feed.conf.default只保留了#src-git telephony https://git.openwrt.org/feed/telephony.git;openwrt-19.07,
其余的#都去掉了,

### (2) 路由器型号和固件版本

A：R7800 最新源码

### (3) 详细日志
[image]

[日志](https://pastebin.com/bH6bSHpM)

zheshifandian, 2020-04-27:
插件没下过来=卡墙上了

kktt007, 2020-04-27:
> 插件没下过来=卡墙上了

好的,谢谢,这都能看出来吗,我以为是编译x86后,重新编译R7800 ,里面东西出现紊乱呢.但是我是全局嘚

kktt007, 2020-04-27:
> 插件没下过来=卡墙上了

我看到后来大屌更新了这个https://github.com/coolsnowwolf/lede/pull/4421 这个golang是不是解决我这个的.速度太快了

closed 2020-04-27 by coolsnowwolf

kktt007, 2020-04-28:
反正还是报错，rm build tem 文件夹了，make clean了，依旧。全局的，还是有问题啊男神