coolsnowwolf/lede #4451: 因luci-app-unblockmusic导致编译失败，小白请教。

hn315, 2020-04-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
因luci-app-unblockmusic导致的编译失败，不知道原因。
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue
没有
## 2. 详细叙述
我电脑：Win10 LTSC + VirtualBox6.16+Ubuntu 18 LTS x64，虚拟机设置2核+2G内存+60G存储。按Lean官方环境和编译命令，各种折腾，编译都出错，最终我取消选择luci-app-unblockmusic后，编译通过。
### (1) 具体问题
make menuconfig默认配置，保存后，每次编译失败后，到make   download  ，会有以下信息，其他命令执行都正常。
A：
make[2]: Entering directory '/home/abc/lede/package/lean/luci-app-unblockmusic'
/home/abc/lede/include/package.mk:327: warning: overriding recipe for target 'prepare-package-install'
/home/abc/lede/include/package.mk:327: warning: ignoring old recipe for target 'prepare-package-install'
/home/abc/lede/include/package.mk:332: warning: overriding recipe for target '/home/abc/lede/bin/targets/x86/64/packages'
/home/abc/lede/include/package.mk:332: warning: ignoring old recipe for target '/home/abc/lede/bin/targets/x86/64/packages'
/home/abc/lede/include/package.mk:339: warning: overriding recipe for target 'force-clean-build'
/home/abc/lede/include/package.mk:339: warning: ignoring old recipe for target 'force-clean-build'
/home/abc/lede/include/package.mk:344: warning: overriding recipe for target 'clean'
/home/abc/lede/include/package.mk:344: warning: ignoring old recipe for target 'clean'
/home/abc/lede/include/package.mk:350: warning: overriding recipe for target 'dist'
/home/abc/lede/include/package.mk:350: warning: ignoring old recipe for target 'dist'
/home/abc/lede/include/package.mk:353: warning: overriding recipe for target 'distcheck'
/home/abc/lede/include/package.mk:353: warning: ignoring old recipe for target 'distcheck'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/abc/lede/package/lean/luci-app-unblockmusic'
time: package/lean/luci-app-unblockmusic/download#0.34#0.07#0.41
### (2) 路由器型号和固件版本
A：
X86 64位
### (3) 详细日志
编译失败日志
A：
package/Makefile:107: recipe for target '/home/abc/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/abc/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/abc/lede'
Build failed - please re-run with -j1 to see the real error message
/home/abc/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

mjyhj, 2020-04-26:
把主页依赖库更新一遍

hn315, 2020-04-26:
> 把主页依赖库更新一遍

更新了的

fengqi, 2020-05-04:
make -j8 download

ElectronicElephant, 2020-05-05:
```
Build failed - please re-run with -j1 to see the real error message
```

这个说得够明白了的吧？

closed 2020-07-04 by coolsnowwolf