coolsnowwolf/lede #4452: luci-app-dockerman 编译报错

jy3344511, 2020-04-26:
用的是ubuntu 20.04，已经编译成功后，第一次已经编译进了dockerman，后来进行第二次编译，提示WARNING: Makefile 'package/lean/luci-app-dockerman/Makefile' has a dependency on 'docker-ce', which does not exist，编译报错。

虚拟机ubuntu，有备份，使用新的ubuntu，执行到make menuconfig后，仍然提示WARNING: Makefile 'package/lean/luci-app-dockerman/Makefile' has a dependency on 'docker-ce', which does not exist，编译不了。

hyird, 2020-04-27:
我的锅,我的锅 只升级了docker-ce 依赖忘了
https://github.com/coolsnowwolf/packages/pull/58
你试试这个pr

lloyd18, 2020-04-27:
./scripts/feeds update -a
./scripts/feeds install -a

jy3344511, 2020-04-27:
> ./scripts/feeds update -a
> ./scripts/feeds install -a

系统有备份，都换全新的了，这两步怎么可能没做呢？

liwenjie119, 2020-04-28:
> > ./scripts/feeds update -a
> > ./scripts/feeds install -a
> 
> 系统有备份，都换全新的了，这两步怎么可能没做呢？

rm -rf ./tmp后再次执行试试

KFERMercer, 2020-04-28:
已经修好了, 关了吧

closed 2020-04-28 by coolsnowwolf