coolsnowwolf/lede #4457: 部分域名通过IPV6解析会解析到0.0.0.0

Cathgao, 2020-04-27:
某些域名，如`raw.githubusercontent.com`、`githubusercontent.com`通过ipv6请求路由解析时会解析到0.0.0.0，但`user-images.githubusercontent.com`又能正确解析，上述网址ipv4全部解析正确。
路由的WAN6所用DNS为阿里的`2400:3200:baba::1`，本地用阿里的ipv6DNS也解析正确。已经开启了SSR+服务并使用PDNSD查询8.8.8.8，但还是偶尔出现DNS污染的情况。
已经取消`禁止解析 IPv6 DNS 记录`的选中，本人网络特殊，需要使用IPV6.
[image]

Cathgao, 2020-04-27:
在路由终端中使用::1解析也是返回正确的结果
[image]

closed 2020-04-27 by Cathgao

geniussar, 2023-05-20:
同样的问题，试过多个方法，还搭建了两个AGH，一个国内一个国外，都不行。
偶然发现了是IPv6的DNS问题，最后在openwrt的IPv6的DNS，手动指定了google的2001:4860:4860::8888解决