coolsnowwolf/lede #4460: ath10k... 这是继续爆炸了吗。。  - -

Aes64X, 2020-04-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

竞斗云 iperf3 内网测速。。。惨不忍睹
固件，最新pull

日志：
分别使用80MHz和外挂9984 160MHz测速，结果一样。。。。各种姿势0速率+丢包

[  357.919591] ath10k_warn: 35 callbacks suppressed
[  357.919612] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.923414] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.930477] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.937779] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.945028] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.952310] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.959622] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.966910] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.974178] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  357.981503] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.328637] ath10k_warn: 28 callbacks suppressed
[  363.328658] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.332457] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.339525] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.346788] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.354043] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.361392] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.368648] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.375958] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.383195] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  363.390536] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.426941] ath10k_warn: 23 callbacks suppressed
[  368.426959] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.430763] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.437837] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.445091] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.452346] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.459692] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.466965] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.474226] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.481549] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  368.488831] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon

AmadeusGhost, 2020-04-27:
revert这个commit呢：https://github.com/coolsnowwolf/lede/commit/4787dbaf3cad44b374b660cc0fef9386963e6795

Aes64X, 2020-04-27:
> revert这个commit呢：[4787dba](https://github.com/coolsnowwolf/lede/commit/4787dbaf3cad44b374b660cc0fef9386963e6795)

一样炸。。。不过，似乎只存在于9984，IPQ40xx本身的无线倒没发现问题。

aglent, 2020-04-27:
acrh17路由器 这个下午最新的版本 5G wifi 有问题 速度几乎为0  2.4G wifi可以用
与上午编译的版本对比下 好像是ath10k-firmware-qca9984-ct 改为ath10k-firmware-qca9984了 ，但4月初和再早些版本 也是用ath10k-firmware-qca9984 就没有问题 可能还有其他问题

Aes64X, 2020-04-27:
> acrh17路由器 这个下午最新的版本 5G wifi 有问题 速度几乎为0 2.4G wifi可以用
> 与上午编译的版本对比下 好像是ath10k-firmware-qca9984-ct 改为ath10k-firmware-qca9984了 ，但4月初和再早些版本 也是用ath10k-firmware-qca9984 就没有问题 可能还有其他问题

跟CT没啥关系，我试过。。。两个都炸 - -

aglent, 2020-04-27:
> 
> 
> > acrh17路由器 这个下午最新的版本 5G wifi 有问题 速度几乎为0 2.4G wifi可以用
> > 与上午编译的版本对比下 好像是ath10k-firmware-qca9984-ct 改为ath10k-firmware-qca9984了 ，但4月初和再早些版本 也是用ath10k-firmware-qca9984 就没有问题 可能还有其他问题
> 
> 跟CT没啥关系，我试过。。。两个都炸 - -

可我上午编译的正常 不知道下午更新的哪部分有问题？

Aes64X, 2020-04-27:
> > > acrh17路由器 这个下午最新的版本 5G wifi 有问题 速度几乎为0 2.4G wifi可以用
> > > 与上午编译的版本对比下 好像是ath10k-firmware-qca9984-ct 改为ath10k-firmware-qca9984了 ，但4月初和再早些版本 也是用ath10k-firmware-qca9984 就没有问题 可能还有其他问题
> > 
> > 
> > 跟CT没啥关系，我试过。。。两个都炸 - -
> 
> 可我上午编译的正常 不知道下午更新的哪部分有问题？

我把内核切到4.14，MAC80211退回4.19再尝试一下。。。

Aes64X, 2020-04-27:
我将IPQ40xx内核切换到4.14并退回mac80211到4.19版本，目前测试一切正常，5GHz iperf3 -s 测速内网吞吐10线程最高550+带宽。

我是退回了，各位请自便 --!

closed 2020-04-29 by coolsnowwolf