coolsnowwolf/lede #4464: lean大或者hxlls大或者Promix953大有空的时候能否进来点拨一下。

norikohxb, 2020-04-27:
lean大hxlls大Promix953大我解决了问题了。我通过adguardhome找到了屏蔽的这个域名securepubads.g.doubleclick.net把它加入到adbyby的不走代理的域名里面。三星安卓5.0的平板+youtube的app就不会有问题饿了。

但是现在我还是想要请教你们这个问题，同样的lede，为什么3月21日的lede不需要特别把securepubads.g.doubleclick.net加入到adbyby白名单？（新老安州手机平板的youtube的app的视频完全不会有问题）而在4月份的lede版本就需要把securepubads.g.doubleclick.net加入到白名单方可以让老的平板上youtube的app（如果家里全部是新的手机也不需要这样操作，一样可以通过）要加入白名单仅仅是为了照顾老的手机可以看youtube的app

不知道能否提点一下这个问题，我以下猜测对不对

https://github.com/coolsnowwolf/lede/issues/4437#issue-606699231

norikohxb, 2020-04-28:
还是没有解决，老的手机，甚至三星s8都是一样。youtube会中断。切换回3月21日的lede了。现在ok了，等待以后lede或者ssrplus有新的更新吧，感觉和adbyby也没有关系。就算是不开 老的手机的youtube app还是一样不行

ElectronicElephant, 2020-04-30:
同一个问题最好不要重新开issue，你可以reopen #4437，方便大家跟踪。另外，标题最好改一下，清晰简洁地描述你遇到的问题，这样大家才有点进来的愿望。

```
甚至直接关闭adbyby 也是一样
```
我这里目前用`git-20.088.29068-cebe120`，只用酸酸乳加，在主流系统上均没有复现楼主遇到的问题。三星s6，够老了的吧。

norikohxb, 2020-04-30:
> 同一个问题最好不要重新开issue，你可以reopen #4437，方便大家跟踪。另外，标题最好改一下，清晰简洁地描述你遇到的问题，这样大家才有点进来的愿望。
> 
> ```
> 甚至直接关闭adbyby 也是一样
> ```
> 
> 我这里目前用`git-20.088.29068-cebe120`，只用酸酸乳加，在主流系统上均没有复现楼主遇到的问题。三星s6，够老了的吧。

好的，之前不懂reopen不过我已经吧第一个关闭了。能否请教一下你是只有一个lede，外面是否还有ros。三星s6和我的galaxy note 12.2感觉是同一个时代的。用的是youtube的app对吧？我是老款型号均如此，新的都没有问题。看来还需要再用排除法。

ElectronicElephant, 2020-04-30:
> 好的，之前不懂reopen不过我已经吧第一个关闭了。能否请教一下你是只有一个lede，外面是否还有ros。三星s6和我的galaxy note 12.2感觉是同一个时代的。用的是youtube的app对吧？我是老款型号均如此，新的都没有问题。看来还需要再用排除法。

我直接使用openwrt拨号。你的环境比较复杂，我个人建议采用比较常见的“最小系统法”排摸。把所有不相关的服务关闭，甚至重刷，看当系统环境比较纯净、精简时，是否仍然能够复现。如果能复现，请把环境告知大家，以便复现，否则问题不太可能解决。

norikohxb, 2020-04-30:
> 我直接使用openwrt拨号。你的环境比较复杂，我个人建议采用比较常见的“最小系统法”排摸。把所有不相关的服务关闭，甚至重刷，看当系统环境比较纯净、精简时，是否仍然能够复现。如果能复现，请把环境告知大家，以便复现，否则问题不太可能解决。

我知道，每次我搞清楚什么都是迫不及待分享的，最近就是因为老款的手机的youtube在折腾。想要用最新的lede，但是无奈家里的老平板不同意。哈哈。新的平板都没有问题，速度都很快。打开也很快。我主要和你不一样还有很依赖adbyby。我继续观察，争取最小化测试

norikohxb, 2020-04-30:
> 我直接使用openwrt拨号。你的环境比较复杂，我个人建议采用比较常见的“最小系统法”排摸。把所有不相关的服务关闭，甚至重刷，看当系统环境比较纯净、精简时，是否仍然能够复现。如果能复现，请把环境告知大家，以便复现，否则问题不太可能解决。
我就是不甘心我的老的平板无法流畅看youtube，我特别编译了一个lean大的默认版本（就是一切menuconfig都是lean大的默认标配的）结果发现没有这个问题。然后我对比我自己编译的极致精简的。我发现少了左边这些黄色的程序。应该是这些程序里面有某一个可以让我的老平板youtube访问迅速，你如果有空可以帮我看一下吗？
[image]

norikohxb, 2020-04-30:
问题找到了，在3月21日之前的lede我编译的时候都是默认不编译upnp的。但是现在最新版本的如果不编译upnp，老的安卓平板播放youtube的apk的时候就会出现点击每个视频开头几秒缓冲播放后就转圈圈了。今天应该算彻底找出真凶了，也就是upnp，其实之前可以
今天还发现一个很有趣的事情，我有一个平板，测试的时候网关是指向刚刚又精简了一次的lede，但是是带了upnp，然后我把服务器的网关从测试的ip修改为我局域网的默认那个正式ip，虽然老的平板还指向那个不存在的网关，但是很奇妙，竟然youtube还可以播放。要知道这个老的平板的网关已经不在了啊

norikohxb, 2020-04-30:
已经解决了4月份之前的lede我精简掉upnp，是不影响作为旁路由的lede的正常功能，4月份后的lede我再精简了upnp，我的老平板的安卓就无法播放youtube了，第一次加载视频缓存播放完毕就几乎不播放了。

norikohxb, 2020-04-30:
> 我直接使用openwrt拨号。你的环境比较复杂，我个人建议采用比较常见的“最小系统法”排摸。把所有不相关的服务关闭，甚至重刷，看当系统环境比较纯净、精简时，是否仍然能够复现。如果能复现，请把环境告知大家，以便复现，否则问题不太可能解决。

有空帮我分析回帖一下，我先关闭了，我确实已经解决了

closed 2020-04-30 by norikohxb