coolsnowwolf/lede #4468: (黑）群晖在openwrt里的远程唤醒问题

aepub, 2020-04-28:
在没有使用openwrt的软路由之前，一直使用的网件6300V2，刷梅林固件，路由器里面在做好ddns、固定IP和arp等之后，能够通过局域网和手机远程唤醒群晖！
但是使用软路由及openwrt后，无论怎么设置，都只能实现局域网唤醒，手机远程唤醒在群晖关机后的短时间内可以，但是群晖关机时间稍长，就无法远程唤醒，不知道是什么原因，曾经把群晖的IP地址放到里IP白名单里面，尝试让它不走代理，但也无果！

ElectronicElephant, 2020-04-30:
暂时可能没有其他人稳定复现这个问题，楼主能否提供更多信息？

blwuer, 2020-04-30:
需要ssh登录到openwrt系统中，使用命令绑定arp。

ip neigh replace （网卡IP） lladdr （网卡MAC） nud permanent dev （设备名称）

例如：需要唤醒的主机ip：192.168.1.11 mac：11:22:33:44:55:66，设备所在的网口名称：eth1，
那么使用命令：
ip neigh replace 192.168.1.11  lladdr 11:22:33:44:55:66 nud permanent dev eth1

aepub, 2020-04-30:
ssh登陆到openwrt的命令行方式与直接设置ip/mac绑定功效还不一样？试试看

aepub, 2020-04-30:
> 暂时可能没有其他人稳定复现这个问题，楼主能否提供更多信息？

这个现象有很多人遇到过，至少我在某电报群里面有三个兄弟遇到过一样的问题

aepub, 2020-05-01:
> ssh登陆到openwrt的命令行方式与直接设置ip/mac绑定功效还不一样？试试看
我仿照命令：ip neigh replace 192.168.1.11 lladdr 11:22:33:44:55:66 nud permanent dev eth1执行后，arp可以看见增加新的一行：
192.168.1.1   0x1         0x6         XX:XX:XX:XX:XX:XX     *        eth1
192.168.1.1   0x1         0x1         XX:XX:XX:XX:XX:XX     *        br-lan
（第二行是以前通过openwrt后台ip/mac绑定产生的，标记为0X1；）
这里可以看见新增行的Flags是0X6，Device是eth1；

但群晖关机，几个小时以后，
这行新增的没有了，只有以前存在的一行：
192.168.1.1   0x1         0x1         XX:XX:XX:XX:XX:XX     *        br-lan
而没有Flage为0X6的，远程唤醒是没办法实现的！

aepub, 2020-05-01:
忘记提一句了，我的是软路由+K3的AP模式！

aepub, 2020-05-21:
始终无法通过正常途径解决，最终的办法是在openwrt-系统-计划任务里面，增加：
0 7-20 * * * ip neigh add 192.168.1.99 lladdr XX:XX:XX:XX nud permanent dev br-lan >/dev/null 2>&1
0 7-20 * * * ip neigh chg 192.168.1.99 dev br-lan nud permanent >/dev/null 2>&1
192.168.1.99是群晖的ip地址

closed 2020-07-04 by coolsnowwolf