coolsnowwolf/lede #4472: Applications 里的 ss plus 看不到了么？

a1392429292, 2020-04-28:
如题，没看到有ss plus的选项了。。。

EnnawYang, 2020-04-28:
哦

hwangjunjie, 2020-04-28:
能不能直接关闭，怎么老是有人问这个问题

a1392429292, 2020-04-28:
> 能不能直接关闭，怎么老是有人问这个问题

所以啊，为何总是有人问呢？是隐藏了吗？还是就去掉了以后也不会有了？还是？？？

EnnawYang, 2020-04-28:
> > 能不能直接关闭，怎么老是有人问这个问题
> 
> 所以啊，为何总是有人问呢？是隐藏了吗？还是就去掉了以后也不会有了？还是？？？

所以能搜到的问题为什么老是有人问呢

hwangjunjie, 2020-04-28:
[image]

a1392429292, 2020-04-28:
> > > 能不能直接关闭，怎么老是有人问这个问题
> > 
> > 
> > 所以啊，为何总是有人问呢？是隐藏了吗？还是就去掉了以后也不会有了？还是？？？
> 
> 所以能搜到的问题为什么老是有人问呢

能够讲一下，为何去掉吗？如果软路由没有了这个功能，软路由意义何在？
是不更新了呢？还是开发者面临违法问题？还是怎样？以后也不会有了吗？

EnnawYang, 2020-04-28:
> > > > 能不能直接关闭，怎么老是有人问这个问题
> > > 
> > > 
> > > 所以啊，为何总是有人问呢？是隐藏了吗？还是就去掉了以后也不会有了？还是？？？
> > 
> > 
> > 所以能搜到的问题为什么老是有人问呢
> 
> 能够讲一下，为何去掉吗？如果软路由没有了这个功能，软路由意义何在？
> 是不更新了呢？还是开发者面临违法问题？还是怎样？以后也不会有了吗？

敢不敢自己动手搜一下呢，兄弟？

closed 2020-04-28 by a1392429292