coolsnowwolf/lede #4474: 刚编译的竞斗云固件，无法进行无线中继！

shenzt68, 2020-04-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法无线中继

### (2) 路由器型号和固件版本

A：p&w-128m

### (3) 详细日志

A：Tue Apr 28 18:49:22 2020 kern.info kernel: [  223.482014] device wlan0 left promiscuous mode
Tue Apr 28 18:49:22 2020 kern.info kernel: [  223.482254] br-lan: port 2(wlan0) entered disabled state
Tue Apr 28 18:49:22 2020 daemon.notice hostapd: Remove interface 'wlan0'
Tue Apr 28 18:49:22 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Tue Apr 28 18:49:22 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Tue Apr 28 18:49:22 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue Apr 28 18:49:22 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Apr 28 18:49:22 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Tue Apr 28 18:49:22 2020 daemon.info avahi-daemon[5167]: Interface wlan0.IPv6 no longer relevant for mDNS.
Tue Apr 28 18:49:22 2020 daemon.info avahi-daemon[5167]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::1e40:e8ff:fe11:a2ea.
Tue Apr 28 18:49:22 2020 kern.warn kernel: [  223.786355] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
Tue Apr 28 18:49:22 2020 daemon.info avahi-daemon[5167]: Withdrawing address record for fe80::1e40:e8ff:fe11:a2ea on wlan0.
Tue Apr 28 18:49:22 2020 daemon.notice netifd: radio0 (8098): command failed: Not supported (-95)
Tue Apr 28 18:49:23 2020 user.notice root: ip link set dev wlan0 up
Tue Apr 28 18:49:24 2020 kern.warn kernel: [  226.090056] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
Tue Apr 28 18:49:24 2020 kern.info kernel: [  226.277878] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Command failed: Request timed out
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Command failed: Not found
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Interface 0 setup failed: WPA_SUPPLICANT_FAILED
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Command failed: Invalid argument
Tue Apr 28 18:51:36 2020 daemon.notice netifd: radio0 (8795): Command failed: Invalid argument
Tue Apr 28 18:51:36 2020 daemon.notice netifd: radio0 (8828): command failed: Not supported (-95)
Tue Apr 28 18:51:42 2020 daemon.notice netifd: radio0 (8957): command failed: Not supported (-95)
Tue Apr 28 18:51:42 2020 user.notice root: ip link set dev wlan0 up
Tue Apr 28 18:51:43 2020 kern.warn kernel: [  365.330218] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
Tue Apr 28 18:51:44 2020 kern.info kernel: [  365.516764] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[image]

shenzt68, 2020-04-28:
还有一些日志：
Tue Apr 28 18:49:22 2020 kern.warn kernel: [  223.786355] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
Tue Apr 28 18:49:22 2020 daemon.info avahi-daemon[5167]: Withdrawing address record for fe80::1e40:e8ff:fe11:a2ea on wlan0.
Tue Apr 28 18:49:22 2020 daemon.notice netifd: radio0 (8098): command failed: Not supported (-95)
Tue Apr 28 18:49:23 2020 user.notice root: ip link set dev wlan0 up
Tue Apr 28 18:49:24 2020 kern.warn kernel: [  226.090056] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
Tue Apr 28 18:49:24 2020 kern.info kernel: [  226.277878] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Command failed: Request timed out
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Command failed: Not found
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Interface 0 setup failed: WPA_SUPPLICANT_FAILED
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Tue Apr 28 18:49:54 2020 daemon.notice netifd: radio0 (8098): Command failed: Invalid argument
Tue Apr 28 18:51:36 2020 daemon.notice netifd: radio0 (8795): Command failed: Invalid argument
Tue Apr 28 18:51:36 2020 daemon.notice netifd: radio0 (8828): command failed: Not supported (-95)
Tue Apr 28 18:51:42 2020 daemon.notice netifd: radio0 (8957): command failed: Not supported (-95)
Tue Apr 28 18:51:42 2020 user.notice root: ip link set dev wlan0 up
Tue Apr 28 18:51:43 2020 kern.warn kernel: [  365.330218] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
Tue Apr 28 18:51:44 2020 kern.info kernel: [  365.516764] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Tue Apr 28 18:52:14 2020 daemon.notice netifd: radio0 (8957): Command failed: Request timed out
Tue Apr 28 18:52:14 2020 daemon.notice netifd: radio0 (8957): Command failed: Not found
Tue Apr 28 18:52:14 2020 daemon.notice netifd: radio0 (8957): Interface 0 setup failed: WPA_SUPPLICANT_FAILED
Tue Apr 28 18:52:14 2020 daemon.notice netifd: radio0 (8957): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Tue Apr 28 18:52:14 2020 daemon.notice netifd: radio0 (8957): Command failed: Invalid argument
Tue Apr 28 18:55:59 2020 auth.info login[9441]: root login on 'pts/0'
Tue Apr 28 18:56:52 2020 kern.warn kernel: [  674.184763] ath10k_ahb a800000.wifi: DFS region 0x0 not supported, will trigger radar for every pulse
Tue Apr 28 18:56:52 2020 kern.warn kernel: [  674.185464] ath10k_ahb a000000.wifi: DFS region 0x0 not supported, will trigger radar for every pulse
Tue Apr 28 18:57:06 2020 daemon.notice hostapd: Remove interface 'wlan1'
Tue Apr 28 18:57:06 2020 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Tue Apr 28 18:57:06 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Tue Apr 28 18:57:06 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Tue Apr 28 18:57:06 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Tue Apr 28 18:57:06 2020 daemon.notice netifd: radio0 (9648): Command failed: Invalid argument
Tue Apr 28 18:57:06 2020 kern.info kernel: [  687.956170] device wlan1 left promiscuous mode
Tue Apr 28 18:57:06 2020 kern.info kernel: [  687.956400] br-lan: port 3(wlan1) entered disabled state
Tue Apr 28 18:57:06 2020 daemon.notice netifd: Network device 'wlan1' link is down
Tue Apr 28 18:57:06 2020 kern.warn kernel: [  688.019361] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 1
Tue Apr 28 18:57:06 2020 daemon.info avahi-daemon[5167]: Interface wlan1.IPv6 no longer relevant for mDNS.
Tue Apr 28 18:57:06 2020 daemon.info avahi-daemon[5167]: Leaving mDNS multicast group on interface wlan1.IPv6 with address fe80::1e40:e8ff:fe11:a2eb.
Tue Apr 28 18:57:06 2020 daemon.info avahi-daemon[5167]: Withdrawing address record for fe80::1e40:e8ff:fe11:a2eb on wlan1.
Tue Apr 28 18:57:06 2020 daemon.notice netifd: radio0 (9688): command failed: Not supported (-95)
Tue Apr 28 18:57:07 2020 daemon.notice netifd: radio1 (9744): command failed: Not supported (-95)
Tue Apr 28 18:57:07 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Tue Apr 28 18:57:08 2020 kern.warn kernel: [  690.351234] ath10k_ahb a800000.wifi: unsupported HTC service id: 1536
Tue Apr 28 18:57:09 2020 kern.info kernel: [  690.476842] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready

AmadeusGhost, 2020-04-29:
可以并入我的commit：https://github.com/AmadeusGhost/lede/commit/500a975aa8a5c635dca01930ca72d836b00a5742 再编译试试能不能中继

waynon, 2020-05-01:
开两个一样的issue有意思吗？#4503 也是你开的issue噢

PR #4518 (2020-05-02): netifd: add dynamic wireless reconfiguration
Set new option 'reconf' in 'wifi-device' section to enable dynamic re-configuration on that radio. Also fix wifi relay and 'netifd: radio1 (9654): Command failed'.
Run-tested-by: @hyird 
fix #4319 , fix #4390 , fix #4474 , fix #4503 , fix #4516
closed 2020-05-02 by coolsnowwolf