coolsnowwolf/lede #4475: 大雕哥，编译的ssr翻墙出不去，不知道缺了哪个插件，插件列表奉上

zheshifandian, 2020-04-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ssr检查谷歌连通性失败，确定订阅没问题，不知道少了什么插件

### (2) 路由器型号和固件版本

A：r619ac

### (3) 详细日志

A：
6in4 - 25-1
UnblockNeteaseMusic - 0.24.1-4
UnblockNeteaseMusicGo - 0.2.0-1
adbyby - 2.7-20200315
ath10k-firmware-qca4019-ct-htt - 2019-10-03-d622d160-1
automount - 1-31
base-files - 200-r2491-11848e8b
block-mount - 2019-08-28-6a61b9ac-3
boost - 1.71.0-4
boost-date_time - 1.71.0-4
boost-program_options - 1.71.0-4
boost-system - 1.71.0-4
busybox - 1.30.0-1
ca-bundle - 20180409-3
ca-certificates - 20180409-3
coreutils - 8.23-3
coreutils-base64 - 8.23-3
default-settings - 1.1-57
dns2socks - 2.1-20200218
dnsmasq-full - 2.80-12
dropbear - 2019.78-3
firewall - 2019-11-22-8174814a-1
fstools - 2019-08-28-6a61b9ac-3
fwtool - 1
getrandom - 2019-06-16-4df34a4d-2
hostapd-common - 2019-08-08-ca8c2bd2-9
htop - 2.2.0-1
ip-full - 5.0.0-1
ip6tables - 1.8.4-1
ip6tables-extra - 1.8.4-1
ip6tables-mod-nat - 1.8.4-1
ipq-wifi-p2w_r619ac - 1
ipset - 7.1-3
ipt2socks - 1.0.2-1
iptables - 1.8.4-1
iptables-mod-fullconenat - 2019-10-21-0cf3b48f-2
iptables-mod-tproxy - 1.8.4-1
iputils-ping - 20101006-1
iputils-ping6 - 20101006-1
iputils-tracepath - 20101006-1
iputils-tracepath6 - 20101006-1
iputils-traceroute6 - 20101006-1
ipv6helper - 1.0-1
iw - 5.4-1
jshn - 2019-06-16-ecf56174-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 4.19.115-1-33d8f24f34bac3c3ffc87bdddb6291b2
kmod-asn1-decoder - 4.19.115-1
kmod-ath - 4.19.115+5.4.27-1-3
kmod-ath10k-ct - 4.19.115+2020-03-25-3d173a47-1
kmod-cfg80211 - 4.19.115+5.4.27-1-3
kmod-crypto-aead - 4.19.115-1
kmod-crypto-arc4 - 4.19.115-1
kmod-crypto-authenc - 4.19.115-1
kmod-crypto-crc32c - 4.19.115-1
kmod-crypto-ecb - 4.19.115-1
kmod-crypto-hash - 4.19.115-1
kmod-crypto-manager - 4.19.115-1
kmod-crypto-null - 4.19.115-1
kmod-crypto-pcompress - 4.19.115-1
kmod-crypto-sha1 - 4.19.115-1
kmod-crypto-user - 4.19.115-1
kmod-cryptodev - 4.19.115+1.10-ipq40xx-1
kmod-fs-exfat - 4.19.115+2018-04-17-01c30ad5-2
kmod-fs-ext4 - 4.19.115-1
kmod-fs-vfat - 4.19.115-1
kmod-fuse - 4.19.115-1
kmod-gpio-button-hotplug - 4.19.115-2
kmod-hwmon-core - 4.19.115-1
kmod-ip6tables - 4.19.115-1
kmod-ip6tables-extra - 4.19.115-1
kmod-ipt-conntrack - 4.19.115-1
kmod-ipt-core - 4.19.115-1
kmod-ipt-fullconenat - 4.19.115+2019-10-21-0cf3b48f-2
kmod-ipt-ipset - 4.19.115-1
kmod-ipt-nat - 4.19.115-1
kmod-ipt-nat-extra - 4.19.115-1
kmod-ipt-nat6 - 4.19.115-1
kmod-ipt-offload - 4.19.115-1
kmod-ipt-raw - 4.19.115-1
kmod-ipt-raw6 - 4.19.115-1
kmod-ipt-tproxy - 4.19.115-1
kmod-iptunnel - 4.19.115-1
kmod-iptunnel4 - 4.19.115-1
kmod-leds-gpio - 4.19.115-1
kmod-lib-crc-ccitt - 4.19.115-1
kmod-lib-crc16 - 4.19.115-1
kmod-lib-textsearch - 4.19.115-1
kmod-mac80211 - 4.19.115+5.4.27-1-3
kmod-macvlan - 4.19.115-1
kmod-mppe - 4.19.115-1
kmod-nf-conntrack - 4.19.115-1
kmod-nf-conntrack6 - 4.19.115-1
kmod-nf-flow - 4.19.115-1
kmod-nf-ipt - 4.19.115-1
kmod-nf-ipt6 - 4.19.115-1
kmod-nf-nat - 4.19.115-1
kmod-nf-nat6 - 4.19.115-1
kmod-nf-nathelper - 4.19.115-1
kmod-nf-nathelper-extra - 4.19.115-1
kmod-nf-reject - 4.19.115-1
kmod-nf-reject6 - 4.19.115-1
kmod-nfnetlink - 4.19.115-1
kmod-nls-base - 4.19.115-1
kmod-nls-cp437 - 4.19.115-1
kmod-nls-iso8859-1 - 4.19.115-1
kmod-nls-utf8 - 4.19.115-1
kmod-ppp - 4.19.115-1
kmod-pppoe - 4.19.115-1
kmod-pppox - 4.19.115-1
kmod-scsi-core - 4.19.115-1
kmod-sit - 4.19.115-1
kmod-slhc - 4.19.115-1
kmod-tcp-bbr - 4.19.115-1
kmod-usb-core - 4.19.115-1
kmod-usb-dwc3 - 4.19.115-1
kmod-usb-dwc3-qcom - 4.19.115-1
kmod-usb-storage - 4.19.115-1
kmod-usb-storage-extras - 4.19.115-1
kmod-usb-storage-uas - 4.19.115-1
kmod-usb3 - 4.19.115-1
libatomic - 7.5.0-1
libblobmsg-json - 2019-06-16-ecf56174-1
libc - 1.1.24-1
libcares - 1.14.0-1
libelf1 - 0.177-2
libev - 4.22-1
libevent2-7 - 2.1.11-1
libgcc - 7.5.0-1
libhttp-parser - 2.9.2-2
libip4tc2 - 1.8.4-1
libip6tc2 - 1.8.4-1
libipset13 - 7.1-3
libiwinfo-lua - 2020-03-22-9f5a7c4f-1
libiwinfo20200105 - 2020-03-22-9f5a7c4f-1
libjson-c - 0.12.1-2
libjson-script - 2019-06-16-ecf56174-1
liblua5.1.5 - 5.1.5-7
libmbedtls12 - 2.16.6-1
libmnl0 - 1.0.4-2
libncurses6 - 6.1-5
libnet-1.2.x - 1.2-rc3-4
libnghttp2-14 - 1.40.0-1
libnl-tiny - 2019-10-29-0219008c-1
libopenssl-conf - 1.1.1g-1
libopenssl1.1 - 1.1.1g-1
libpcap - 1.9.0-1
libpcre - 8.41-2
libpthread - 1.1.24-1
librt - 1.1.24-1
libsodium - 1.0.16-1
libstdcpp - 7.5.0-1
libubox20170601 - 2019-06-16-ecf56174-1
libubus-lua - 2018-10-06-221ce7e7-1
libubus20191227 - 2018-10-06-221ce7e7-1
libuci - 2018-08-11-4c8b4d6e-1
libuci-lua - 2018-08-11-4c8b4d6e-1
libuclient20160123 - 2018-11-24-3ba74ebc-1
libustream-openssl20150806 - 2020-01-05-30cebb4f-1
libuuid - 2.33-1
libuv1 - 1.34.2-1
libxtables12 - 1.8.4-1
logd - 2019-06-16-4df34a4d-2
lua - 5.1.5-7
luci - git-20.117.60969-420c61a-1
luci-app-adbyby-plus - 2.0-71
luci-app-autoreboot - 1.0-8
luci-app-cpufreq - 1-5
luci-app-filetransfer - 1-2
luci-app-firewall - git-20.117.60969-420c61a-1
luci-app-flowoffload - 1.0-15
luci-app-netdata - 1.0-3
luci-app-ramfree - 1.0-1
luci-app-ssr-plus - 177-1
luci-app-unblockmusic - 2.3.1-42
luci-app-vlmcsd - 1.0-5
luci-app-xlnetacc - 1.0.3-9
luci-base - git-20.117.60969-420c61a-1
luci-i18n-adbyby-plus-zh-cn - 2.0-71
luci-i18n-autoreboot-zh-cn - 1.0-8
luci-i18n-base-zh-cn - git-20.117.60969-420c61a-1
luci-i18n-cpufreq-zh-cn - 1-5
luci-i18n-filetransfer-zh-cn - 1-2
luci-i18n-firewall-zh-cn - git-20.117.60969-420c61a-1
luci-i18n-flowoffload-zh-cn - 1.0-15
luci-i18n-netdata-zh-cn - 1.0-3
luci-i18n-ramfree-zh-cn - 1.0-1
luci-i18n-ssr-plus-zh-cn - 177-1
luci-i18n-unblockmusic-zh-cn - 2.3.1-42
luci-i18n-vlmcsd-zh-cn - 1.0-5
luci-lib-fs - 1.0-1
luci-lib-ip - git-20.117.60969-420c61a-1
luci-lib-jsonc - git-20.117.60969-420c61a-1
luci-lib-nixio - git-20.117.60969-420c61a-1
luci-mod-admin-full - git-20.117.60969-420c61a-1
luci-proto-ipv6 - git-20.117.60969-420c61a-1
luci-proto-ppp - git-20.117.60969-420c61a-1
luci-ssl-openssl - git-20.117.60969-420c61a-1
luci-theme-bootstrap - git-20.117.60969-420c61a-1
microsocks - 1.0-1
mtd - 23
netdata - 1.20.0-1
netifd - 2019-08-05-5e02f944-1
node - v12.15.0-1
ntfs-3g - 2017.3.23-1-fuseint
odhcp6c - 2019-01-11-e199804b-16
odhcpd-ipv6only - 2020-01-14-6db312a6-3
openssh-sftp-server - 8.0p1-2
openssl-util - 1.1.1g-1
opkg - 2020-01-25-c09fe209-1
pdnsd-alt - 1.2.9b-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d
ppp - 2.4.8-5
ppp-mod-pppoe - 2.4.8-5
procd - 2020-03-13-77a6782d-2
redsocks2 - 0.67-3
rpcd - 2020-01-05-efe51f41-2
rpcd-mod-rrdns - 20170710
shadowsocks-libev-ss-local - 3.3.4-1
shadowsocks-libev-ss-redir - 3.3.4-1
shadowsocksr-libev-alt - 2.5.6-5
shadowsocksr-libev-server - 2.5.6-5
shadowsocksr-libev-ssr-local - 2.5.6-5
shellsync - 0.2-2
simple-obfs - 0.0.5-5
swconfig - 12
tcpping - 0.2-d890cc1bd8e3951390ceeff1ccb092a5d802850c
terminfo - 6.1-5
trojan - 1.15.1-1
ubi-utils - 2.0.2-1
ubox - 2019-06-16-4df34a4d-2
ubus - 2018-10-06-221ce7e7-1
ubusd - 2018-10-06-221ce7e7-1
uci - 2018-08-11-4c8b4d6e-1
uclient-fetch - 2018-11-24-3ba74ebc-1
uhttpd - 2019-08-17-6b03f960-4
uhttpd-mod-ubus - 2019-08-17-6b03f960-4
urandom-seed - 1.0-1
urngd - 2020-01-21-c7f7b6b6-1
v2ray - 4.23.1-1
v2ray-plugin - 1.3.0-2
vlmcsd - svn1112-1
wget - 1.20.3-3
wireless-regdb - 2017-10-20-4343d359
wpad - 2019-08-08-ca8c2bd2-9
zlib - 1.2.11-2

zheshifandian, 2020-04-28:
谷歌连通失败，反正就是处于报废状态，开了还不如不开(开了连gayhub都打不开)

dgulan, 2020-04-29:
奇怪的很：自从本次编译之后，V2就再也连接不上。
打开s s r之后，github打不开；关了S S R，github就可以打开。

更奇怪的是：我用winscp可以登录v p s 服务器，但是windows 端的v 2 ray N一样登陆不上。

zheshifandian, 2020-04-29:
> 奇怪的很：自从本次编译之后，V2就再也连接不上。
> 打开s s r之后，github打不开；关了S S R，github就可以打开。
> 
> 更奇怪的是：我用winscp可以登录v p s 服务器，但是windows 端的v 2 ray N一样登陆不上。
> 
> 因为windows 版本 v 2 ray N客户端一样无法连接，我同样怀疑是不是v2被 土蔷 了？
> winscp可以连接此IP的。

我用大雕的潘多拉，ssr没问题，op就gg

closed 2020-04-29 by coolsnowwolf