coolsnowwolf/lede #4476: x86固件[esxi]open-vm-tools内存占用持续增长

host2318, 2020-04-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

 没有类似的issue

## 2. 详细叙述
esxi平台版本：6.7.0 Update 2 (Build 13473784)

### (1) 具体问题

vmtoolsd 11.0.5-1 内存占用持续增长，重启该进程内存得以恢复；替换到旧版本10.3.0-1即可正常

 2592     1 root     S     254m  52%   0% /bin/vmtoolsd

[image]

### (2) 路由器型号和固件版本

A：OpenWrt R20.4.8 / LuCI Master (git-20.088.29068-cebe120)

### (3) 详细日志
已经从4月初观察至今，100%重现

[image]
[image]

fonlan, 2020-04-29:
我也遇到了，手动重启下vmtoolsd服务能暂时释放内存，但这应该是vmtools本身的问题，在lean这儿提没用

host2318, 2020-04-29:
> 我也遇到了，手动重启下vmtoolsd服务能暂时释放内存，但这应该是vmtools本身的问题，在lean这儿提没用
收到，谢谢回复。

ElectronicElephant, 2020-04-30:
去了解一下esxi或者kvm平台的内存balloon机制吧。这是完全正常的，可以把寄生系统暂时不需要的内存释放给宿主系统。

fonlan, 2020-04-30:
> 去了解一下esxi或者kvm平台的内存balloon机制吧。这是完全正常的，可以把寄生系统暂时不需要的内存释放给宿主系统。

这样啊，那应该影响不大，就是不太好看而已

hxlls, 2020-05-02:
VMTOOL本来就是处理资源膨胀机制的一个东西，你划分1G内存，，实际只用到512M，那VMTOOL 就会把剩余是占用，占用部分会返还给宿主使用

woaihsw, 2020-05-08:
> 去了解一下esxi或者kvm平台的内存balloon机制吧。这是完全正常的，可以把寄生系统暂时不需要的内存释放给宿主系统。

不对吧...

```
[root@localhost:~] vsish -e get /config/Mem/intOpts/CtlMaxPercent
Vmkernel Config Option {
   Default value:65
   Min value:0
   Max value:99
   Current value:65
   hidden config option:0
   Description:vmmemctl limit as percentage of VM max size
}
```

默认配置下是最多回收65%的内存, 那么问题来了, 我分了1G过去, 一礼拜后都能被吃得剩下60M...

这怎么也不像是被限制到65%啊...

ElectronicElephant, 2020-05-10:
> 不对吧...
>
> ```
> [root@localhost:~] vsish -e get /config/Mem/intOpts/CtlMaxPercent
> Vmkernel Config Option {
>    Default value:65
>    Min value:0
>    Max value:99
>    Current value:65
>    hidden config [...]

https://nerdblurt.com/vmware-memory-ballooning-sched-mem-maxmemctl-versus-mem-ctlmaxpercent/
这篇比较详细，可见它是由两个参数共同控制的

host2318, 2020-05-13:
> > > 去了解一下esxi或者kvm平台的内存balloon机制吧。这是完全正常的，可以把寄生系统暂时不需要的内存释放给宿主系统。
> > 
> > 
> > 不对吧...
> > ```
> > [root@localhost:~] vsish -e get /config/Mem/intOpts/CtlMaxPercent
> > Vmkernel Config Option {
> >    Default value:65
> >    Min value:0
> >    Max value:99
> >    Current value:65
> >    hidden config option:0
> >    Description:vmmemctl limit as percentage of VM max size
> > }
> > ```
> > 
> > 
> > 默认配置下是最多回收65%的内存, 那么问题来了, 我分了1G过去, 一礼拜后都能被吃得剩下60M...
> > 这怎么也不像是被限制到65%啊...
> 
> https://nerdblurt.com/vmware-memory-ballooning-sched-mem-maxmemctl-versus-mem-ctlmaxpercent/
> 这篇比较详细，可见它是由两个参数共同控制的
> 
> 比较迷惑，但应该不会影响使用

做了了个调整，现在内存占用不会持续下降了。 调整内容：设置esxi内 “Mem.IdleTax”参数值为40
但是同时升级到了OpenWrt R20.5.9 / LuCI Master (git-20.117.60969-420c61a)版本，不确定升级是否有优化，也懒得再尝试了

host2318, 2020-05-15:
> 做了了个调整，现在内存占用不会持续下降了。 调整内容：设置esxi内 “Mem.IdleTax”参数值为40
> 但是同时升级到了OpenWrt R20.5.9 / LuCI Master (git-20.117.60969-420c61a)版本，不确定升级是否有优化，也懒得再尝试了

刚回到家，明显觉得连接速度变慢，登陆路由器，内存已经又降低到80MB，昨晚此时还是330+空闲
[image]
[image]
[image]

closed 2020-07-04 by coolsnowwolf