coolsnowwolf/lede #4477: 这几天的固件编译出来ssr-plus不见了

winddriver, 2020-04-29:
这几天的固件编译出来ssr-plus不见了，配置文件里肯定是开启了ssr-plus的
[image]
[config.zip](https://github.com/coolsnowwolf/lede/files/4549368/config.zip)

winddriver, 2020-04-29:
[image]
这是固件服务那一栏，里面没有ssr-plus

EnnawYang, 2020-04-29:
非issue 请关闭

winddriver, 2020-04-29:
> 非issue 请关闭

这如果不叫问题，请问什么叫问题？

a1392429292, 2020-04-29:
> > 非issue 请关闭
> 
> 这如果不叫问题，请问什么叫问题？

这里，总是有那么几个人，回答很及时，但就是不正面回答你的问题，无非就是  非issue，搜以往issue。。。。。反正就是不直接回答你的问题，与其有时间答非所问，不如一针见血的回答。

因为 ssr-plus 现在隐藏掉了，详细可以看 
https://github.com/coolsnowwolf/lede/issues/4410#issuecomment-618436351

winddriver, 2020-04-29:
> > > 非issue 请关闭
> > 
> > 
> > 这如果不叫问题，请问什么叫问题？
> 
> 这里，总是有那么几个人，回答很及时，但就是不正面回答你的问题，无非就是 非issue，搜以往issue。。。。。反正就是不直接回答你的问题，与其有时间答非所问，不如一针见血的回答。
> 
> 因为 ssr-plus 现在隐藏掉了，详细可以看
> [#4410 (comment)](https://github.com/coolsnowwolf/lede/issues/4410#issuecomment-618436351)

感谢解惑，我去试试

closed 2020-04-29 by winddriver