coolsnowwolf/lede #4479: 4.29版本看油管拖动时跑飞，自动重启，必然重现。

GreatMichaelFu, 2020-04-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：测试条件及问题如下（不清楚是否以下条件与issue的必然条件，仅仅客观描述）
1.路由端使能ipsec vpn, 安卓10.0手机，使用4G+ipsec vpn连入路由内网
2.使能SSR科学上网，科学上网配置搬瓦工vps。使能DNSForwarder插件，添加解析策略文件，国内地址走国内dns，国外地址走境外DNS
3. 手机ipsec连入内网后，打开youtube应用，搜索8K视频，点击播放任意一个（调到最高1080p), 进行拖动。 拖动后，软路由自动重启，固件跑飞。

### (2) 路由器型号和固件版本

A：OpenWrt SNAPSHOT r2491-11848e8b / LuCI Master (git-20.117.60969-420c61a) 

### (3) 详细日志

A：无，固件自动重启，重启后就是内核日志及网络和配置初始化日志，无意义，未见lastlog,未见segment fault等有帮助debug的信息。

GreatMichaelFu, 2020-04-29:
如需要日志，请告知如何搜集

Sheep-12, 2020-05-02:
ipsec网速突然上去就会重启,我记得我之前只要一下载文件就会导致路由器重启。（之前=半年前）

hcym, 2020-05-02:
8k没拖过，路由器还没咋样电脑cpu到顶了，4k TV经常快进倒是正常，拖不动基本是网络限制

yunmi323, 2020-05-02:
建议把系统日志写到挂载点上，系统里可以设置，这样自动重启还保留之前日志

LitingLin, 2020-05-07:
先确定是不是机器有负载高就重启的问题，供电电路或散热有问题也会这样

closed 2020-07-04 by coolsnowwolf