coolsnowwolf/lede #4482: 编译时多个插件提示Warning

hoticebeijing, 2020-04-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
在全新的系统下，按照说明，一步一步操作，最终，多个luci-app提示Warning找不到依赖库。

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
使用的阿里云香港服务器，Ubuntu 18 64位
sudo apt-get update
sudo apt-get upgrade
并且按要求，安装 了所有库文件，首次编译。
执行到这里时，
./scripts/feeds update -a

多个地方提示了依赖无法找到的警告。

### (1) 具体问题

A：多个地方提示了依赖无法找到的警告。

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'kmod-fs-exfat', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'ntfs-3g', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/ipt2socks/Makefile' has a build dependency on 'libuv', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/lean/ksmbd-tools/Makefile' has a dependency on 'avahi-daemon', which does not exist
WARNING: Makefile 'package/lean/ksmbd-tools/Makefile' has a build dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/libidn2/Makefile' has a dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not e                                           xist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exi                                           st
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exis                                           t
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exis                                           t
[... 111 lines cut ...]
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'jansson', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libarchive', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'acl', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'attr', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'libtasn1/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'rpcsvc-proto/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'python3/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'perl/host', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a build dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/tcpping/Makefile' has a dependency on 'libnet-1.2.x', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-system', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray-plugin/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

EnnawYang, 2020-04-29:
此处警告无视即可

gold2wave, 2020-04-30:
同样的问题

lxwq, 2020-04-30:
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

closed 2020-04-30 by coolsnowwolf