coolsnowwolf/lede #4485: 能把luci-app-frpc换成https://github.com/kuoruan/luci-app-frpc/吗？

ardanzhu, 2020-04-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [*] 没有类似的issue

## 2. 详细叙述

package里自带的luci frpc比较单一，能否更换成https://github.com/kuoruan/luci-app-frpc/的？
这样编译时可以减少一下自行替换的工夫，谢谢

lxwq, 2020-04-30:
1手动删除/package/lean/luci-app-frpc
2编辑feeds.config.default
添加src-git luci-app-frpc https://github.com/kuoruan/luci-app-frpc
3./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
以上供参考

ardanzhu, 2020-04-30:
> 1手动删除/package/lean/luci-app-frpc
> 2编辑feeds.config.default
> 添加src-git luci-app-frpc https://github.com/kuoruan/luci-app-frpc
> 3
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> 以上供参考

这是自行手工更新。我说的是更新lean仓库里的。

closed 2020-04-30 by ardanzhu