coolsnowwolf/lede #4493: 如何让luci概况显示AMD系CPU温度

PilotSnail, 2020-04-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：貌似所有AMD的CPU均无法显示工作时温度，请问是否可以通过修改配置文件捕获温度感应器数据呢？

### (2) 路由器型号和固件版本

A：HP 705 G2 MINI ELITEDESK 
AMD PRO A6-8500B R5, 6 Compute Cores 2C+4G : 2 Core 2 Thread
OpenWrt R20.4.8 IPv4 5.1[2020] Compiled by eSir / LuCI Master (git-20.117.60969-420c61a)

### (3) 详细日志

A：

PilotSnail, 2020-04-30:
[image]

guyezi, 2020-05-01:
opkg install lm-sensors lm-sensors-detect

`<%#
 Copyright 2008 Steven Barth <steven@midlink.org>
 Copyright 2008-2011 Jo-Philipp Wich <jow@openwrt.org>
 Licensed to the public under the Apache License 2.0.
-%>

<%
	local fs = require "nixio.fs"
	local util = require "luci.util"
	local stat = require "luci.tools.status"
	local ver = require "luci.version"

	local has_ipv6 = fs.access("/usr/sbin/ip6tables")
	local has_dhcp = fs.access("/etc/config/dhcp")
	local has_wifi = ((fs.stat("/etc/config/wireless", "size") or 0) > 0)

	local sysinfo = luci.util.ubus("system", "info") or { }
	local boardinfo = luci.util.ubus("system", "board") or { }
	local unameinfo = nixio.uname() or { }
	local temperinfo = luci.util.exec("sensors  | grep -E 'temp1|temp2|Package|Core' | sed 's/temp2/<br>temp2/g' | sed 's/Package/<br>Package/g' | sed 's/Core/<br>Core/g' ")

	local meminfo = sysinfo.memory or {
		total = 0,
		free = 0,
		buffered = 0,
		shared = 0
	}
	
	local mem_cached = luci.sys.exec("sed -e '/^Cached: /!d; s#Cached: *##; s# kB##g' /proc/meminfo")

	local swapinfo = sysinfo.swap or {
		total = 0,
		free = 0
	}

	local has_dsl = fs.access("/etc/init.d/dsl_control")

	if luci.http.formvalue("status") == "1" then
		local ntm = require "luci.model.network".init()
		local wan = ntm:get_wannet()
		local wan6 = ntm:get_wan6net()

		local conn_count = tonumber(
			fs.readfile("/proc/sys/net/netfilter/nf_conntrack_count") or "") or 0

		local conn_max = tonumber(luci.sys.exec(
			"sysctl -n -e net.nf_conntrack_max net.ipv4.netfilter.ip_conntrack_max"
		):match("%d+")) or 4096

		local cpu_info = luci.sys.exec("cpuinfo")

		local eth_info = luci.sys.exec("ethinfo")

		local user_info = luci.sys.exec("cat /proc/net/arp | grep '0x2' | wc -l") or 0
		
		local cpu_usage = (luci.sys.exec("expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')") or "6") .. "%"

		local rv = {
			cpuusage    = cpu_usage,
			cpuinfo    = cpu_info,
			ethinfo    = eth_info,
			userinfo    = user_info,
			uptime     = sysinfo.uptime or 0,
			localtime  = os.date(),
			loadavg    = sysinfo.load or { 0, 0, 0 },
			memory     = meminfo,
			memcached  = mem_cached,
			swap       = swapinfo,
[... 759 lines cut ...]
<fieldset class="cbi-section">
       <legend><%:DSL%></legend>
       <table width="100%" cellspacing="10">
               <tr><td width="33%" style="vertical-align:top"><%:DSL Status%></td><td>
                       <table><tr>
                               <td id="dsl_i" style="width:16px; text-align:center; padding:3px">[image]/icons/ethernet_disabled.png" /><br /><small>?</small></td>
                               <td id="dsl_s" style="vertical-align:middle; padding: 3px"><em><%:Collecting data...%></em></td>
                       </tr></table>
               </td></tr>
       </table>
</fieldset>
<% end %>

<% if has_wifi then %>
<fieldset class="cbi-section">
	<legend><%:Wireless%></legend>

	<table id="wifi_status_table" width="100%" cellspacing="10">
		<tr><td><em><%:Collecting data...%></em></td></tr>
	</table>
</fieldset>
<% end %>

<%+footer%>
`

PilotSnail, 2020-05-01:
谢谢解答，不过我用了vim命令，报错了
[image]

PilotSnail, 2020-05-04:
解决了，感谢
[image]

closed 2020-05-04 by PilotSnail