coolsnowwolf/lede #4494: 精简固件一族进来看看吧，upnp这个luci插件不要精简了。旁路由也要留着

norikohxb, 2020-04-30:
精简固件一族进来看看吧，upnp这个luci插件不要精简了。
我自从学会了编译lede以后就喜欢精简自己的lede多次给lean大批评教育，小白不要胡乱精简
也确实精简的固件每次在lean大有重大改动的时候受害最深。每次都要重新调校
然后现在发现了一个有趣的事实也要告诉精简固件一族
在4月份之前的固件我都是用精简掉upnp的固件。因为我好像之前也发帖过了。好像大家都赞成作为旁路由根本就不需要upnp所以去年到3月份，我几乎都不编译upnp进去的。
也相安无事用到现在。
在4月后可能lean大重新改了内核之类的。具体高深技术我肯定是不懂，因为我没有编译upnp进去。导致了我家里的安卓8.0和以前的平板的youtube都无法播放，点开一个视频，进入视频以后前面几秒的视频缓存播放完毕就转圈圈。当然最新的安卓10没有这个问题。
也在论坛多次发帖求助无果。从4月1日到今天一直折磨自己，因为又想要用最新的lede，家里的老平板在刷牙漱口洗澡的时候还想要看youtube。所以很冲突。今天终于解决了。
[image]

我不知道lean大或者其他大神会不会来看这个帖子，如果会，我截图里面的黄色的哪个服务也不建议精简掉（精简掉会影响性能的，能麻烦告诉我吗。因为在这之前我其实都是阉割掉的）

norikohxb, 2020-04-30:
而且我还发现一个神奇的现象，比如你的平板的网关是某个ip，在你播放youtube的时候，你悄悄地把网关修改为另外一个ip，也就是平板的网关已经彻底没了，都不存在网关，然后你的youtube播放其他视频不影响。一样播放？估计lean大在4月以后的内核确实和3月份不一样。搞得这个upnp飞开不可，如果有懂得技术的进来指教一下吧。不胜感激

badcrazy, 2020-04-30:
就算没有旁路由upnp也是需要的啊，各类下载软件都需要upnp，除非你愿意一个个手动开端口

badcrazy, 2020-04-30:
只要你有nat就需要upnp，旁路由也要nat，除非你愿意一个个手动开端口

norikohxb, 2020-04-30:
> 只要你有nat就需要upnp，旁路由也要nat，除非你愿意一个个手动开端口

可是旁路由的lede其实是一个路由啊，纯粹的路由，应该不是NAT，NAT是叫做端口地址映射吧。还需要吧源IP给转化成为LEDE的本地的IP，然后经过端口号转换，把上网的电脑统统伪装成为LEDE所在的那个LAN的ip，难道不是吗？既然是在同一个内网。那么其实上网的电脑所在的网段和ROS（一级路由）也是在同一个网段，彼此之间不是都是相互可见，都在同一个255.255.255.0的网段里面，这个时候根本不需要端口地址转换吧？ 我一直以为  路由+（ip和端口的伪装修改）=NAT，而旁路由就是一个彻彻底底不折不扣的路由器嘛。

guyezi, 2020-05-01:
UPNP这功能只是端口映射,对于旁路由不做任何外网访问内网,根本就不需要这插件功能

hxlls, 2020-05-01:
请正确理解网上所传说的“旁”路由和真正的旁路由(bypass)的区别，还有NAT 的意思。你就会知道为什么了。只有正确理解这些东西，才能避免小白问题

norikohxb, 2020-05-01:
> 请正确理解网上所传说的“旁”路由和真正的旁路由(bypass)的区别，还有NAT 的意思。你就会知道为什么了。只有正确理解这些东西，才能避免小白问题

hxlls非常感谢你终于出现了。是的。首先确实是小白，就是希望您指导一下，还有我在3月28日之前的编译真的是都不带upnp，家里的电脑平板好像使用也都还ok。如果有空的时候能否简要的指导一下。

hxlls, 2020-05-01:
> > 请正确理解网上所传说的“旁”路由和真正的旁路由(bypass)的区别，还有NAT 的意思。你就会知道为什么了。只有正确理解这些东西，才能避免小白问题
> 
> hxlls非常感谢你终于出现了。是的。首先确实是小白，就是希望您指导一下，还有我在3月28日之前的编译真的是都不带upnp，家里的电脑平板好像使用也都还ok。如果有空的时候能否简要的指导一下。

我从来不用精简，也不打算创造一个精简的环境去帮别人解决问题，因为我觉得精简一点意义都没，软件不用会休眠，生成的镜像容量是在编译时候设置的，已经固定了大小，我不缺几百K的空间或者几M的磁盘，大部分软件在不开启不会使用资源，除了固件的磁盘以外，我也没有那种强迫症，多一个菜单或者少一个菜单什么的，对我没啥意义，所以我都是保证最简单使用和不会影响兼容性情况下，集成部分日常软件

norikohxb, 2020-05-02:
> 请正确理解网上所传说的“旁”路由和真正的旁路由(bypass)的区别，还有NAT 的意思。你就会知道为什么了。只有正确理解这些东西，才能避免小白问题

其实应该是说没有在乎那一点点系统的消耗，你没有编译进去的程序就不会运行了，左边是lean大的默认标配的，右边是我精简掉的。黄色的是我没有运行的程序，这些程序不在内存运行甚至都没有执行是不是会让软路由更快一些？

badcrazy, 2020-05-02:
> > 请正确理解网上所传说的“旁”路由和真正的旁路由(bypass)的区别，还有NAT 的意思。你就会知道为什么了。只有正确理解这些东西，才能避免小白问题
> 
> 其实应该是说没有在乎那一点点系统的消耗，你没有编译进去的程序就不会运行了，左边是lean大的默认标配的，右边是我精简掉的。黄色的是我没有运行的程序，这些程序不在内存运行甚至都没有执行是不是会让软路由更快一些？

其实有些你确定不需要的组件可以在启动项里停止，然后禁止启动，这样这个组件就彻底不会运行了

hxlls, 2020-05-02:
> > 请正确理解网上所传说的“旁”路由和真正的旁路由(bypass)的区别，还有NAT 的意思。你就会知道为什么了。只有正确理解这些东西，才能避免小白问题
> 
> 其实应该是说没有在乎那一点点系统的消耗，你没有编译进去的程序就不会运行了，左边是lean大的默认标配的，右边是我精简掉的。黄色的是我没有运行的程序，这些程序不在内存运行甚至都没有执行是不是会让软路由更快一些？

[image]
除了我用htop 命令查看外，无非就是几个登录的守护进程使用了CPU，其他使用率都是0，何以来个更省资源的说法？比0% 更省资源的意思难道还能负值来增加硬件性能？有些东西还是要靠实际数据来支撑，而不是靠猜测至于内存，这个根本不需要考虑，X86的完全加载IMG也就256M足够还有剩余，别说现在X86还有人用256M内存的，至于普通的ARM 版本，也远远足够加载固件的所有需求，如果不够，你连WEB 升级都升级不了，因为WEB 升级是上传到内存，然后内存写入覆盖，也就是起码有剩余容量足够你固件大小，你才能上传

a1392429292, 2020-05-02:
我也是极度精简主义者，一般编译会操作如下

Extra packages > autosamba  取消smb

LuCI ---> Applications ---> luci-app-ddns   		#动态域名 DNS（集成阿里DDNS客户端）
LuCI ---> Applications ---> luci-app-firewall   	        #添加防火墙
LuCI ---> Applications ---> luci-app-flowoffload  	#Turbo ACC网络加速
LuCI ---> Applications ---> luci-app-mwan3   		#MWAN3负载均衡
LuCI ---> Applications ---> luci-app-ssr-plus   	        #乳酸菌饮料

使用下来，无感，没有觉得有任何的不舒服，500M电信宽带跑620

norikohxb, 2020-05-02:
> 我也是极度精简主义者，一般编译会操作如下
> 
> Extra packages > autosamba 取消smb
> 
> LuCI ---> Applications ---> luci-app-ddns #动态域名 DNS（集成阿里DDNS客户端）
> LuCI ---> Applications ---> luci-app-firewall #添加防火墙
> LuCI ---> Applications ---> luci-app-flowoffload #Turbo ACC网络加速
> LuCI ---> Applications ---> luci-app-mwan3 #MWAN3负载均衡
> LuCI ---> Applications ---> luci-app-ssr-plus #乳酸菌饮料
> 
> 使用下来，没有觉得有任何的不舒服，500M电信宽带跑620
> 
> UPNP，我一开始也是勾选上的，但是，后来觉得【防火墙】里手动配置端口更实在，但是玩端口转发的前提就是内网设备的内网ip不能变更，需要将租赁时间设置为infinite（永久）；
> 
> 甚至一段时间，我还在考虑【网络】里的 主机名/ipmac绑定/静态路由 之间是什么关系？

根据你的说法我推测你是主路由，而且感觉你也没有用老的安卓+youtube的app看youtube，你的手动添加应该是添加端口映射。比如对迅雷这种吧，象我的老安卓的youtube的app是哪个端口我都雾煞煞。

a1392429292, 2020-05-02:
是的，主路由，我一直在精简，精简一切用不到的功能；
【网络】-【防火墙】-【端口转发】
openwrt管理后台绕开80端口用8088；
群晖的5000端口映射出去；
群晖的5005的webdav映射出去；
我讲究什么设备做什么事，不怎么喜欢all in 。
.................
其实，我看了下你上面的描述，感觉不是upnp的事情，应该是【ssr plus】-【访问控制】部分的设置问题，初步感觉是这样。。。

norikohxb, 2020-05-02:
> [image]
> 除了我用htop 命令查看外，无非就是几个登录的守护进程使用了CPU，其他使用率都是0，何以来个更省资源的说法？比0% 更省资源的意思难道还能负值来增加硬件性能？有些东西还是要靠实际数据来支撑，而不是靠猜测至于内存，这个根本不需要考虑，X86的完全加载IMG也就256M足够还有剩余，别说现在X86还有人用256M内存的，至于普通的ARM [...]

hxlls非常感谢你的热心回答，其实我并不是要节省那个空间，那个100mb还是1g，我其实都无所谓，我的硬盘是512gb就跑ros和lede，所以就是给他100g的空间我也愿意。我精简的目的是觉得对我无用的东西越少越少，不驻留内存就更不会影响系统稳定性。就好像之前我装好操作系统以后，windows的主题，print spool， 都是先去关闭，这样关闭是为了节省系统资源也为了让系统更稳定，感谢lede一直以来的更新，比如在adbyby里面加入了可以添加订阅（但是那个订阅你们说是全局的）还有在ssrplus里面也加上广告过滤（那个竟然也是全局的，把国内的淘宝相关的一些域名也跟着封杀了）lean大的工作当然是为了更好更方便的让我们使用，但是其实坦白说这个开发也只是开发一半，我相信现在在ssr还开屏蔽广告的人很少。那个adbyby订阅的也很少，因为我感觉正常的ssrplus的广告屏蔽应该只是屏蔽ssr出去的翻墙这部分的，不应该屏蔽你国内的。当然我们也不能奢求lean大去完成，毕竟lean大的贡献很大了，但是从事实上来说我觉得ssr的广告过滤是个半成品。adbyby的订阅列表如果是全局的（我觉得也是半成品他应该只过滤国内不应该过滤翻墙的）当然我不会看源代码，我说了那么多不是在说lean大不好，只是说，程序越多，就越容易冲突或者是有不稳定因数，比如我不需要ddns，我理所当然就精简掉了，避免他留在里面碍事，我不需要qbittorrent不需要用lede下载我自然也把它精简掉。主要目的是为了整个系统更纯净。

不知道你是否能告诉我这几个能否精简掉。
urngd
sysfsutils
ipsec
done
autocore
urandom_seed
这个urandom好像是一个产生随机函数，我如果把它去掉会不会影响我的ssr性能和国内上网性能。

norikohxb, 2020-05-02:
> 是的，主路由，我一直在精简，精简一切用不到或重复的功能；
> 【网络】-【防火墙】-【端口转发】
> openwrt管理后台绕开80端口用8088；
> 群晖的5000端口映射出去；
> 群晖的5005的webdav映射出去；
> 我讲究什么设备做什么事，不怎么喜欢all in 。
> .................
> 其实，我看了下你上面的描述，感觉不是upnp的事情，应该是【ssr plus】-【访问控制】部分的设置问题，初步感觉是这样。。。

不是ssrplus的设置问题，都是默认设置，大陆白名单。然后我就一个trojan自己假设的gcp，现在·还剩下30美元。 因为我知识有限，纯粹喜欢软路由和翻墙，并不是像lean大那种真正程序打交道的，我觉得是   翻墙上youtube，如果你的youtube的app是用upnp上网，那么你也要开通upnp，你单独ros开没屁用，ros开upnp是给国内的网站的upnp用的，你翻墙的是lede，而lede你精简了upnp，把它阉割了，那么你平时就无法用upnp去访问youtube视频。然后我相信在3月底之前的lede就算编译的时候不带upnp，那么他系统对于ssr的upnp还是自带支持的，而到了4月份的lede，lean大把ssr里面的upnp阉割了，这个功能专门交给upnp这个插件。而我精简了，变成了上不去。当然这个是我的拙见，我之所以说出来是希望hxlls或者lean大看到帖子可以批评教育我一下

a1392429292, 2020-05-02:
https://downloads.openwrt.org/releases/19.07.2/targets/x86/64/  刷这里的包，再加上你想要的ipk软件，可以说是最纯粹的精简了。

norikohxb, 2020-05-02:
> https://downloads.openwrt.org/releases/19.07.2/targets/x86/64/ 刷这里的包，再加上你想要的ipk软件，可以说是最纯粹的精简了。
这个是openwrt原版的，我现在喜欢lean大的lede，虽然我不知道他有什么好处（不是程序猿）但是一路研究起来顺风顺水。

a1392429292, 2020-05-03:
其实，用原版的官方openwrt固件，就是最极简；
ssrplus好像现在恩山有专人在维护更新；
我个人感觉，用原版官方openwrt固件配合ipk，真的是原汁原味。

norikohxb, 2020-05-03:
> 其实，用原版的官方openwrt固件，就是最极简；
> ssrplus好像现在恩山有专人在维护更新；
> 我个人感觉，用原版官方openwrt固件配合ipk，真的是原汁原味。

装了一下，没有觉得是极简的，如果真的要用也是自己去编译openwrt，不过目前只要lean大还没有放弃lede，我就会用lede下去，你那个并不精简，当然也可以去自己编译更精简。但是现在还是lede用的习惯

9l, 2020-05-03:
基本服务包不要乱简，一般把 luci app 去掉就行了，比如你说那个 upnp，
`# CONFIG_PACKAGE_luci-app-upnp is not set`
把这个luci app去掉就行了，就不会在服务里出现了

norikohxb, 2020-05-04:
> 基本服务包不要乱简，一般把 luci app 去掉就行了，比如你说那个 upnp，
> `# CONFIG_PACKAGE_luci-app-upnp is not set`
> 把这个luci app去掉就不会在服务里出现了
> 你看我这个简不简，只是去掉了不需要的luci app而已
> [image]

感谢你的回答，我是吧luci-app-upnp关掉了好几个月了，在4月份之前都没有发现问题，就是4月份之后发现我的老安卓平板新的固件如果去掉unpnp无法播放youtube的视频（用youtube的apk）所以只能又开启了。

9l, 2020-05-04:
0202 年了，老手机可以淘汰了

norikohxb, 2020-05-04:
> 0202 年了，老手机可以淘汰了

卫生间，洗脸刷牙洗澡的时候看，舍不得用高端平板。说实话。
[image]
[image]

ElectronicElephant, 2020-05-04:
之前和楼主在另一个issue里聊过，但是我水平不够，就没有后续跟进。

我很赞同

> 
> 
> 基本服务包不要乱简，一般把 luci app 去掉就行了，比如你说那个 upnp，
> `# CONFIG_PACKAGE_luci-app-upnp is not set`
> 把这个luci app去掉就不会在服务里出现了
> 你看我这个简不简，只是去掉了不需要的luci app而已
> [image]

badcrazy, 2020-05-07:
> > https://downloads.openwrt.org/releases/19.07.2/targets/x86/64/ 刷这里的包，再加上你想要的ipk软件，可以说是最纯粹的精简了。
> > 非常感谢你的推荐，你这个是openwrt原版的，我现在喜欢lean大的lede，虽然我不知道他有什么好处（不是程序猿）但是一路研究起来顺风顺水。其实这个ssrplus应该是lean的作品，所以我觉得平台也应该用它的lede应该会好一些。我要是用clash之类的我可能就会用纯正的openwrt。

我现在k3就是官方op加上自己编译的lean插件，这样确实是最精简的，当然直接编译lean大的精简也可以，但是不过我是云编译，lean大经常调整默认的软件配置模板，等编译好了发现多了组件再去修改编译配置精简比较麻烦，比如按照默认设置再加上自己需要的组件编译并输出ipk，在官方op安装需要的ipk即可

norikohxb, 2020-05-28:
https://www.right.com.cn/forum/thread-3708700-1-1.html
问题还是在这里，其实很多人都发现了
我是一开始就发现的3月20日的lede那个时候是都不会，受影响的是老版本的安卓+youtube。
安卓10不会有这个问题。

closed 2020-05-30 by norikohxb