coolsnowwolf/lede #4497: 直播软件引起路由重启问题

Aes64X, 2020-05-01:
没有日志，因为它二话不说直接重启了。。。。
设备：竞斗云    源码非最新pull，大约半个月前的没更新，仅在此处问问各位有遇见过类似情况吗。

N1盒子装的直播软件。
路由放着不动，随便联网都没事。只要一开直播软件，路由立刻重启。即使PPPOE断掉，无外网状态，只要点开直播软件，二话不说。。。重启。
我醉了。。。。。。

Aes64X, 2020-05-01:
我在将 packages 部分退回到 cd354e8b1abc165617c519e24cd0e55fb3b35018 后，此问题暂未出现。我会进行长时间测试。

R3pl4c3r, 2020-05-06:
有没有装docker？如果装了docker可能是 coolsnowwolf/packages@6689ca74fb96e32b6e0dfdbea0d40ba50ec8f6ff docker版本更新的问题。据我所知，docker会影响UPnP和UDP，会造成一些莫名其妙的问题

Aes64X, 2020-05-06:
> 有没有装docker？如果装了docker可能是 [coolsnowwolf/packages@6689ca7](https://github.com/coolsnowwolf/packages/commit/6689ca74fb96e32b6e0dfdbea0d40ba50ec8f6ff) docker版本更新的问题。据我所知，docker会影响UPnP和UDP，会造成一些莫名其妙的问题

没有docker，我已经确定，在我提到的源码节点之后的更新中，某个组件会引起此问题。我退回到我提到的部分，已测试3天，没有任何问题。

closed 2020-05-08 by Aes64X