coolsnowwolf/lede #4501: 编译的时候，给的Root Filesystem 越大，failed to allocate 4989 blocks 也越大

li-xunhuan, 2020-05-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译的时候，给的Root Filesystem 越大，failed to allocate 4989 blocks 也越大，当前配置60+271

### (2) 路由器型号和固件版本

A：软路由

### (3) 详细日志

A：

[image]

9l, 2020-05-02:
你啥机子，多少容量的空间，我一直都是64+1024都没问题

li-xunhuan, 2020-05-03:
> 你啥机子，多少容量的空间，我一直都是64+1024都没问题
> 对应的文件系统模块选了么，比如:
> CONFIG_PACKAGE_kmod-fs-ext4
> CONFIG_PACKAGE_kmod-fs-squashfs

J1900，4+32G
该选的都选了，我就是觉得镜像太大了，想压缩小一点儿。
就只新增了一个Argon主题、一个OpenClash、一个PassWall，我觉得没必要给450的 Root Filesystem 啊！！
但是给小了就报错，上头……
而且有时候加大了之后反而`failed to allocate xxx blocks` 的值更大了，很迷惑！！

closed 2020-07-04 by coolsnowwolf