coolsnowwolf/lede #4504: 斐讯K2P编译失败

KevinMX, 2020-05-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
斐讯K2P / MT7621 编译失败

### (1) 具体问题

A：斐讯K2P编译失败，似乎是与新增加的Redmi AC2100支持有关，Revert commit [
01aeb5d
](https://github.com/coolsnowwolf/lede/commit/01aeb5df0ed4c2119c9cbdb072872089d4739a58)之后恢复正常。

### (2) 路由器型号和固件版本

A：Phicomm K2P / R20.4.8

### (3) 详细日志

A：部分log:
```
$ ionice -c 3 nice -n 20 make -j8 V=s CONFIG_DEBUG_SECTION_MISMATCH=y 2>&1 | tee build.log
.....
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xzwifi_creativebox-v1.dtb /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xzwifi_creativebox-v1.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_mir3p.dtb.tmp
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/arch/mips/boot/dts -I/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/arch/mips/boot/dts/include -I/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/include/ -undef -D__DTS__  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_zbtlink_zbt-we1326.dtb.tmp ../dts/mt7621_zbtlink_zbt-we1326.dts
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

ERROR: Input tree has errors, aborting (use -f to force output)
make[5]: *** [Makefile:202: /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb] Error 2
make[5]: *** Waiting for unfinished jobs....
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_mir3g-v2.dtb.tmp
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_youhua_wr1200js.dtb /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_youhua_wr1200js.dtb.tmp
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_youku_yk-l2.dtb /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_youku_yk-l2.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaoyu_xy-c5.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xzwifi_creativebox-v1.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_youhua_wr1200js.dtb.tmp
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_zbtlink_zbt-we1326.dtb /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_zbtlink_zbt-we1326.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_youku_yk-l2.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_zbtlink_zbt-we1326.dtb.tmp
make[5]: Leaving directory '/tmp/lede/target/linux/ramips/image'
make[4]: *** [Makefile:26: install] Error 2
make[4]: Leaving directory '/tmp/lede/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/tmp/lede/target/linux'
time: target/linux/install#54.58#16.21#48.83
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/tmp/lede'
make[1]: *** [target/Makefile:19: /tmp/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/tmp/lede'
make: *** [/tmp/lede/include/toplevel.mk:227：world] 错误 2
```
config: https://github.com/KevinMX/openwrt-phicomm-k2p-build/blob/master/.config
完整log:
[build.log](https://github.com/coolsnowwolf/lede/files/4563918/build.log)

waynon, 2020-05-01:
make download一下

KevinMX, 2020-05-01:
> make download一下

前面已经 `make download` 过了，问题不在这，问题应该是出在AC2100的dts。
刚刚看到 @AmadeusGhost 貌似已经修了（[PR#4500](https://github.com/coolsnowwolf/lede/pull/4500)），不过Lean还没合PR，等修好我再关issue吧。

AmadeusGhost, 2020-05-01:
可以合并我的pr试试，我就理论改了下，没试
```
## pwd:lede
wget https://github.com/coolsnowwolf/lede/commit/8a2097ae61a9f90884d17bb304b1a3ca365e3d37.patch
## rename filename to 1.patch
git apply 1.patch
```

KevinMX, 2020-05-01:
> 可以合并我的pr试试，我就理论改了下，没试
> 
> ```
> ## pwd:lede
> wget https://github.com/coolsnowwolf/lede/commit/8a2097ae61a9f90884d17bb304b1a3ca365e3d37.patch
> ## rename filename to 1.patch
> git apply 1.patch
> ```

```
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_wevo_11acnas.dtb.tmp
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_wevo_w2914ns-v2.dtb.tmp
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_mir3g.dtb /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_mir3g.dtb.tmp

/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"

ERROR: Input tree has errors, aborting (use -f to force output)
```
应用patch之后也还是在报错……

AmadeusGhost, 2020-05-01:
> 应用patch之后也还是在报错……尝试make clean过，没有效果。

我知道了，还有几处地方写错了没注意，我上下编译机看看

AmadeusGhost, 2020-05-01:
> 应用patch之后也还是在报错……尝试make clean过，没有效果。

https://github.com/AmadeusGhost/lede/commit/7ce65890a4844d40615a362437c8c6141f0dab86
现在应该可以编译过了，这个实在是,,,ramips 4.14的写法太草了😂

KevinMX, 2020-05-01:
> > 应用patch之后也还是在报错……尝试make clean过，没有效果。
> 
> ```
> wget https://github.com/AmadeusGhost/lede/commit/7ce65890a4844d40615a362437c8c6141f0dab86.patch
> ```
> 
> 现在应该可以编译过了，这个实在是,,,ramips 4.14的写法太草了

刚才fail了一次，我make clean再试试😂

AmadeusGhost, 2020-05-01:
> 刚才fail了一次，我make clean再试试😂

再看了下，还有一处漏了改😂
```
wget https://github.com/coolsnowwolf/lede/commit/17342e7eb902135aad8bff3f4084d9e5e8b4b3d7.patch
```

KevinMX, 2020-05-01:
> > 刚才fail了一次，我make clean再试试joy
> 
> 再看了下，还有一处漏了改
> 
> ```
> wget https://github.com/coolsnowwolf/lede/commit/17342e7eb902135aad8bff3f4084d9e5e8b4b3d7.patch
> ```

出现了，是新的报错🙈
```
rm -f /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_wevo_w2914ns-v2.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/arch/mips/boot/dts -I/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/arch/mips/boot/dts/include -I/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.176/include/ -undef -D__DTS__  -o /tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_youhua_wr1200js.dtb.tmp ../dts/mt7621_youhua_wr1200js.dts
/tmp/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_xiaomi_redmi-router-ac2100.dtb: ERROR (path_references): Reference to non-existent node or label "gmac0"

ERROR: Input tree has errors, aborting (use -f to force output)
```

AmadeusGhost, 2020-05-01:
> 出现了，是新的报错🙈

git pull一下再打patch，之前那个commit已修复，我自己测了下编译ok

KevinMX, 2020-05-01:
> > 出现了，是新的报错see_no_evil
> 
> git pull一下再打patch，之前那个commit已修复，我自己测了下编译ok

没注意到前面已经merge了，现在没问题了，可以Merge了 xD

PR #4508 (2020-05-01): ramips: fix build error
这个实在是太坑了，我没注意到 Orz
fix #4504
closed 2020-05-01 by KevinMX