coolsnowwolf/lede #4507: 请教一下关于TP9343（AR71xx）增加32M Flash的兼容问题

w634319428, 2020-05-01:
在最新版本的Openwrt中编译过程中，由于16M Flash可以加的插件实在太少了，我想折腾一下试试32M的闪存。CPU是TP9343（AR71xx），在target/linux/image的tiny-tp-link.mk下把原来的$(Device/tplink-8mlzma)改成了$(Device/tplink-32mlzma)，并在相同目录下的common-tp-link.mk文件增加了如下代码：
`define Device/tplink-32mlzma
  $(Device/tplink-lzma)
  TPLINK_FLASHLAYOUT := 32Mlzma
  IMAGE_SIZE := 31744k
endef`

然后在tools/firmware-utils目录下的mktplinkfw.c增加了这段代码：
`       {
		.id		= "32M",
		.fw_max_len	= 0x‭1f00000‬,
		.kernel_la	= 0x80060000,
		.kernel_ep	= 0x80060000,
		.rootfs_ofs	= 0x140000,
	}, {
		.id		= "32Mlzma",
		.fw_max_len	= 0x‭1f00000‬,
		.kernel_la	= 0x80060000,
		.kernel_ep	= 0x80060000,
		.rootfs_ofs	= 0x100000,
	}, {
		.id		= "32Mppc",
		.fw_max_len	= 0x‭1f00000‬‬,
		.kernel_la	= 0x00000000 ,
		.kernel_ep	= 0xc0000000,
		.rootfs_ofs	= 0x2a0000,
	}`
这样加完还有其他需要修改的嘛？因为我看好像没有和TL有关的dst文件。
刚开始接触Openwrt，想请教一下大佬，谢谢。

haxc, 2020-05-04:
目前好像ar系列芯片的32m支持没有做有代码支持，因为以前旧版openwrt代码的32m支持无法使用在新版openwrt上

lunatickochiya, 2020-05-08:
默认支持32M 直接用吧 具体要想知道哪个commit的话 https://github.com/openwrt/openwrt/commit/3fa85af5add3eea8a7c2bbceb89988d660a17579#diff-de1e25af1f1991eda5b590895d00d19d  
参阅这里吧

closed 2020-07-04 by coolsnowwolf