coolsnowwolf/lede #4511: k2p a1 移动光猫桥接pppoe后会间歇性断网并且无法自动重连

WROIATE, 2020-05-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

移动光猫桥接openwrt后连接上pppoe一段时间网会连不上，但是WAN口是有ip的，概况界面的ipv6接口会消失显示问号，需要重新再接口界面的WAN口处点击连接才能恢复，并且日志并无报错，只能看到ssr显示断连表示网络断开
路由器设置了vlan单线复用，wan-lan同线

### (2) 路由器型号和固件版本

k2p a1
OpenWrt R20.4.8 / LuCI Master (git-20.117.60969-420c61a)

### (3) 详细日志

断开网络连接时的日志，标志是ssr连接失败
```
Sat May  2 07:19:03 2020 daemon.err nlbwmon[2329]: Unable to dump conntrack: No buffer space available
Sat May  2 07:19:50 2020 authpriv.info dropbear[811]: Child connection from 192.168.1.104:11664
Sat May  2 07:19:50 2020 authpriv.info dropbear[811]: Exit before auth: Error reading: Connection reset by peer
Sat May  2 07:20:01 2020 authpriv.info dropbear[822]: Child connection from 192.168.1.104:11669
Sat May  2 07:20:05 2020 authpriv.notice dropbear[822]: Password auth succeeded for 'root' from 192.168.1.104:11669
Sat May  2 07:27:28 2020 kern.err /usr/bin/ssr-redir[13733]: server recv: Connection reset by peer
Sat May  2 07:29:51 2020 cron.info crond[2208]: crond (busybox 1.30.0) started, log level 8
Sat May  2 07:29:53 2020 cron.info crond[2331]: crond (busybox 1.30.0) started, log level 8
Sat May  2 07:33:24 2020 kern.err /usr/bin/ssr-redir[13725]: server recv: Connection reset by peer
Sat May  2 07:33:27 2020 kern.err /usr/bin/ssr-redir[13727]: server recv: Connection reset by peer
Sat May  2 07:33:27 2020 kern.err /usr/bin/ssr-redir[13731]: server recv: Connection reset by peer
Sat May  2 07:35:33 2020 authpriv.info dropbear[23484]: Exit (root): Exited normally
Sat May  2 07:38:40 2020 kern.err /usr/bin/ssr-redir[13723]: remote recv: Connection reset by peer
Sat May  2 07:46:31 2020 authpriv.info dropbear[7557]: Child connection from 192.168.1.104:12006
Sat May  2 07:46:32 2020 authpriv.warn dropbear[7557]: Bad password attempt for 'root' from 192.168.1.104:12006
Sat May  2 07:46:34 2020 authpriv.notice dropbear[7557]: Password auth succeeded for 'root' from 192.168.1.104:12006
Sat May  2 07:46:45 2020 kern.err /usr/bin/ssr-redir[13703]: connect: Permission denied
Sat May  2 07:46:45 2020 kern.err /usr/bin/ssr-redir[13707]: connect: Permission denied
Sat May  2 07:46:47 2020 kern.err /usr/bin/ssr-redir[13731]: connect: Permission denied
Sat May  2 07:46:48 2020 kern.err /usr/bin/ssr-redir[13711]: connect: Permission denied
Sat May  2 07:46:49 2020 kern.err /usr/bin/ssr-redir[13727]: connect: Permission denied
Sat May  2 07:46:49 2020 kern.err /usr/bin/ssr-redir[13719]: connect: Permission denied
```

重新再接口界面点击连接之后的日志
```
Sat May  2 07:50:01 2020 kern.err /usr/bin/ssr-redir[13711]: connect: Permission denied
Sat May  2 07:50:02 2020 kern.err /usr/bin/ssr-redir[13713]: connect: Permission denied
Sat May  2 07:50:02 2020 kern.err /usr/bin/ssr-redir[13715]: connect: Permission denied
Sat May  2 07:50:02 2020 daemon.notice netifd: Interface 'wan_6' is now down
Sat May  2 07:50:02 2020 daemon.notice netifd: Network alias '' link is down
Sat May  2 07:50:02 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Sat May  2 07:50:02 2020 daemon.notice netifd: Interface 'wan_6' is disabled
Sat May  2 07:50:02 2020 daemon.info pppd[29448]: Terminating on signal 15
Sat May  2 07:50:02 2020 daemon.info pppd[29448]: Connect time 44.3 minutes.
Sat May  2 07:50:02 2020 daemon.info pppd[29448]: Sent 4507367 bytes, received 45416693 bytes.
Sat May  2 07:50:02 2020 daemon.err miniupnpd[2898]: ioctl(s, SIOCGIFADDR, ...): Address not available
Sat May  2 07:50:02 2020 daemon.err miniupnpd[2898]: Failed to get IP for interface pppoe-wan
Sat May  2 07:50:02 2020 daemon.warn miniupnpd[2898]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sat May  2 07:50:02 2020 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sat May  2 07:50:07 2020 daemon.notice netifd: Interface 'wan' is now down
Sat May  2 07:50:07 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sat May  2 07:50:07 2020 daemon.err insmod: module is already loaded - slhc
Sat May  2 07:50:07 2020 daemon.err insmod: module is already loaded - ppp_generic
Sat May  2 07:50:07 2020 daemon.err insmod: module is already loaded - pppox
Sat May  2 07:50:07 2020 daemon.err insmod: module is already loaded - pppoe
Sat May  2 07:50:07 2020 daemon.notice netifd: wan (8938): uci: Entry not found
Sat May  2 07:50:07 2020 daemon.info pppd[8982]: Plugin rp-pppoe.so loaded.
Sat May  2 07:50:07 2020 daemon.info pppd[8982]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sat May  2 07:50:08 2020 daemon.notice pppd[8982]: pppd 2.4.8 started by root, uid 0
Sat May  2 07:50:08 2020 daemon.info pppd[8982]: PPP session is 42818
Sat May  2 07:50:08 2020 daemon.warn pppd[8982]: Connected to 00:00:5e:00:01:15 via interface eth0.41
Sat May  2 07:50:08 2020 daemon.info pppd[8982]: Renamed interface ppp0 to pppoe-wan
Sat May  2 07:50:08 2020 daemon.info pppd[8982]: Using interface pppoe-wan
Sat May  2 07:50:08 2020 daemon.notice pppd[8982]: Connect: pppoe-wan <--> eth0.41
Sat May  2 07:50:11 2020 daemon.info pppd[8982]: syncppp not active
Sat May  2 07:50:11 2020 daemon.info pppd[8982]: Remote message: Authentication success,Welcome!
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: PAP authentication succeeded
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: peer from calling number xx:00:5E:00:01:xx authorized
Sat May  2 07:50:11 2020 daemon.err miniupnpd[2898]: Failed to get IP for interface pppoe-wan
Sat May  2 07:50:11 2020 daemon.warn miniupnpd[2898]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sat May  2 07:50:11 2020 daemon.err miniupnpd[2898]: Failed to get IP for interface pppoe-wan
Sat May  2 07:50:11 2020 daemon.warn miniupnpd[2898]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: local  IP address 10.233.161.33
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: remote IP address 10.233.0.1
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: primary   DNS address 211.139.5.27
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: secondary DNS address 211.139.5.28
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: local  LL address xxxx::747d:24a7:4293:xxxx
Sat May  2 07:50:11 2020 daemon.notice pppd[8982]: remote LL address xxxx::0200:5eff:fe00:xxxx
Sat May  2 07:50:11 2020 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sat May  2 07:50:11 2020 daemon.notice netifd: Interface 'wan' is now up
Sat May  2 07:50:11 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sat May  2 07:50:11 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Sat May  2 07:50:11 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sat May  2 07:50:11 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Sat May  2 07:50:12 2020 kern.err /usr/bin/ssr-redir[13721]: connect: Permission denied
Sat May  2 07:50:13 2020 kern.err /usr/bin/ssr-redir[13713]: connect: Permission denied
Sat May  2 07:50:14 2020 kern.err /usr/bin/ssr-redir[13731]: connect: Permission denied
Sat May  2 07:50:15 2020 daemon.notice netifd: Interface 'wan_6' is now up
Sat May  2 07:50:16 2020 kern.err /usr/bin/ssr-redir[13709]: server recv: Connection reset by peer
Sat May  2 07:50:22 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sat May  2 07:50:24 2020 cron.info crond[9490]: crond (busybox 1.30.0) started, log level 8
Sat May  2 07:50:25 2020 cron.info crond[9600]: crond (busybox 1.30.0) started, log level 8
Sat May  2 07:50:27 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Sat May  2 07:50:29 2020 cron.info crond[9961]: crond (busybox 1.30.0) started, log level 8
Sat May  2 07:50:30 2020 cron.info crond[10071]: crond (busybox 1.30.0) started, log level 8
Sat May  2 07:50:42 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sat May  2 07:50:44 2020 cron.info crond[10485]: crond (busybox 1.30.0) started, log level 8
Sat May  2 07:50:46 2020 cron.info crond[10595]: crond (busybox 1.30.0) started, log level 8
```

WROIATE, 2020-05-02:
已解决，是防火墙未使用MASQUERADE进行包装pppoe口的流量

closed 2020-05-02 by WROIATE

ZhinianQIN, 2020-05-24:
请问你是如何解决的，能给出具体的解决方法吗，我也是遇到这个问题，48小时后电信强制中断我的pppoe，然后dhcpv6失败

ZhinianQIN, 2020-05-24:
@WROIATE

WROIATE, 2020-07-08:
> 请问你是如何解决的，能给出具体的解决方法吗，我也是遇到这个问题，48小时后电信强制中断我的pppoe，然后dhcpv6失败

抱歉没注意你的@，主要是添加这条命令`iptables -t nat -I POSTROUTING -o pppoe-wan -j MASQUERADE`